zwave-js/zwave-js #5691: Motion sensor connected via USB power stops detecting motion

ocalvo, 2023-04-22:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

What do you observe?
After restarting the host system, some the Aeotec multisensor 7 sensors stop sending motion events.

What did you expect to happen?
Never

Steps to reproduce the behavior:
  1. Restart the ZwaveJS
  2. Trigger motion event in a multisensor 7 device.

Log is filled with:
```
2023-04-22T02:45:04.494Z DRIVER « [Node 020] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -73 dBm
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_NoSPAN
2023-04-22T02:45:04.498Z CNTRLR » [Node 020] No SPAN is established yet, cannot decode command. Requesting a non
                                  ce...
2023-04-22T02:45:04.510Z SERIAL » 0x012000a90114149f027f01d6ce365a83c881a8168e0b874c6c55ba05000000009 (34 bytes)
                                  ec2
2023-04-22T02:45:04.512Z DRIVER » [Node 020] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x05
                                  │ callback id:      158
                                  └─[Security2CCNonceReport]
                                      sequence number:  127
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0xd6ce365a83c881a8168e0b874c6c55ba
2023-04-22T02:45:04.519Z SERIAL « [ACK]                                                                   (0x06)
2023-04-22T02:45:04.522Z SERIAL « 0x010401a90152                                                       (6 bytes)
2023-04-22T02:45:04.524Z SERIAL » [ACK]                                                                   (0x06)
2023-04-22T02:45:04.526Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2023-04-22T02:45:04.534Z SERIAL « 0x011d00a99e00000100b77f7f7f7f00000300000000030100007f7f7f7f7f1d    (31 bytes)
2023-04-22T02:45:04.535Z SERIAL » [ACK]                                                                   (0x06)
2023-04-22T02:45:04.538Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            158
                                    transmit status:        OK, took 10 ms
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 100 kbit/s
                                    ACK RSSI:               -73 dBm
                                    ACK channel no.:        0
                                    TX channel no.:         0
2023-04-22T02:45:04.547Z SERIAL « 0x012a00a8000114219f0367011241badc6634d40626ed61ea8238361db49db6900 (44 bytes)
                                  bf276a24eac39ded800b890
2023-04-22T02:45:04.551Z SERIAL » [ACK]                                                                   (0x06)
2023-04-22T02:45:04.554Z CNTRLR   [Node 020] [~] [Battery] level: 100 => 100                        [Endpoint 0]
2023-04-22T02:45:04.556Z CNTRLR   [Node 020] [~] [Battery] isLow: false => false                    [Endpoint 0]
2023-04-22T02:45:04.559Z DRIVER « [Node 020] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -72 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 103
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0xbadc6634d40626ed61ea8238361db49d
                                    └─[BatteryCCReport]
                                        level:  100
                                        is low: false
```

### Device information

Manufacturer: Aeotec
Model name: Multisensor 7
Node ID in your network: 20

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 
`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

yes (please describe)

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/11300105/zwave_js.log)

AlCalzone, 2023-04-27:
I don't see anything obviously wrong with that log. That error you mention happens when the node sends it first report where the shared encryption state between Z-Wave JS and the node is out of sync. This is then re-synchronized and the node re-transmits its command, which is a battery report:
```
2023-04-22T02:45:04.494Z DRIVER « [Node 020] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -73 dBm
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_NoSPAN
2023-04-22T02:45:04.512Z DRIVER » [Node 020] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x05
                                  │ callback id:      158
                                  └─[Security2CCNonceReport]
                                      sequence number:  127
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0xd6ce365a83c881a8168e0b874c6c55ba
2023-04-22T02:45:04.559Z DRIVER « [Node 020] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -72 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 103
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0xbadc6634d40626ed61ea8238361db49d
                                    └─[BatteryCCReport]
                                        level:  100
                                        is low: false
```

The other occurences are on other S2-capable nodes, so I guess you recently restarted?

AlCalzone, 2023-04-27:
I suppose you're running into this bug in the sensor: 
[image]

Try a firmware update: https://aeotec.freshdesk.com/support/solutions/articles/6000255731-update-multisensor-7-v1-4-2

ocalvo, 2023-05-03:
I attempted the firmware update.  However, after the update the device was not able to be interviewed. 
This sensor is in the ceiling, it will be very hard for me to try to reset the device.

```
2023-05-03T00:19:16.552Z CNTRLR « [Node 020] received information for parameter #1:
                                  parameter name:      Beaming Enable/Disable
                                  value format:        Enumerated
                                  value size:          1 bytes
                                  min value:           0
                                  max value:           1
                                  default value:       0
                                  is read-only:        false
                                  is advanced (UI):    false
                                  has bulk support:    false
                                  alters capabilities: false
2023-05-03T00:19:16.557Z CNTRLR » [Node 020] querying parameter #2 information...
2023-05-03T00:19:16.716Z DRIVER   Dropping message because it could not be deserialized: The command class Appli
                                  cation Status is not implemented (ZW0303)
2023-05-03T00:19:17.658Z CNTRLR   [Node 020] failed to decode the message, retrying with SPAN extension...
2023-05-03T00:19:17.679Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2023-05-03T00:19:17.918Z DRIVER   Dropping message because it could not be deserialized: The command class Appli
                                  cation Status is not implemented (ZW0303)
2023-05-03T00:19:18.470Z CNTRLR « [Node 020] received information for parameter #2:
                                  parameter name:      Motion Retrigger Time
                                  value format:        UnsignedInteger
                                  value size:          2 bytes
                                  min value:           0
                                  max value:           3600
                                  default value:       30
                                  is read-only:        false
                                  is advanced (UI):    false
                                  has bulk support:    false
                                  alters capabilities: false
2023-05-03T00:19:18.472Z CNTRLR » [Node 020] querying parameter #3 information...
2023-05-03T00:19:18.651Z DRIVER   Dropping message because it could not be deserialized: The command class Appli
                                  cation Status is not implemented (ZW0303)
2023-05-03T00:19:19.668Z CNTRLR   [Node 020] Timed out while waiting for a response from the node (ZW0201)
2023-05-03T00:19:19.670Z CNTRLR « [Node 020] received information for parameter #3:
                                  parameter name:      Motion Untrigger Time
                                  value format:        UnsignedInteger
                                  value size:          2 bytes
                                  min value:           30
                                  max value:           3600
                                  default value:       240
                                  is read-only:        false
                                  is advanced (UI):    false
                                  has bulk support:    false
                                  alters capabilities: false
2023-05-03T00:19:19.676Z CNTRLR » [Node 020] querying parameter #4 information...
2023-05-03T00:19:19.738Z CNTRLR   [Node 020] failed to decode the message, retrying with SPAN extension...
2023-05-03T00:19:20.320Z CNTRLR   [Node 020] failed to decode the message, retrying with SPAN extension...
2023-05-03T00:19:20.342Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2023-05-03T00:19:20.525Z DRIVER   Dropping message because it could not be deserialized: The command class Appli
                                  cation Status is not implemented (ZW0303)
2023-05-03T00:19:21.055Z DRIVER   Dropping message because it could not be deserialized: The command class Appli
                                  cation Status is not implemented (ZW0303)
2023-05-03T00:19:21.592Z CNTRLR « [Node 020] received information for parameter #4:
                                  parameter name:      Motion Sensitivity
                                  value format:        UnsignedInteger
                                  value size:          1 bytes
                                  min value:           0
                                  max value:           11
                                  default value:       11
                                  is read-only:        false
                                  is advanced (UI):    false
                                  has bulk support:    false
                                  alters capabilities: false
2023-05-03T00:19:21.597Z CNTRLR » [Node 020] querying parameter #5 information...
[... 222 lines cut ...]
2023-05-03T00:20:25.546Z CNTRLR   [Node 020] Timed out while waiting for a response from the node (ZW0201)
2023-05-03T00:20:25.549Z CNTRLR « [Node 020] received information for parameter #5:
                                  parameter name:      Motion Report Type
                                  value format:        Enumerated
                                  value size:          1 bytes
[... cut ...]
2023-05-03T00:20:28.322Z CNTRLR   [Node 020] failed to decode the message, retrying with SPAN extension...
2023-05-03T00:20:28.787Z CNTRLR   [Node 020] failed to decode the message after re-transmission with SPAN extens
                                  ion, dropping the message.
2023-05-03T00:20:28.791Z CNTRLR   [Node 020] Failed all interview attempts, giving up.
2023-05-03T00:20:28.851Z DRIVER   Dropping message because it could not be deserialized: The command class Appli
                                  cation Status is not implemented (ZW0303)
```

ocalvo, 2023-05-03:
Should I restart the Zwave addon?
What should be the next step other than remove the device and readd ?

AlCalzone, 2023-05-03:
Ok looks like you're now running into an issue I [fixed](https://github.com/zwave-js/node-zwave-js/pull/5731) yesterday. Will make sure to get a new release out soon.

ocalvo, 2023-05-05:
After forcing interview multiple times I recovered the device.
However, the lock up issue is still happening, even with firmware 1.4.2

[image]

AlCalzone, 2023-05-05:
> After forcing interview multiple times I recovered the device.

That does sound like you have some connectivity issues and the reports just get lost.

ocalvo, 2023-05-05:
> > After forcing interview multiple times I recovered the device.
> 
> That does sound like you have some connectivity issues and the reports just get lost.

Aeotec support reply with a beta version of a new firmware: 1.5.3 beta.
I will update to this version and report back here.

ocalvo, 2023-05-09:
Unfortunately, even in version 1.5.3 of the firmware the issue still repro.

AlCalzone, 2023-05-10:
You may want to temporarily switch to Z-Wave JS UI and look into the connection strength with a health check.
Although if the firmware update went through, this can't be that bad - or did you move it for the update?

ocalvo, 2023-05-10:
> You may want to temporarily switch to Z-Wave JS UI and look into the connection strength with a health check.
> Although if the firmware update went through, this can't be that bad - or did you move it for the update?

How do I switch to zwave js UI without loosing all the settings?
I there instructions somewhere?

AlCalzone, 2023-05-10:
https://www.home-assistant.io/integrations/zwave_js/#how-do-i-switch-between-the-official-z-wave-js-add-on-and-the-z-wave-js-ui-add-on

Minus step 5, do not uninstall if you plan to switch back. Switching back works similarly, but you check the "use the supervisor addon" box.

ocalvo, 2023-05-12:
I failed to perform the migration to ZwaveJS-UI. 
Problems:
1. The new addon does not allow me to edit the security config:
[image]
2. The add-on fails to start correcty:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------
 Add-on: Z-Wave JS UI
 Fully configurable Z-Wave JS gateway and control panel
-----------------------------------------------------------
 Add-on version: 1.12.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 10.1  (aarch64 / yellow)
 Home Assistant Core: 2023.5.2
 Home Assistant Supervisor: 2023.04.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
Log level is set to INFO
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-zwave-js-ui: starting
s6-rc: info: service init-nginx: starting
s6-rc: info: service init-nginx successfully started
s6-rc: info: service init-zwave-js-ui successfully started
s6-rc: info: service zwave-js-ui: starting
s6-rc: info: service zwave-js-ui successfully started
s6-rc: info: service discovery: starting
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
[08:48:06] INFO: Starting the Z-Wave JS UI...
2023-05-12 08:48:13.660 INFO APP: Version: 8.15.0
2023-05-12 08:48:13.672 INFO APP: Application path:/opt
  ______  __          __                      _  _____     _    _ _____ 
 |___  /  \ \        / /                     | |/ ____|   | |  | |_   _|
    / /____\ \  /\  / /_ ___   _____         | | (___     | |  | | | |  
   / /______\ \/  \/ / _' \ \ / / _ \    _   | |\___ \    | |  | | | |  
  / /__      \  /\  / (_| |\ V /  __/   | |__| |____) |   | |__| |_| |_ 
 /_____|      \/  \/ \__,_| \_/ \___|    \____/|_____/     \____/|_____|
2023-05-12 08:48:13.722 WARN STORE: scenes.json not found
2023-05-12 08:48:13.725 WARN STORE: nodes.json not found
2023-05-12 08:48:13.728 WARN STORE: users.json not found
2023-05-12 08:48:13.759 INFO APP: Listening on port 44920 host 0.0.0.0 protocol HTTP
[08:48:13] INFO: Starting NGinx...
2023-05-12 08:48:14.116 INFO MQTT: MQTT is disabled
2023-05-12 08:48:14.135 WARN BACKUP: Store backup is disabled
2023-05-12 08:48:14.140 WARN BACKUP: Nvm backup is disabled
2023-05-12 08:48:14.147 WARN Z-WAVE: Z-Wave driver not inited, no port configured
2023-05-12 08:48:21.313 INFO APP: GET /health/zwave 200 62.203 ms - 2140
2023-05-12 08:48:48.684 INFO APP: GET / 200 6.196 ms - 2140
2023-05-12 08:48:48.787 INFO APP: GET /static/js/126.6652f99f93c7eece9082.js 200 16.170 ms - 44168
2023-05-12 08:48:48.815 INFO APP: GET /static/css/app.6e9a0e56f46532006ef8.css 200 23.402 ms - 696282
2023-05-12 08:48:48.871 INFO APP: GET /static/js/app.7c25fc4ced5cca9e71dc.js 200 15.903 ms - 2204086
2023-05-12 08:48:51.632 INFO APP: GET /api/auth-enabled 200 28.710 ms - 29
2023-05-12 08:48:51.869 INFO APP: GET //static/fonts/MaterialIcons-Regular.2d80174.woff2 200 4.432 ms - 125116
2023-05-12 08:48:51.965 INFO APP: GET /static/logo.png 200 67.860 ms - 26171
2023-05-12 08:48:52.224 INFO APP: GET /api/settings 200 318.541 ms - 12910
2023-05-12 08:48:52.289 INFO APP: GET /api/auth-enabled 304 7.448 ms - -
2023-05-12 08:49:00.370 WARN Z-WAVE: Zwavejs driver is not ready yet, statistics will be enabled on driver initialization
2023-05-12 08:49:00.377 INFO APP: POST /api/statistics 200 53.394 ms - 89
2023-05-12 08:49:10.613 INFO Z-WAVE: Calling api getProvisioningEntries with args: [ [length]: 0 ]
2023-05-12 08:49:10.621 ERROR APP: Unhandled Rejection, reason: TypeError: Cannot read properties of undefined (reading 'isInBootloader')
2023-05-12 08:49:11.964 INFO APP: GET /api/settings 200 161.288 ms - 12956
2023-05-12 08:49:12.031 INFO APP: GET /api/auth-enabled 304 6.957 ms - -
2023-05-12 08:49:15.092 INFO Z-WAVE: Calling api _getScenes with args: [ [length]: 0 ]
2023-05-12 08:49:15.096 ERROR APP: Unhandled Rejection, reason: TypeError: Cannot read properties of undefined (reading 'isInBootloader')
2023-05-12 08:49:17.412 INFO APP: GET /api/store 200 12.376 ms - 584
2023-05-12 08:49:19.731 INFO Z-WAVE: Calling api refreshNeighbors with args: [ [length]: 0 ]
2023-05-12 08:49:19.734 ERROR APP: Unhandled Rejection, reason: TypeError: Cannot read properties of undefined (reading 'isInBootloader')
2023-05-12 08:50:21.327 INFO APP: GET /health/zwave 200 4.868 ms - 2140
```

ocalvo, 2023-05-12:
These are the stats for Node 20 (the motion sensor). After this I am trying to recover the network:
[image]

ocalvo, 2023-05-12:
I went back to the old ZwaveJS. I dont have the stomach to have the network going down.

AlCalzone, 2023-05-12:
You need to configure the same path and security keys in the addon's web ui under settings that  you have in the core addon config.
Then it should just be a matter of waiting until all devices are interviewed.

ocalvo, 2023-05-13:
I have completed the fast migration to ZwaveJS-UI, however initially the network was running super slow. Turning off/on a light takes up to 30 seconds.
It seems that after 10 minutes, it recover itself, I hope this is normal.
What do you need me todo in the ZwaveUI control panel for node 20?
[image]

AlCalzone, 2023-05-15:
Go to the network map, reload graph, click Node 20 there, click "check health" in the popup panel, keep the controller (node 1) as the target, start a health check, wake up the node so it actually starts.

ocalvo, 2023-05-15:
Node 20:
[image]

Check result:
(Running)

ocalvo, 2023-05-15:
Interestingly enough, after I did the check health test, the sensor stopped sending motion events.

AlCalzone, 2023-05-15:
The first few attempts aren't looking too reliable, the last few (I assume that's after the route change) are much better.
Not sure if the node is also using that route now, or if it still has a different one.
But if it still drops commands, it might not be using that route for unsolicited reports. You can try healing just that node now, hoping that the controller will assign it that good route.

Unfortunately we don't know if it doesn't send the commands or if they get lost. You'd have to use a Zniffer to figure that out.

---

Also side question: Is your Z-Wave stick on a USB extension?

ocalvo, 2023-05-15:
I have an usb extension and the dongle is in middle air:
[image]

I will try to restart zwave addon next time that the issue happens. 

I do have an additional zwave dongle, will install the sniffer to learn how to detect events coming from the sensor.

AlCalzone, 2023-05-16:
Aeotec Z-Stick 7 I see. EU or US? The EU version has connectivity issues.

Could be worth a try switching to the other stick (which one specifically?) temporarily - you can do that relatively easy using Z-Wave JS UI:

- Do an NVM backup of the current stick
- switch sticks, make sure the path is correct
- restore the NVM backup onto the other stick

If that doesn't help, you can just switch sticks again and everything should be back to the previous state.

ocalvo, 2023-05-16:
Thanks, I am only using US radios.

ocalvo, 2023-05-21:
After modifing the motion retirgger timeout from 30 seconds to 240 seconds the sensor has been working better, so far the issue has not repro.
However, after migration to ZWaveJS-UI I [started to see another issue where ZwaveJS fails completely in home-assistant](https://github.com/home-assistant/core/issues/93326)

[image]

closed 2023-06-08 by zwave-js-assistant[bot]