zwave-js/zwave-js #5708: Missing device configuration: <enter manufacturer and device name>

jewelgeronimo, 2023-04-25:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Kwikset Smart Lock ZWave

### Manufacturer ID

0x0090

### Product Type

0x0003

### Product ID

0x0042

### Firmware Version

4.10

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/2268

### Do you have a manual?

[Install_Guide_914C_03_09_2017.pdf](https://github.com/zwave-js/node-zwave-js/files/11325651/Install_Guide_914C_03_09_2017.pdf)

AlCalzone, 2023-05-22:
@zwave-js-bot import config 2268 from zwa

AlCalzone, 2023-05-23:
Ok, looks like this is an (older?) variant of the 914C [we already have](https://devices.zwave-js.io/?jumpTo=0x0090:0x0003:0x0446:0.0) under a different ID.

However, this one here:

- is missing config params 1-30
- has 2 association groups

AlCalzone, 2023-05-23:
@zwave-js-bot add fingerprint `0x0003:0x0042` to `0x0090:0x0003:0x0446:0.0`

closed 2023-05-23 by AlCalzone

jewelgeronimo, 2023-05-23:
This may be a dumb question, but is the next step to just exclude the re-run inclusion?

jewelgeronimo, 2023-05-27:
AMAZING. Just like that and poof! Magic. Thank you for the amazing work. Bless you.

PR #5834 (2023-05-23): fix(config): add fingerprint `0x0003:0x0042` to Kwikset 914C
fixes: #5708