zwave-js/zwave-js #5709: Adding Z-wave battery Vent, log reports " does NOT support CC Battery (0x80)"

riccol1966, 2023-04-26:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
  On interview or re-interview no battery attributes become available, interview and re-interview complete successfully.  
What do you observe?
  Device gets interviewed successfully but is missing battery level attribute, which did work when device was connected to a Veralite controller. Log initially says "  supported CCs:...    · Battery (0x80)" but further down in the log it says "querying the CC version for Battery... 2023-04-26T11:52:26.465Z CNTRLR   [Node 025]   does NOT support CC Battery (0x80) "
What did you expect to happen?
 To see battery command classes being identified so that I can monitor when battery runs low. This used to work when I used a veralite controller but am now using a Z00z 700 stick.  

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: Econet Controls
Model name: EV100-CM
Node ID in your network: 
24

Manufacturer Product Link: https://products.z-wavealliance.org/products/1044?selectedFrequencyId=-1

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 2023.4.6
`node-zwave-js` branch: 0.1.79
`zwave-js-ui` branch: 

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### If yes, where did it work?

I assume this has never worked in HA but battery info was reported when using the devices in Veralite. Using HA with a Zooz 700 stick updated to latest firmware. I have a
[zwave-interview-js-debug-log.txt](https://github.com/zwave-js/node-zwave-js/files/11332583/zwave-interview-js-debug-log.txt)
ttached debug log from z-wave JS  add-on. 

### Attach Driver Logfile

[core_zwave_js_2023-04-26T12-16-47.495Z.log](https://github.com/zwave-js/node-zwave-js/files/11332653/core_zwave_js_2023-04-26T12-16-47.495Z.log)

AlCalzone, 2023-04-26:
The logfile you attached is **not** on loglevel debug.

riccol1966, 2023-04-27:
sorry I was setting to debug and restarting but I guess that was not setting debug. So I stopped it manually and then set the debug level and restarted. Tried an inclusion and then a re-interview. Debug logs attached. 
[core_zwave_js_2023-04-27T09-41-00.440Z.log](https://github.com/zwave-js/node-zwave-js/files/11342021/core_zwave_js_2023-04-27T09-41-00.440Z.log)
[core_zwave_js_2023-04-27T09-38-40.373Z.log](https://github.com/zwave-js/node-zwave-js/files/11342024/core_zwave_js_2023-04-27T09-38-40.373Z.log)

AlCalzone, 2023-04-27:
I'm afraid you missed the most important part here:
https://www.home-assistant.io/integrations/zwave_js#how-to-access-the-z-wave-logs
> You need to keep this browser tab open for logging to be active.

The interview log just contains the last 2-3 messages.

riccol1966, 2023-04-27:
Ahh I see, that was a bit fiddly alright to get the logging to appear in the browser tab... I think this is it... I did an exclude, then include, then re-interview. 
[zwave_js (5).log](https://github.com/zwave-js/node-zwave-js/files/11342269/zwave_js.5.log)

AlCalzone, 2023-04-27:
including again wouldn't have been necessary, FYI

riccol1966, 2023-04-27:
Is there any solution to this do you think?

AlCalzone, 2023-04-27:
@zwave-js-bot add compat flag to `0x0157:0x0100:0x0100`
```jsonc
		// The device supports Battery CC, despite the Version query telling otherwise
		"commandClasses": {
			"add": {
				"0x80": {
					"isSupported": true,
					"version": 1
				}
			}
		}
```

riccol1966, 2023-04-27:
do I need to somehow fix this myself or does this get rolled out in some update.. sorry for the noobie questions...

AlCalzone, 2023-04-27:
will get fixed next release

PR #5712 (2023-04-27): fix(config): force-add Battery CC support for EcoNet Controls EV100
fixes: #5709

**TODO:**
- [ ]	Change PR title to be more specific
closed 2023-04-27 by AlCalzone