zwave-js/zwave-js #5725: Air temperature goes to 0 on several devices.

Miesje, 2023-05-02:


AlCalzone, 2023-05-02:
> the device is reporting 0

This.

AlCalzone, 2023-05-02:
While this is a device bug, we may be able to work around it by disabling the regular queries for the sensor values. I'll need to know the device fingerprint and firmware version to do that.

Miesje, 2023-05-02:
Dear AlCalzone,

This applies to al my devices of type  NAS-PD03Z:
[image]

Just to mention, this started from the 25 fifth of march, before there was no problem.
On the 25fith of March I updated the zwave to the latest version.
Now I have gone back to:
zwave-js-ui: 8.13.0.02c66b0
zwave-js: 10.13.0
home id: 4093854822
home hex: 0xf4034466
It did not solve the problem because I got one device already  produced a 0 value.
Maybe it was version v10.12.0 and 8.11.1 for UI

If you need more information please let me know.

Thanks in advance.
Miesje

Miesje, 2023-05-03:
I'm not a programmer but the device is at least 6 years old, for domotica that is an century.
Some devices have the problem less then the other.
The thing is that I don't now don't know because the device is old or that it is a bug.

AlCalzone, 2023-05-03:
Oh it's definitely a device bug, but it also seems that the Z-Wave certification wasn't as thorough back in the days.

PR #5739 (2023-05-03): fix(config): disable periodic refresh for Shenzhen Neo NAS-PD03Z
fixes: #5725
closed 2023-05-03 by zwave-js-bot