zwave-js/zwave-js #5736: Duewi 05437 - ZW ZS 3500 not pairing with Aeotec ZST10-700

Goldwing1973, 2023-05-03:


AlCalzone, 2023-05-03:
Please check that the RF region is set correctly. There were some issues where that magically changed after firmware updates:
[image]

If that doesn't help, I'm afraid you're out of options. The EU version of the Aeotec Z-Stick 7 has measurable RF performance issues - maybe the Z-Pi 7 (GPIO module) does too.

Goldwing1973, 2023-05-03:
RF Region was set to Europe.
And the performance issues, that could be, but why does a Fibaro Plug in the same wall socket pair without problems.
I even had battery powered motion sensors connected 15 meters away wich worked without problems.
Connecting the Deuwi to my  Z‐Stick Gen5 USB Controller works flawlessly.
Also, the Deuwi is the only device that won't connect (i have over 60 z-wave devices)

AlCalzone, 2023-05-03:
> Connecting the Deuwi to my Z‐Stick Gen5 USB Controller works flawlessly

The Gen5 definitely has a much better antenna.

>  why does a Fibaro Plug in the same wall socket pair without problems

Maybe the Fibaro has a better antenna than the Deuwi.

Goldwing1973, 2023-05-03:
`Just to be sure, did you exclude the Deuwi from the other network before trying to include it into the new one? `

yes, multiple times to be sure.
I’ll try to pair the Deuwi right next to the Z-Pi 7 to rule out the antenna and RF problem.

closed 2023-05-03 by Goldwing1973