zwave-js/zwave-js #5744: Zooz ZST39 Driver error - Failed to initialize the driver

reheat33, 2023-05-04:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

8.14.2.c234053

### ZwaveJS version

10.16.0

### Describe the bug

I'm using a Zooz zst39 stick and upon startup I receive the following error: Failed to initialize the driver: TypeError: Cannot read properties of undefined (reading 'valueDB').... (log file attached).

I am running via docker and passing the device through by id. I've tried various solutions like enable/disable soft reset and bootloader only. I've also tried unplugging the device several times. I consistently get this error. Any help would be appreciated. Thank you.

### To Reproduce

Start container with device installed and passed through by id.

### Expected behavior

Driver loading properly.

### Additional context

[logfile.txt](https://github.com/zwave-js/zwave-js-ui/files/11384763/logfile.txt)

robertsLando, 2023-05-04:
@reheat33 Just to know could you try to use `master` tag? It has 10.17.0 version that may fix it

AlCalzone, 2023-05-04:
Looks like this breaks because the stick does not report any node being part of the network - not even itself:
```
2023-05-03T15:46:51.733Z CNTRLR   received additional controller information:
                                    Z-Wave API version:         9 (legacy)
                                    Z-Wave chip type:           EFR32ZG23 / ZGM230S
                                    node type                   Controller
                                    controller role:            primary
                                    controller is the SIS:      true
                                    controller supports timers: false
                                    nodes in the network:       
```

PR #5745 (2023-05-04): fix: startup crash when controller returns empty list of nodes
fixes: #5744
closed 2023-05-04 by zwave-js-bot