zwave-js/zwave-js #5755: Thermostat showing Celsius instead of Fahrenheit

WizKid, 2023-05-05:
Starting a few months ago sometimes my thermostat randomly started showing temperature in Celsius even if it is configured to return Fahrenheit. I finally turned on Silly logging to see what is going on.

```
2023-05-04T13:25:41.658Z CNTRLR » [Node 042] Multilevel Sensor CC values may be stale, refreshing...
2023-05-04T13:25:41.659Z CNTRLR » [Node 042] querying Air temperature sensor reading...
2023-05-04T13:25:41.672Z SERIAL » 0x011700132a109f037700d3afa074d7fd97c82d80d8c025f097                (25 bytes)
2023-05-04T13:25:41.674Z DRIVER » [Node 042] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      240
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 119
                                    └─[MultilevelSensorCCGet]
                                        sensor type: Air temperature
                                        scale:       Celsius
2023-05-04T13:25:42.240Z SERIAL « [ACK]                                                                   (0x06)
2023-05-04T13:25:42.249Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2023-05-04T13:25:42.250Z SERIAL » [ACK]                                                                   (0x06)
2023-05-04T13:25:42.253Z DRIVER « [RES] [SendData]
                                    was sent: true
2023-05-04T13:25:42.276Z SERIAL « 0x01180013f000000200c97f7f7f7f0101030000000002010000cf              (26 bytes)
2023-05-04T13:25:42.276Z SERIAL » [ACK]                                                                   (0x06)
2023-05-04T13:25:42.277Z DRIVER « [REQ] [SendData]
                                    callback id:            240
                                    transmit status:        OK, took 20 ms
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    ACK RSSI:               -55 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1
2023-05-04T13:25:42.331Z SERIAL « 0x011a0004002a129f037b007792b963b8b71bfb3059df96bc79c900c2          (28 bytes)
2023-05-04T13:25:42.332Z SERIAL » [ACK]                                                                   (0x06)
2023-05-04T13:25:42.333Z CNTRLR   [Node 042] [translateValueEvent: metadata updated]
                                    commandClass: Multilevel Sensor
                                    endpoint:     0
                                    property:     Air temperature
                                    propertyKey:  undefined
                                    internal:     false
                                    secret:       false
                                    event source: undefined
2023-05-04T13:25:42.334Z CNTRLR   [Node 042] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
2023-05-04T13:25:42.334Z CNTRLR   [Node 042] [translateValueEvent: metadata updated]
                                    is root endpoint:        true
                                    is application CC:       true
                                    should hide root values: false
2023-05-04T13:25:42.335Z CNTRLR   [Node 042] [translateValueEvent: value updated]
                                    commandClass: Multilevel Sensor
                                    endpoint:     0
                                    property:     Air temperature
                                    propertyKey:  undefined
                                    internal:     false
                                    secret:       false
                                    event source: undefined
2023-05-04T13:25:42.335Z CNTRLR   [Node 042] [~] [Multilevel Sensor] Air temperature: 69 => 20.5    [Endpoint 0]
2023-05-04T13:25:42.335Z CNTRLR   [Node 042] [translateValueEvent: value updated]
                                    is root endpoint:        true
                                    is application CC:       true
                                    should hide root values: false
2023-05-04T13:25:42.336Z DRIVER « [Node 042] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 123
                                    └─[MultilevelSensorCCReport]
                                        type:  Air temperature
                                        scale: Celsius
                                        value: 20.5
2023-05-04T13:25:42.338Z CNTRLR « [Node 042] received current Air temperature sensor reading: 20.5 °C
```

It seems like when the value is stale and we ask for an updated value we ask for the value in Celsius. Then later when the temperature changes I get this:

```
2023-05-04T17:01:59.868Z SERIAL « 0x011a0004002a129f034f00a894baa493accbb0ad8553a19cc7c90024          (28 bytes)
2023-05-04T17:01:59.870Z SERIAL » [ACK]                                                                   (0x06)
2023-05-04T17:01:59.872Z CNTRLR   [Node 042] [translateValueEvent: metadata updated]
                                    commandClass: Multilevel Sensor
                                    endpoint:     0
                                    property:     Air temperature
                                    propertyKey:  undefined
                                    internal:     false
                                    secret:       false
                                    event source: undefined
2023-05-04T17:01:59.872Z CNTRLR   [Node 042] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
2023-05-04T17:01:59.873Z CNTRLR   [Node 042] [translateValueEvent: metadata updated]
                                    is root endpoint:        true
                                    is application CC:       true
                                    should hide root values: false
2023-05-04T17:01:59.874Z CNTRLR   [Node 042] [translateValueEvent: value updated]
                                    commandClass: Multilevel Sensor
                                    endpoint:     0
                                    property:     Air temperature
                                    propertyKey:  undefined
                                    internal:     false
                                    secret:       false
                                    event source: undefined
2023-05-04T17:01:59.874Z CNTRLR   [Node 042] [~] [Multilevel Sensor] Air temperature: 20.5 => 71    [Endpoint 0]
2023-05-04T17:01:59.875Z CNTRLR   [Node 042] [translateValueEvent: value updated]
                                    is root endpoint:        true
                                    is application CC:       true
                                    should hide root values: false
2023-05-04T17:01:59.876Z DRIVER « [Node 042] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 79
                                    └─[MultilevelSensorCCReport]
                                        type:  Air temperature
                                        scale: Fahrenheit
                                        value: 71
```

I don't have old logs so I can't say if the stale value detected is new or that it asks for Celsius is new.

Let me know if you need more logs.

WizKid, 2023-05-05:
Not sure if it is useful but here is the debug info for the device: [thermostat_device_info.txt](https://github.com/zwave-js/node-zwave-js/files/11403004/thermostat_device_info.txt)

WizKid, 2023-05-05:
@kpine: Ah, I'll do that and sees if that fixes things. But still feels like a bug. It was working for a really long. And I already configured the device to return Fahrenheit:
[image]

It also seems to be a huge amount of temperature sensor types. So to make sure I get Fahrenheit everywhere I need to do
[image]

AlCalzone, 2023-05-05:
Yeah, that does seem like it's not working as intended. The query should re-use the last scale the device reported with, unless you override it with preferred scales.

Couple of questions:
- Which driver version are you on? Make sure it's at least 10.13.0
- How often does the device report the temperature on its own?
- Does it have another multilevel sensor that is not reported?

WizKid, 2023-05-05:
I'm running:
zwave-js-ui: 8.13.1.33a6f57
zwave-js: 10.14.0

And last update before the stale poll was slightly over 6 hours before in both cases:
```
2023-05-04T07:14:03.325Z CNTRLR   [Node 042] [~] [Multilevel Sensor] Air temperature: 71 => 69      [Endpoint 0]
2023-05-04T07:15:23.323Z CNTRLR   [Node 042] [~] [Multilevel Sensor] Air temperature: 69 => 71      [Endpoint 0]
2023-05-04T07:18:03.332Z CNTRLR   [Node 042] [~] [Multilevel Sensor] Air temperature: 71 => 69      [Endpoint 0]
2023-05-04T07:22:03.323Z CNTRLR   [Node 042] [~] [Multilevel Sensor] Air temperature: 69 => 71      [Endpoint 0]
2023-05-04T07:23:03.326Z CNTRLR   [Node 042] [~] [Multilevel Sensor] Air temperature: 71 => 69      [Endpoint 0]
2023-05-04T13:25:42.335Z CNTRLR   [Node 042] [~] [Multilevel Sensor] Air temperature: 69 => 20.5    [Endpoint 0]
2023-05-04T17:01:59.874Z CNTRLR   [Node 042] [~] [Multilevel Sensor] Air temperature: 20.5 => 71    [Endpoint 0]
2023-05-04T23:10:40.304Z CNTRLR   [Node 042] [~] [Multilevel Sensor] Air temperature: 71 => 21.5    [Endpoint 0]
```

2023-05-04T13:25:42.335Z and 2023-05-04T23:10:40.304Z are the polls for stale values.

And no I don't have any other multilevel sensors that are not reported automatically. The difference is that my other sensors reports 0.1 degree differences so they updates much more often. It never goes 6 hours before an update for them. This Thermostat only report 1 degree differences so it very rarely updates.

WizKid, 2023-05-06:
Was looking into the code and I realized a strange thing. Neither https://github.com/zwave-js/node-zwave-js/blob/bbb914e958713b664f9ea1d186f54546a9bffaf9/packages/cc/src/cc/MultilevelSensorCC.ts#L133 nor https://github.com/zwave-js/node-zwave-js/blob/bbb914e958713b664f9ea1d186f54546a9bffaf9/packages/cc/src/cc/MultilevelSensorCC.ts#L142 is logged.

When it polls the thermostat for humidity I see: "No scale preference for sensor type 5, using the last-used scale 0".

So it must be something else that decides to use Celsius. I guess the easiest would be if I just build from Github source and add some more logging.

AlCalzone, 2023-05-06:
Do you still have the preferred scales active? In that case one of the earlier checks should determine the scale to use from that.

WizKid, 2023-05-06:
No I'm just looking at the logs when it happened originally. At that point I didn't know preferred scales existed.

After adding prefer scales I decided to remove them to see if the bug still happens. I also updated to 10.17.1 in case that does something.

AlCalzone, 2023-05-17:
@WizKid did you have any luck debugging? If not, would you mind making me a driver log for a re-interview on loglevel "debug"?
That should show some more information than the node diagnostics.

WizKid, 2023-05-26:
Finally had time to add some logging and debug it.

I hit this line: https://github.com/zwave-js/node-zwave-js/blob/master/packages/cc/src/cc/MultilevelSensorCC.ts#L122 and it say my preferred temperature scale is Celcius. It logs `scaleGroup preferred Celsius, temperature` when I log `scaleGroup preferred ${preferred}, ${scaleGroup.name}` But I don't understand how because in settings my preferred scales are empty:

[image]

WizKid, 2023-05-27:
Was trying to see where the Celsius comes from and found https://github.com/zwave-js/node-zwave-js/blob/72478787e9f75dd9be1529e606ceadab7acd117c/packages/zwave-js/src/lib/driver/Driver.ts#L254

Doesn't that mean we always use Celsius for temperature here https://github.com/zwave-js/node-zwave-js/blob/72478787e9f75dd9be1529e606ceadab7acd117c/packages/cc/src/cc/MultilevelSensorCC.ts#L122 . And we never do the fallback to previous value here https://github.com/zwave-js/node-zwave-js/blob/72478787e9f75dd9be1529e606ceadab7acd117c/packages/cc/src/cc/MultilevelSensorCC.ts#L125-L139 because preferred is already set.

AlCalzone, 2023-05-27:
Uhh, you're probably right

closed 2023-05-30 by zwave-js-bot

GaTechThomas, 2023-06-12:
When should we get these bits on our devices? Can we manually upgrade to the new version? Here's my version info:

Home Assistant 2023.6.1
Supervisor 2023.06.1
Operating System 10.2
Frontend 20230608.0 - latest

AlCalzone, 2023-06-12:
https://zwave-js.github.io/which-version/

That's driver version 10.23.0
The HA Core addon needs to be updated first according to that overview.