zwave-js/zwave-js #5757: [Bug]: Node-red crashes on a restart with an error that "z-wave driver" is not ready (see paste of logs from NOde-red below)

ReziioAdmin, 2023-05-05:
### What happened?

I have noticed this happening  on a few different Hubs in the past few weeks.  the Zwave interface becomes unresponsive and then Node-Red dives into a restart.  Most times it eventually recovers but I have one Hub right now that will just not get past the "uncaught error".

I am using a Zooz 700 Z-wave dongle.  One possibility is that the firmware is 7.17 and the latest version is 7.18.

### How to reproduce?

Please list steps to reproduce. If the problem is hardware-specific, please list which devices are affected.

### Version

8.0

### Node-RED Version

3.0.2

### What hardware are you seeing the problem on?

Other Linux

### Relevant Z-Wave-JS log as a text file

```
May 02 20:51:12 rosedale-apt-multi-N200 Node-RED[2746]: 2 May 20:51:12 - [error] ZWaveError: The controller is not yet ready! (ZW0103)
May 02 20:51:12 rosedale-apt-multi-N200 Node-RED[2746]:     at Driver.get controller [as controller] (/home/pi/.node-red/node_modules/zwave-js/src/lib/driver/Driver.ts:708:10)
May 02 20:51:12 rosedale-apt-multi-N200 Node-RED[2746]:     at Driver.shouldDiscardCC (/home/pi/.node-red/node_modules/zwave-js/src/lib/driver/Driver.ts:3607:21)
May 02 20:51:12 rosedale-apt-multi-N200 Node-RED[2746]:     at Driver.serialport_onData (/home/pi/.node-red/node_modules/zwave-js/src/lib/driver/Driver.ts:2895:14)
May 02 20:51:12 rosedale-apt-multi-N200 systemd[1]: nodered.service: Main process exited, code=exited, status=1/FAILURE
May 02 20:51:12 rosedale-apt-multi-N200 systemd[1]: nodered.service: Failed with result 'exit-code'.
```

closed 2023-05-05 by zwave-js-bot

ReziioAdmin, 2023-05-05:
@AlCalzone Thank you for this fix.  All is working again !

AlCalzone, 2023-05-05:
Huh, I didn't release it yet

ReziioAdmin, 2023-05-05:
Thanks Marcus.  I temporarily resolved the problem by removing the nodes which were calling the zwave driver immediately after a reboot or re-deploy which is probably a preferable design in any event but look forward to the release where the fix is released.

marcus-j-davies, 2023-05-05:
Makes perfect sense.

AlCalzone, 2023-05-06:
The problem was caused by the driver, not user code.