zwave-js/zwave-js #5765: TODO: Provide @raman325 device dumps

raman325, 2023-05-06:
1. Device dump with a config parameter that is a bitfield
2. Mock device dump for Energy Production CC

Thanks!

AlCalzone, 2023-05-11:
Energy Production CC:
[zwave_js-50da63f28934190600d0d68b43b2d834-Node 2-94ff446df95cec353669c8c8f7b88406.json.txt](https://github.com/zwave-js/node-zwave-js/files/11451605/zwave_js-50da63f28934190600d0d68b43b2d834-Node.2-94ff446df95cec353669c8c8f7b88406.json.txt)

AlCalzone, 2023-05-11:
Configuration CC with BitField:
```
Logger: aiohttp.server
Source: components/zwave_js/diagnostics.py:138
First occurred: 2:43:02 PM (2 occurrences)
Last logged: 2:43:15 PM

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py", line 433, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 85, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 235, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 146, in handle
    result = await result
             ^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/diagnostics/__init__.py", line 265, in get
    data = await info.device_diagnostics(hass, config_entry, device)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zwave_js/diagnostics.py", line 138, in async_get_device_diagnostics
    client: Client = hass.data[DOMAIN][config_entry.entry_id][DATA_CLIENT]
                     ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
KeyError: '50da63f28934190600d0d68b43b2d834'
```

AlCalzone, 2023-06-02:
I don't think 1. is necessary anymore. After reworking how bit fields are handled, the default behavior is good IMO:
[image]

closed 2023-06-02 by AlCalzone