zwave-js/zwave-js #5769: Unable to create Group Association

guiggess, 2023-05-08:
Problem/Motivation

>Tried to create a groupe association between a NodOn Soft Remote (FW: v2.3) and a Fibaro FGR223 (FW: v5.1).
Zwave hub is a Z‐Stick Gen5 USB Controller (FW: v1.1 SDK: v6.51.10).

Expected behavior

>When association is added to the NodOn Remote, it appears in the groups list and the device responds accordingly

Actual behavior

>When the association is created in the groups menu of the NodOn remote, it doesn't appear in the list and there is no response of the device when buttons are pressed. No "Target Endpoint" selection menu to be seen here.

Steps to reproduce

In NodOn remote:

>Configuration v1: Every setup on "Default" (also tried with "Scene Type" = Scene Activation)

In GROUPS Menu:

>Click "Add"
Select "Root Endpoint" and "Button 1"
Select Relevant "Target Node"
Click "Add"
Done

guiggess, 2023-05-27:
I tried today with the Fibaro keyfob, with the same result...
[zwavejs_2023-05-27.log](https://github.com/zwave-js/node-zwave-js/files/11582385/zwavejs_2023-05-27.log)

robertsLando, 2023-05-29:
cc @AlCalzone

AlCalzone, 2023-05-30:
I checked myself since I have two nodes with different security. All that happens when trying to add an impossible association is a warning in the application logs:
```
2023-05-30 15:08:18.635 WARN Z-WAVE: [Node 005] Unable to add Node 3 to Group 1 of Node 5
```
However, this problem isn't surfaced to the UI.

Currently when trying to add such an association, all one sees is this, which looks like success:
[image]

IMO the available options should be filtered instead of silently failing when adding the associations:
[image]