zwave-js/zwave-js #5776: Missing device configuration: <Yale / Yale Assure 2 Lock® SL Touchscreen Deadbolt>

tamorgen, 2023-05-08:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

YRD420-ZW3

### Manufacturer ID

0x0129

### Product Type

0x8104

### Product ID

0x45d2

### Firmware Version

2.17.0

tamorgen, 2023-05-08:
[YRC612 Manual.pdf](https://github.com/zwave-js/node-zwave-js/files/11425596/YRC612.Manual.pdf)

Added manual (Nov 2022) for the following deadbolts:

YRD216-ZW3, YRD226-ZW3, YRD256-ZW3,
YRC216-ZW3, YRC226-ZW3, YRC256-ZW3,
YRD652-ZW3, NF-YRD622-ZW3, NF-YRD612-ZW3,
YRC652-ZW3, NF-YRC622-ZW3, NF-YRC612-ZW3,
YRD622-ZW3, YRD642-ZW3, YRC622-ZW3, YRC642-ZW3,
YRD410-ZW3, YRD420-ZW3, YRD430-ZW3, YRD450-ZW3

AlCalzone, 2023-05-09:
Not obvious from the issue title, but this information is already posted/requested in https://github.com/zwave-js/node-zwave-js/issues/5650.

closed 2023-05-09 by AlCalzone