zwave-js/zwave-js #5786: Missing device configuration: 0x200 and neo cool cam

Touma02, 2023-05-10:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Cool cam NAD PD01

### Manufacturer ID

0x0258

### Product Type

0x0200

### Product ID

0x1036

### Firmware Version

3.3.0

AlCalzone, 2023-05-22:
This needs some more information to write the config file. At the very least a manual.

AlCalzone, 2023-09-27:
This does exist: https://devices.zwave-js.io/?jumpTo=0x0258:0x0200:0x1036:0.0

closed 2023-09-27 by AlCalzone