zwave-js/zwave-js #5799: Qubino flush dimmer 0-10V (ZMNHVD) included but do not want to use repeaters

Kratax, 2023-05-15:
Hi,
I have an issue with one ZMNHVD flush dimmer. Although alive and connected (included and seems to be working well), it is not willing to use any repeaters although should be available and better connection. On the map is shown as grey (named wallbox)... Healing does not help neither....
Any advice how to solve this?
Thanks a lot.
Regards,
Kratax

[image]

[image]

[image]

AlCalzone, 2023-05-15:
Another case of "don't trust that map" --> https://github.com/zwave-js/zwave-js-ui/issues/2517

In the 2nd screenshot you can see that it does route through nodes 34 and 39, but apparently they are not considered neighbors by the stick or device firmware 🤷‍♂️

closed 2023-06-02 by zwave-js-assistant[bot]

robertsLando, 2023-06-05:
Just for info, now with latest zwave-js-ui version you can set priority routes on nodes from network graph