zwave-js/zwave-js #5825: Fibaro covers doesn't report state after update to HA-Addon 1.13.1

memphi2, 2023-05-21:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

After updating to 1.13.1 my fibaro covers doesn't report the state.
I can see in the zwave log that the covers are sending the update, but in HA there is no change in the state.

What causes the bug?

Change in the state of a fibaro cover (open/close/ set to 50%)

no state change in HA

What did you expect to happen?

correct  state in HA

### Device information

Manufacturer:  fibaro
Model name:  FGR223 
Node ID in your network: 2-7

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 1.13.1
`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Updated from  1.13.0 ro  1.13.1

### Did this work before?

Yes (please describe)

### If yes, where did it work?

worked in 1.13.0

### Attach Driver Logfile

2023-05-21 14:30:53.407 INFO Z-WAVE: [Node 003] Value updated: 38-1-currentValue 81 => 99
2023-05-21 14:30:53.419 INFO Z-WAVE: [Node 003] Value updated: 38-1-targetValue 99 => 99

AlCalzone, 2023-05-22:
What the bot said :)

lucas-clemente, 2023-05-23:
I’m seeing the same issue, started a few days ago. I’ll try to get you a log file tonight.

Rob19284, 2023-05-23:
I'm having exact same problem with the fibaro covers since the last update.

AlCalzone, 2023-05-23:
Please don't comment without a logfile, that's just noise.

AlCalzone, 2023-05-23:
relevant log snippet:
```
2023-05-23T15:18:44.251Z DRIVER » [Node 004] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      11
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 36
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0x8d77687108e3641dd78a7c927bb3c8a3
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      0
                                      │ destination: 1
                                      └─[SupervisionCCGet]
                                        │ session id:      56
                                        │ request updates: true
                                        └─[MultilevelSwitchCCSet]
                                            target value: 56
2023-05-23T15:18:44.255Z SERIAL « [ACK]                                                                   (0x06)
2023-05-23T15:18:44.262Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2023-05-23T15:18:44.263Z SERIAL » [ACK]                                                                   (0x06)
2023-05-23T15:18:44.264Z DRIVER « [RES] [SendData]
                                    was sent: true
2023-05-23T15:18:44.286Z SERIAL « 0x011800130b00000200d27f7f7f7f00000300000000030100002e              (26 bytes)
2023-05-23T15:18:44.287Z SERIAL » [ACK]                                                                   (0x06)
2023-05-23T15:18:44.288Z DRIVER « [REQ] [SendData]
                                    callback id:            11
                                    transmit status:        OK, took 20 ms
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 100 kbit/s
                                    ACK RSSI:               -46 dBm
                                    ACK channel no.:        0
                                    TX channel no.:         0
2023-05-23T15:18:44.357Z SERIAL « 0x011d00040004159f03d400e3f9d2cae3fc3e0311da0443b3babbe6d1d00046    (31 bytes)
2023-05-23T15:18:44.362Z SERIAL » [ACK]                                                                   (0x06)
2023-05-23T15:18:44.364Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 212
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      1
                                      │ destination: 0
                                      └─[SupervisionCCReport]
                                          session id:          56
                                          more updates follow: true
                                          status:              Working
                                          duration:            25s
2023-05-23T15:18:47.098Z SERIAL « 0x0122000400041a9f03d500895d9f7037d4910233f5e718a302bf73b8691dcc047 (36 bytes)
                                  4d00031
2023-05-23T15:18:47.104Z SERIAL » [ACK]                                                                   (0x06)
2023-05-23T15:18:47.110Z CNTRLR   [Node 004] [Meter] value[66049]: metadata updated                 [Endpoint 1]
2023-05-23T15:18:47.113Z CNTRLR   [Node 004] [~] [Meter] value[66049]: 0 => 108.2                   [Endpoint 1]
2023-05-23T15:18:47.116Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 213
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      1
                                      │ destination: 0
                                      └─[MeterCCReport]
                                          type:       Electric
                                          scale:      W
                                          rate type:  Consumed
                                          value:      108.2
                                          time delta: 0 seconds
2023-05-23T15:18:59.085Z SERIAL « 0x011d00040004159f03d600d911f386a4b70c3682098d57ab97faa8d2d20016    (31 bytes)
2023-05-23T15:18:59.094Z SERIAL » [ACK]                                                                   (0x06)
2023-05-23T15:18:59.097Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 214
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      1
                                      │ destination: 0
                                      └─[SupervisionCCReport]
                                          session id:          56
                                          more updates follow: false
                                          status:              Success
                                          duration:            0s
```

no value update as result of the success.

thecobra666, 2023-05-24:
I have the same issue, since updating my docker container to the latest version, I can see the request showing up in the logs after a clicked a button or tried to switch on/off a device, but it does not report back to HA. 

Does not matter which device. I have it with fibaro, qubino, sensative..  
 
 
• Application Logs are disabled. Please enable it on "Settings > General" in order to see ZwaveJS logs

Filter logs
Type to filter logs, case sensitive
14:01:19.692 SERIAL » [ACK] (0x06)
14:01:19.693 DRIVER « [RES] [SendData]
was sent: true
14:01:19.712 SERIAL « 0x011800131900000200c77f7f7f7f000003000000000301000029 (26 bytes)
14:01:19.713 SERIAL » [ACK] (0x06)
14:01:19.715 DRIVER « [REQ] [SendData]
callback id: 25
transmit status: OK, took 20 ms
routing attempts: 1
protocol & route speed: Z-Wave, 100 kbit/s
ACK RSSI: -57 dBm
ACK channel no.: 0
TX channel no.: 0
14:01:19.724 CNTRLR [Node 013] [~] [Binary Switch] currentValue: false => false [Endpoint 0]
14:01:20.738 SERIAL » 0x010900130d029840251a0d (11 bytes)
14:01:20.740 DRIVER » [Node 013] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 26
└─[SecurityCCNonceGet]
14:01:20.743 SERIAL « [ACK] (0x06)
14:01:20.747 SERIAL « 0x0104011301e8 (6 bytes)
14:01:20.749 SERIAL » [ACK] (0x06)
14:01:20.750 DRIVER « [RES] [SendData]
was sent: true
14:01:20.767 SERIAL « 0x011800131a00000200c77f7f7f7f00000300000000030100002a (26 bytes)
14:01:20.769 SERIAL » [ACK] (0x06)
14:01:20.770 DRIVER « [REQ] [SendData]
callback id: 26
transmit status: OK, took 20 ms
routing attempts: 1
protocol & route speed: Z-Wave, 100 kbit/s
ACK RSSI: -57 dBm
ACK channel no.: 0
TX channel no.: 0
14:01:20.782 SERIAL « 0x01120004000d0a98801786998d612c01cec70036 (20 bytes)
14:01:20.784 SERIAL » [ACK] (0x06)
14:01:20.786 DRIVER « [Node 013] [REQ] [ApplicationCommand]
└─[SecurityCCNonceReport]
nonce: 0x1786998d612c01ce
14:01:20.794 SERIAL » 0x011d00130d1698815e57145de7f37e4f834c1e17d83c7a62769c989f251b7f (31 bytes)
14:01:20.796 DRIVER » [Node 013] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 27
└─[SecurityCCCommandEncapsulation]
│ nonce id: 23
└─[BinarySwitchCCGet]
14:01:20.799 SERIAL « [ACK] (0x06)
14:01:20.805 SERIAL « 0x0104011301e8 (6 bytes)
14:01:20.807 SERIAL » [ACK] (0x06)
14:01:20.808 DRIVER « [RES] [SendData]
was sent: true
14:01:20.827 SERIAL « 0x011800131b00000200c77f7f7f7f00000300000000030100002b (26 bytes)
14:01:20.830 SERIAL » [ACK] (0x06)
14:01:20.832 DRIVER « [REQ] [SendData]
callback id: 27
transmit status: OK, took 20 ms
routing attempts: 1
protocol & route speed: Z-Wave, 100 kbit/s
ACK RSSI: -57 dBm
ACK channel no.: 0
TX channel no.: 0
[... 911 lines cut ...]
14:16:31.348 DRIVER « [RES] [SendData]
was sent: true
14:16:31.367 SERIAL « 0x011800132700000200cd7f7f7f7f01010300000000020100001c (26 bytes)
14:16:31.369 SERIAL » [ACK] (0x06)
14:16:31.370 DRIVER « [REQ] [SendData]
callback id: 39
transmit status: OK, took 20 ms
routing attempts: 1
protocol & route speed: Z-Wave, 40 kbit/s
ACK RSSI: -51 dBm
ACK channel no.: 1
TX channel no.: 1
14:16:31.381 CNTRLR [Node 023] [~] [Binary Switch] currentValue: false => false [Endpoint 0]
14:16:31.467 SERIAL « 0x010b0004001703250300cd000f (13 bytes)
14:16:31.469 SERIAL » [ACK] (0x06)
14:16:31.471 CNTRLR [Node 023] [~] [Binary Switch] currentValue: false => false [Endpoint 0]
14:16:31.473 CNTRLR [Node 023] Scheduled poll canceled because expected value was received
14:16:31.474 DRIVER « [Node 023] [REQ] [ApplicationCommand]
└─[BinarySwitchCCReport]
current value: false
14:16:32.160 SERIAL » 0x010900131c0225022528d1 (11 bytes)
Made with ❤️ by
Daniel Lando
- Enjoying it? 
[Support me 🙏](https://github.com/sponsors/robertsLando)

AlCalzone, 2023-05-24:
I know what the issue is. working on it.

thecobra666, 2023-05-24:
> I know what the issue is. working on it.

Great to hear that. What is the last working release? I'm already down to 8.15.0, still does not work or is not related to zwave-js-ui?

PR #5840 (2023-05-24): fix: bind `SET_VALUE` and `POLL_VALUE` to the correctly proxied API instance
Well, this was fun. As it turns out, the arrow functions we used for the API-specific SET_VALUE and POLL_VALUE implementations made it impossible to pass in `commandOptions`.

A huge refactor later, this is now possible.

fixes: #5825
closed 2023-05-24 by zwave-js-bot