zwave-js/zwave-js #5826: Changed configuration parameters for MCO-Home micro switch MH-S220

dxfoxd, 2023-05-22:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

MCO-Home MH-S220

### Manufacturer ID

0x015f

### Product Type

0x2201

### Product ID

0x1252

### Firmware Version

3.2

### Is the device listed on the Z-Wave Alliance website?

https://opensmarthouse.org/zwavedatabase/1153/reference/MH-S220-2021-v3.2.pdf

### Do you have a manual?

The configuration parameters of this device are changed from fw 3.2. I have created (and tested) a working config file for this fw and updated/renamed the existing device config.
[mh-s220.zip](https://github.com/zwave-js/node-zwave-js/files/11528688/mh-s220.zip)

AlCalzone, 2023-05-22:
Can you open a pull request with the changes? Makes it easier to see what changed and make sure it's correct.

dxfoxd, 2023-05-22:
I tried in my first instance, but I got stuck in "Comparing changes". It's my first PR and no idea what to select here.

AlCalzone, 2023-05-22:
Does this help?
https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork
base repository/branch is node-zwave-js/zwave-js, master
head repository/branch is your repo, your branch