zwave-js/zwave-js #5846: Missing device configuration: Qubino Shades Remote Controller, ZMNKGD1

KattenJansson1974, 2023-05-26:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Qubino Shades Remote Controller, ZMNKGD1

### Manufacturer ID

0x0159

### Product Type

0x0300

### Product ID

Cannot be found

### Is the device listed on the Z-Wave Alliance website?

No

### Do you have a manual?

[Qubino_Shades_Remote.pdf](https://github.com/zwave-js/node-zwave-js/files/11578179/Qubino_Shades_Remote.pdf)

AlCalzone, 2023-05-26:
The product ID should either be shown in Z-Wave JS UI when you open the dropdown for the node, or in a diagnostics dump in Home Assistant.

KattenJansson1974, 2023-05-30:
Updated with correct Manufacturer ID, Product Type and Product ID

PR #6335 (2023-10-11): feat(config): add Qubino Shades Remote Controller
This should close #5846 . There are no parameters and no association groups.

PR description here
[Qubino_Shades_Remote.pdf](https://github.com/zwave-js/node-zwave-js/files/12750157/Qubino_Shades_Remote.pdf)

memo: How this device appears to work 1) Includes in network (gets ID) 2) Includes Qubino blinds during operation.  Probably picks up zwave transmissions. 3) operate blinds with remote. Unclear how many blinds it can control.
closed 2023-10-11 by AlCalzone