zwave-js/zwave-js #5849: Missing device configuration: PM-B400ZW-N

techinpark, 2023-05-27:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

PM-B400ZW-N

### Manufacturer ID

396

### Product Type

66

### Product ID

0x5

### Firmware Version

78.40

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/1870

### Do you have a manual?

[image]

AlCalzone, 2023-06-02:
@zwave-js-bot import config 1870 from ZWA

AlCalzone, 2023-06-02:
@zwave-js-bot import config 1870 from zwa

PR #6421 (2023-10-24): feat(config): add PM-B400ZW-N
First extract attempt from ZWA to close #5849 (help wanted issue). Tried to pick up the bot import #5871, but had unexpected message (separate repository?), so started from fresh. Only real area of concern for me is param 1. It appears from the manual that once a value over 0 is added, the device enables the 4th byte (0x01). I do not see how to do that separately with masking. It seems the function of the parameter is to turn off the switch if the value falls below the setting value. Other than that, I used the 'label' to explain the enable_disable for params 3 &  [... cut]
closed 2023-10-24 by AlCalzone