zwave-js/zwave-js #5851: Missing device configuration: sunricher SR-ZV9101SAC-HP

MarcHagen, 2023-05-27:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

sunricher SR-ZV9101SAC-HP

### Manufacturer ID

0x0330

### Product Type

0x0200

### Product ID

0xd00c

### Firmware Version

1.26

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/2918

### Do you have a manual?

https://www.sunricher.com/media/resources/manual/SR-ZV9101SAC-HP-V3-Instruction.pdf

AlCalzone, 2023-06-02:
@zwave-js-bot import config 2918 from zwa

AlCalzone, 2023-06-02:
We have an (older?) variant of this device, which the linked alliance DB entry refers to (product ID 0xd005).
The manual seems to have 3 new config parameters.

Let's stay conservative and:
- copy https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0330/zv-9101.json into a new file
- change the product ID
- add the 3 new config parameters