zwave-js/zwave-js #5855: Multiple double tap events firing?

lkroeker, 2023-05-29:
Problem:
After a recent update (to home assistant), when double pressing my Jasco ZWave Dimmers the system registers four 'double presses' a few milliseconds apart. These events show up identically other than the unique id. This wouldn't be an issue, except that I use multiple double presses to trigger different actions in my system.

Expected behavior:
Ideally there would only be one event sent per double press.

Actual behavior:
Four events appear of the type: event_type: zwave_js_value_notification

Steps to reproduce:
Bind device to double tap on the controller. Double tap light switch.

AlCalzone, 2023-05-30:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) of the issue, loglevel `debug` and attach it here as a file (drag & drop into the text field).

lkroeker, 2023-06-01:
[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/11619665/zwave-js-ui-store.zip)
[node_51.zip](https://github.com/zwave-js/node-zwave-js/files/11619667/node_51.zip)

Here's what I see in HA:
[HA Logs - zwave_js_value_notificaitons.txt](https://github.com/zwave-js/node-zwave-js/files/11619672/HA.Logs.-.zwave_js_value_notificaitons.txt)

AlCalzone, 2023-06-01:
I see one instance where the event is repeated 4x, but this is because the corresponding command is received 4x from the device:
```
2023-06-01T13:19:44.981Z SERIAL « 0x010900040033032001ff1c                                            (11 bytes)
2023-06-01T13:19:44.988Z SERIAL » [ACK]                                                                   (0x06)
2023-06-01T13:19:44.989Z SERIAL « 0x010900040033032001ff1c                                            (11 bytes)
2023-06-01T13:19:44.990Z SERIAL » [ACK]                                                                   (0x06)
2023-06-01T13:19:44.991Z SERIAL « 0x010900040033032001ff1c                                            (11 bytes)
2023-06-01T13:19:44.995Z SERIAL » [ACK]                                                                   (0x06)
2023-06-01T13:19:44.995Z SERIAL « 0x010900040033032001ff1c                                            (11 bytes)
2023-06-01T13:19:44.997Z SERIAL » [ACK]                                                                   (0x06)
```

This is unrelated to host software updates and typically caused by physical RF connectivity issues where the sending device doesn't get the target's acknowledgement in time and starts re-transmitting.

Here are some links you can work through to try and fix the connectivity issues:
- https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting
- https://zwave-js.github.io/node-zwave-js/#/troubleshooting/network-health?id=testing-the-connection-strength

lkroeker, 2023-06-10:
I did a full heal of my network, and it seems to be working normally now.  It was odd that it was so consistently sending multiple commands, and from multiple switches.  But it seems to be working normally again, I assume one of the key routing nodes was not behaving correctly.  I'm going to close this as it appears to be related to the zwave mesh or interference like you suggested.   Thanks for you help!

closed 2023-06-10 by lkroeker

AlCalzone, 2023-06-10:
It doesn't mean that multiple commands are sent but that the same command is in flight multiple times due to re-transmission by the nodes along the route. This happens because the nodes don't notice that their commands were received.