zwave-js/zwave-js #5864: Improve handling of unknown (254) values

AlCalzone, 2023-05-30:
In reference to https://github.com/zwave-js/node-zwave-js/discussions/4804

The current solution of using the string `"unknown"` or `undefined` for numbers is not ideal (one is not a number, the other does not get emitted in events). On the other hand, the signal value 254 lies outside of the min/max range which applications need to be able to rely on.

The preferred solution is using two different signal values:
- `null` for (known to be) unknown --> 254
- `undefined` for not (yet) known --> not queried/no response

sanderlv, 2026-08-30:
What's the point of these unknown entities that spoil my entities database?

[image]

[image]

They suddenly just "appeared" on many of my devices... can't they be just left way if it's nothing (as before)?

AlCalzone, 2026-08-30:
Please open a new issue and share the device diagnostics