zwave-js/zwave-js #5888: Bootstrapping node on behalf of inclusion controller does nothing - possibly HA/Server issue

AlCalzone, 2023-06-05:
```
15:05:01.398 CNTRLR   Node 4 was included by another controller:
                      basic device class:    Routing Slave
                      generic device class:  Binary Switch
                      specific device class: Unused
                      supported CCs: 
                        · Z-Wave Plus Info (0x5e)
                        · Security 2 (0x9f)
                        · Security (0x98)
                        · Supervision (0x6c)
                        · Transport Service (0x55)
                        · Binary Switch (0x25)
15:05:01.398 CNTRLR   [Node 004] Waiting for initiate command to bootstrap node...
15:05:11.401 CNTRLR   [Node 004] no initiate command received, bootstrapping node...
15:05:11.401 CNTRLR » [Node 004] Assigning SUC return route...
15:05:11.406 SERIAL » 0x0105005104c36c                                                     (7 bytes)
15:05:11.406 DRIVER » [Node 004] [REQ] [AssignSUCReturnRoute]
                        payload: 0x04c3
15:05:11.408 SERIAL « [ACK]                                                                   (0x06)
15:05:11.417 SERIAL « 0x0104015101aa                                                       (6 bytes)
15:05:11.418 SERIAL » [ACK]                                                                   (0x06)
15:05:11.419 DRIVER « [RES] [AssignSUCReturnRoute]
                        was executed: true
15:05:11.468 SERIAL « 0x01210051c30000000000000000000000000000000000000000000000000000000 (35 bytes)
                      0004c
15:05:11.468 SERIAL » [ACK]                                                                   (0x06)
15:05:11.468 DRIVER « [REQ] [AssignSUCReturnRoute]
                        callback id:     195
                        transmit status: OK
15:05:11.471 SERIAL » 0x010e00a90104029f042500000000c425                                  (16 bytes)
15:05:11.471 DRIVER » [Node 004] [REQ] [SendDataBridge]
                      │ source node id:   1
                      │ transmit options: 0x25
                      │ callback id:      196
                      └─[Security2CCKEXGet]
15:05:11.474 SERIAL « [ACK]                                                                   (0x06)
15:05:11.475 SERIAL « 0x010401a90152                                                       (6 bytes)
15:05:11.475 SERIAL » [ACK]                                                                   (0x06)
15:05:11.476 DRIVER « [RES] [SendDataBridge]
                        was sent: true
15:05:11.488 SERIAL « 0x011d00a9c400000100ba7f7f7f7f00000300000000030100007f7f7f7f7f4a    (31 bytes)
15:05:11.489 SERIAL » [ACK]                                                                   (0x06)
15:05:11.489 DRIVER « [REQ] [SendDataBridge]
                        callback id:            196
                        transmit status:        OK, took 10 ms
                        routing attempts:       1
                        protocol & route speed: Z-Wave, 100 kbit/s
                        ACK RSSI:               -70 dBm
                        ACK channel no.:        0
                        TX channel no.:         0
15:05:11.495 SERIAL « 0x010f00a8000104069f050002018300bafb                                (17 bytes)
15:05:11.495 SERIAL » [ACK]                                                                   (0x06)
15:05:11.495 DRIVER « [Node 004] [REQ] [BridgeApplicationCommand]
                      │ RSSI: -70 dBm
                      └─[Security2CCKEXReport]
                          echo:                       false
                          supported schemes:
                          · KEXScheme1
                          supported ECDH profiles:
                          · Curve25519
                          CSA requested:              false
                          requested security classes:
                          · S2_Unauthenticated
                          · S2_Authenticated
                          · S0_Legacy
```
can be tested with the `RT_SUCAndSISFunctionality_Rev01` certification test

raman325, 2023-06-06:
I think there's an inherent assumption in the design that inclusion is initiated by the primary controller. Currently the only way for an application to process the `grant security classes` or `validate dsk and enter pin` steps is by providing callbacks as part of calling `Controller.beginInclusion` right? I see in the docs that these callbacks can be specified in the driver options, maybe we need to switch to using that in the server so that there is always a callback available to the controller/driver?

AlCalzone, 2023-06-06:
> I see in the docs that these callbacks can be specified in the driver options, maybe we need to switch to using that in the server so that there is always a callback available to the controller/driver?

Yep, I added that setting while working on support for Inclusion controllers for this exact reason.

raman325, 2023-06-06:
> > I see in the docs that these callbacks can be specified in the driver options, maybe we need to switch to using that in the server so that there is always a callback available to the controller/driver?
> 
> Yep, I added that setting while working on support for Inclusion controllers for this exact reason. The problem here was though that only one application can provide those, so it is probably Z-Wave JS UI when it is the one hosting the server.

can we make it optionally a list that the driver loops through? And maybe add a method to add callbacks outside of options? We can then add our callbacks immediately after we start the server which will work in both the embedded and standalone context

AlCalzone, 2023-06-06:
The only way I could imagine a list working would be to call them all at once and wait until the first one resolves, but then the others won't get notified that the callback already resolved.

AlCalzone, 2023-06-14:
Just checked. You can already change the `inclusionUserCallbacks` using `driver.updateOptions(...)`. Only one set of callbacks can be active at any given time, but that should be okay.
Maybe when using the core addon, HA should handle the callbacks. And if using Z-UI, Z-UI should handle them.

AlCalzone, 2023-06-14:
Also I confirmed with Z-UI which also gets this wrong. The driver calls the callbacks, but they are never handled.

raman325, 2023-06-16:
> Maybe when using the core addon, HA should handle the callbacks. And if using Z-UI, Z-UI should handle them.

Before we do this, is there any alternative that allows both applications to be able to handle this in parallel? It feels like a bad user experience to have things work differently depending on the addon you are using.

AlCalzone, 2023-06-20:
Thought about this a bit more. Pretty sure Z-UI can only handle such a callback if the UI is visible, I guess the same would be true for HA.
Maybe this should be solved on both sides similarly to how HA does driver logs:
"subscribe" by configuring the callbacks when opening the relevant UI, "unsubscribe" by removing them again when leaving it.
/cc @robertsLando

robertsLando, 2023-06-20:
What if there are 2 users logged then?

AlCalzone, 2025-05-09:
This has been fixed

closed 2025-05-09 by AlCalzone