zwave-js/zwave-js #5900: GE/Jasco 14942 Device File Error - Incorrect Value Size

Nick2253, 2023-06-07:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

The device file for GE 14294 dimmer switches appears to have the incorrect "valueSize" parameter for the three dimming time parameters.

The [14294 device file](https://github.com/zwave-js/node-zwave-js/blob/1ada8fce8751f3cefceacb05766d30b08aff7a94/packages/config/config/devices/0x0063/14294_zw3005.json#L55-L56) refers to the [Jasco template file](https://github.com/zwave-js/node-zwave-js/blob/1ada8fce8751f3cefceacb05766d30b08aff7a94/packages/config/config/devices/0x0063/templates/jasco_template.json#L67-L69), which refers to the [master template file "dimming_timing"](https://github.com/zwave-js/node-zwave-js/blob/1ada8fce8751f3cefceacb05766d30b08aff7a94/packages/config/config/devices/templates/master_template.json#L222-L229).  

"dimming_timing" has a `valueSize` parameter of 1, but the 14294 switch requires a `valueSize` of 2 in order to successfully set the parameter.

This was discussed and confirmed in a [thread on the Home Assistant Forum](https://community.home-assistant.io/t/zwave-js-parameters-not-being-set-for-switch/578813/5).  When manually setting the parameter through the Z-Wave JS UI with the 2-byte size, the 14294 switch accepted the parameter updates.  With the specified 1-byte size, it did not.

The attached log shows an attempt at changing parameter 12 "Dimmer Rate (All-on/All-off)" to value "1".  The logs show transmit is "OK", but when it queries the devices afterwards, the device reports back the previous value of "3".

### Device information

Manufacturer: GE/Jasco
Model name: 14294 / ZW3005

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Zwave JS: 1.28
Driver: 10.22.3

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### Attach Driver Logfile

```
Log Level changed to: Debug
2023-06-07T21:56:19.332Z SERIAL » 0x010c00130c0570040c010125fe4a                                      (14 bytes)
2023-06-07T21:56:19.338Z DRIVER » [Node 012] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      254
                                  └─[ConfigurationCCSet]
                                      parameter #:      12
                                      reset to default: false
                                      value size:       1
                                      value format:     UnsignedInteger
                                      value:            1
2023-06-07T21:56:19.345Z SERIAL « [ACK]                                                                   (0x06)
2023-06-07T21:56:19.350Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2023-06-07T21:56:19.351Z SERIAL » [ACK]                                                                   (0x06)
2023-06-07T21:56:19.356Z DRIVER « [RES] [SendData]
                                    was sent: true
2023-06-07T21:56:19.364Z SERIAL « 0x01070013fe00000217                                                 (9 bytes)
2023-06-07T21:56:19.365Z SERIAL » [ACK]                                                                   (0x06)
2023-06-07T21:56:19.367Z DRIVER « [REQ] [SendData]
                                    callback id:     254
                                    transmit status: OK
2023-06-07T21:56:20.389Z SERIAL » 0x010a00130c0370050c25ff4a                                          (12 bytes)
2023-06-07T21:56:20.392Z DRIVER » [Node 012] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      255
                                  └─[ConfigurationCCGet]
                                      parameter #: 12
2023-06-07T21:56:20.397Z SERIAL « [ACK]                                                                   (0x06)
2023-06-07T21:56:20.402Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2023-06-07T21:56:20.403Z SERIAL » [ACK]                                                                   (0x06)
2023-06-07T21:56:20.403Z DRIVER « [RES] [SendData]
                                    was sent: true
2023-06-07T21:56:20.418Z SERIAL « 0x01070013ff00000216                                                 (9 bytes)
2023-06-07T21:56:20.419Z SERIAL » [ACK]                                                                   (0x06)
2023-06-07T21:56:20.419Z DRIVER « [REQ] [SendData]
                                    callback id:     255
                                    transmit status: OK
2023-06-07T21:56:20.430Z SERIAL « 0x010c0004000c0670060c02000386                                      (14 bytes)
2023-06-07T21:56:20.432Z SERIAL » [ACK]                                                                   (0x06)
2023-06-07T21:56:20.442Z CNTRLR   [Node 012] [~] [Configuration] 12: 1 => 3
2023-06-07T21:56:20.449Z DRIVER « [Node 012] [REQ] [ApplicationCommand]
                                  └─[ConfigurationCCReport]
                                      parameter #: 12
                                      value size:  2
                                      value:       3
```

apella12, 2023-10-12:
Did some investigation into this. There are two GE/Jasco devices 14294 with identical numbers (Type:Id) (0x4944:0x3038).  One version has 1 byte parameters 8, 10, 12 and the other 2 byte.
1 Byte) https://products.z-wavealliance.org/products/1442
2 Byte) https://[products.z-wavealliance.org/products/2105](https://products.z-wavealliance.org/products/2105)

Sometimes the version can be used to separate devices.  What version do you have? It would under the version tab in Zwave-js-ui.  The concern would be changing it globally might cause problems with the 1 byte owners. Also since there is a workaround with the custom configuration, maybe this isn't worth the effort?

FYI-I also found a couple of other GE/Jasco dimmers with two byte parameters 8, 10, 12. Because of the nature of the parameter  (not really changed much) there are others device owners that would have a problem changing the parameter if they tried.  AFAICT all the GE/Jasco devices in Zwave-JS are setup with 1 byte.

AlCalzone, 2023-10-12:
@JascoProducts can you shed some light onto this?

Nick2253, 2023-10-19:
As far as I can tell, the only difference between those two devices is that 2105 is the updated firmware (v5.26).  Maybe I'm missing something (especially since I don't understand why they'd have to get recertified for a firmware update), but I think it's just the firmware change.  If you check for ZW3010 device, you'd also see two different certifications for two different firmware versions.

If I'm correct, then the only thing we'd need to check on is the firmware version.

AlCalzone, 2023-10-19:
That sounds likely