zwave-js/zwave-js #5943: Button doesn't work randomly

dxfoxd, 2023-06-26:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

_What causes the bug?_
Pressing the button twice
_What do you observe?_
The button does not provide the outcome as expected (starting an automation)
_What did you expect to happen?_
Start the automation
_Steps to reproduce the behavior:_
Pressing the button twice > nothing happens
Pressing the button twice again > nothing happens
Pressing the button twice again > behavior as expected.
Sometimes the button behavior is as expected after the second pressing sequence, or multiple instances, or not at all. The button also works instantly in some occasions.

The log shows for the first two instances of the node (Node 109) "[Node 109] in the process of replying to a NonceGet, won't send another NonceReport", but also "Dropping message with invalid payload". I don't know if the last error is referring to a message of node 109.

The node is listed as a S0 legacy device as advised to prevent assigning multiple non existing capabilities to the device. Attached is a node debug file 
[node_109_debug.txt](https://github.com/zwave-js/node-zwave-js/files/11868819/node_109_debug.txt)

### Device information

Manufacturer: FibarGroup
Model name: Push Button Central Scene Controller
Node ID in your network:  109

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwave-js-ui: 8.16.1.b06c3ff
zwave-js: 10.20.0

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Sometimes the button works as expected.

### Attach Driver Logfile

[zwave-js_log_node109.txt](https://github.com/zwave-js/node-zwave-js/files/11869106/zwave-js_log_node109.txt)

dxfoxd, 2023-06-26:
Thx!

* The advices from the  first link were performed already several weeks ago due to assigning of non existing device capabilities and poor network performance. This brought some improvements.
* Both devices (button and the controller) are in the same room with (at the time of the healing process) a distance of 30cm. I  already tried several healings of the device, but strange enough and even after I removed all device associations the healing did not lead to a better connection. A hub device in the second next room (hallway on a lower floor) is selected (appr. 9m distance with 3 walls between them.
[image]

* Several attempts to perform a health check didn’t work either. The app keeps running without any results of with results, but keeps running (even I limited them to 5), see screenshot.
[image]

Strange enough, the controller is using a distant device as a hub to route all messages. You should expect that the controller will accept multiple direct connections, especially where multiple devices are in the same room at a and within a shorter distance. For example, the controller and WS-5 are within a 30cm distance. WS-5 is the same device as WS-3, but between the controller and WS-3 there is a distance of 3m including a 40cm thick brick wall.

How can I ensure that the controller accepts multiple devices as a direct connection (without the obvious heal after taking off the power of the device)?

dxfoxd, 2023-06-26:
Thx for the tip, I just upgraded and I'm impressed of the new network map and prio route feature.

After upgrading tested and worked in first instance, but the next attempt didn't work.
Errors
- in the process of replying to a NonceGet, won't send another NonceReport
- Dropping message with invalid payload
- Nonce 0xac expired, cannot decode security encapsulated command

Does this mean that I need to consider more time between 2 button press events?

Attached is the log from after restarting the latest version.
[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/11873872/zwavejs_current.log)

dxfoxd, 2023-06-26:
Additionally, I did run a diagnose for another device (WS-5 30cm of the controller) and this provided errors from the controller to the device. Is it true that not all devices support the diagnose feature when receiving from the controller?
[image]
If so, how should I interpret a 0/10 rating (meaning 0 out of 10, or as in rating to/from)?

AlCalzone, 2023-06-27:
> Is it true that not all devices support the diagnose feature when receiving from the controller?

Not all devices support all health-check related features, but it looks like yours does. This one seems to have problems reaching the controller though, or it is not receiving the acknowledgements. I'm guessing the latter is true - the commands reach the controller but the ACKs don't find their way back to the devices.

0/10 rating means 0 out of 10, so the worst possible score. This is because none of the pings from device to controller are successful, which means this direction of communication seems to be broken.

> Attached is the log from after restarting the latest version

Is node 109 also in close proximity to the controller? At least the first couple of communication attempts seem fine (direct connection at 100 kbit/s), then later it swaps back to a 2-hop route at 40k.

Other nodes have trouble reaching the controller too, e.g. Node 028 tries around 40 times (although there are probably repeaters involved) to get a report to the controller. This kind of traffic severely impacts the reliability of the network for other devices.

---

Considering all of this, I'm making a guess here:
You're in EU and using an Aeotec Z-Stick 7? If that's the case, I strongly suggest switching to a different stick, preferably 500 series. I don't know what it is with 700 series sticks and the EU frequency, but this combination has terrible RF performance.

The Aeotec Gen5+ stick is a good choice and [can be updated](https://aeotec.freshdesk.com/support/solutions/articles/6000252294-z-stick-gen5-v1-02-firmware-update) to support most features like SmartStart.
Using Z-Wave JS UI, you can easily swap. Just do an NVM backup on the 700 series, and restore that onto the 500 series (needs the aforementioned firmware upgrade).

robertsLando, 2023-06-27:
> @robertsLando any clue why that happens? 5 rounds configured, currently stuck at 9?!

IMO that was an old issue that should be fixed now

AlCalzone, 2023-06-27:
Yeah, looks like it works now.

dxfoxd, 2023-07-07:
> Not all devices support all health-check related features, but it looks like yours does. This one seems to have problems reaching the controller though, or it is not receiving the acknowledgements. [...]

I received the Aeotec GEN 5+ stick and installed it. I will do more tuning of the network and the devices and see if the performance gets better. One thing though, I want to disable the LED (parameter 81) but there is no parameter setting in the UI. Using the zwave_js.set_config_parameter service in HA won't work either because the controller has no device_id or entity_id. Is there a code snippet to update the parameter in a js function?

AlCalzone, 2023-07-10:
Or a driver function, as documented [here](https://zwave-js.github.io/node-zwave-js/#/usage/custom?id=send-a-custom-message-to-the-controller).

dxfoxd, 2023-07-10:
Thx. The command line worked. For completeness, toggling the led on again was succesful with 
`sudo echo -e -n "\x01\x08\x00\xF2\x51\x01\x01\x05\x01\x50" > /dev/[USB device] `
where [USB device] is the device used (mine is referred to as `/dev/serial/by-id/usb-0658_0200-if00`)
I also tried to use the code into the custom driver function, but this lead to an error ("Message not defined")

closed 2023-07-29 by zwave-js-assistant[bot]