zwave-js/zwave-js #5947: Zwave controller stops responding

ocalvo, 2023-06-26:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

What do you observe?

What did you expect to happen?
Network to automatically reset controller. Today a maual restart of the addon is required.

Steps to reproduce the behavior:
  1. After 24-48 hours, the controller stops responding.

### Device information

Manufacturer: Aeotec 
Model name: USB 7
Node ID in your network: TBD

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

```
2023-06-26 15:10:01.303 INFO Z-WAVE: [Node 057] Value updated: 113-0-alarmLevel 0 => 0
2023-06-26 15:10:01.310 INFO Z-WAVE: [Node 057] Value updated: 113-0-Home Security-Motion sensor status 8 => 0
2023-06-26 15:10:01.436 INFO Z-WAVE: [Node 057] Metadata updated: 48-0-Motion
2023-06-26 15:10:01.440 INFO Z-WAVE: [Node 057] Value updated: 48-0-Motion true => false
2023-06-26 15:10:05.763 INFO Z-WAVE: [Node 126] Value updated: 113-0-alarmType 0 => 0
2023-06-26 15:10:05.769 INFO Z-WAVE: [Node 126] Value updated: 113-0-alarmLevel 0 => 0
2023-06-26 15:10:05.775 INFO Z-WAVE: [Node 126] Value updated: 113-0-Home Security-Motion sensor status 8 => 0
2023-06-26 15:10:05.832 INFO Z-WAVE: [Node 126] Metadata updated: 48-0-Motion
2023-06-26 15:10:05.837 INFO Z-WAVE: [Node 126] Value updated: 48-0-Motion true => false
[... 67 lines cut ...]
                                  n 1100 ms.
2023-06-26 15:12:08.250 ERROR Z-WAVE-SERVER: Timeout while waiting for an ACK from the controller (ZW0200)
ZWaveError: Timeout while waiting for an ACK from the controller (ZW0200)
    at Driver.sendMessage (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:4322:23)
    at Driver.sendCommandInternal (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:4519:28)
    at Driver.sendSupervisedCommand (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:4564:27)
    at Driver.sendCommand (/opt/node_modules/zwave-js/src/lib/driver/Driver.ts:4612:30)
    at Proxy.set (/opt/node_modules/@zwave-js/cc/src/cc/MultilevelSwitchCC.ts:248:24)
    at Proxy.<anonymous> (/opt/node_modules/@zwave-js/cc/src/cc/MultilevelSwitchCC.ts:324:17)
    at ZWaveNode.setValue (/opt/node_modules/zwave-js/src/lib/node/Node.ts:996:39)
    at NodeMessageHandler.handle (/opt/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:23:38)
    at Object.node (/opt/node_modules/@zwave-js/server/dist/lib/server.js:41:96)
    at Client.receiveMessage (/opt/node_modules/@zwave-js/server/dist/lib/server.js:106:99)
2023-06-26T21:12:09.271Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2023-06-26T21:12:10.375Z CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try i
                                  n 1100 ms.
2023-06-26T21:12:12.482Z CNTRLR   [Node 012] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
2023-06-26T21:12:13.249Z CNTRLR » [Node 019] Multilevel Sensor CC values may be stale, refreshing...
2023-06-26T21:12:13.251Z CNTRLR » [Node 019] querying Dew point sensor reading...
2023-06-26T21:12:13.257Z CNTRLR   [Node 019] No scale preference for sensor type 11, using the first supported s
                                  cale 0
2023-06-26T21:12:13.993Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
```

AlCalzone, 2023-06-27:
Duplicate of #2723

That said, I've actually never seen 700 series sticks suddenly go unresponsive, so this might be a hardware or OS issue.
I'm not sure how you are running Z-Wave JS, is a VM with USB passthrough involved somehow?

jriguera, 2023-06-28:
I am facing a similar issue (not sure if it is the same). The controller is a aeotec ZST10-700 (FW: v7.19.2 / SDK: v7.19.2) running on Raspberry Pi with docker and using the latest version of Zwave JS UI (zwave-js-ui: 8.19.0.487524b ; zwave-js: 11.0.0). I am still investigating a bit the behavior. 

This is some output:

```
2023-06-28 18:08:41.997 ERROR Z-WAVE-SERVER: Timeout while waiting for an ACK from the controller (ZW0200)
ZWaveError: Timeout while waiting for an ACK from the controller (ZW0200)
    at Driver.sendMessage (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4366:23)
    at Driver.sendCommandInternal (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4563:28)
    at Driver.sendCommand (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4674:15)
    at Proxy.set (/usr/src/app/node_modules/@zwave-js/cc/src/cc/MultilevelSwitchCC.ts:251:24)
    at Proxy.<anonymous> (/usr/src/app/node_modules/@zwave-js/cc/src/cc/MultilevelSwitchCC.ts:327:17)
    at ZWaveNode.setValue (/usr/src/app/node_modules/zwave-js/src/lib/node/Node.ts:1019:39)
    at NodeMessageHandler.handle (/usr/src/app/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:20:43)
    at Object.node (/usr/src/app/node_modules/@zwave-js/server/dist/lib/server.js:41:96)
    at Client.receiveMessage (/usr/src/app/node_modules/@zwave-js/server/dist/lib/server.js:106:99)
    at WebSocket.<anonymous> (/usr/src/app/node_modules/@zwave-js/server/dist/lib/server.js:50:45)
2023-06-28 18:10:00.371 ERROR Z-WAVE-SERVER: Failed to send the message after 3 attempts (ZW0202)
ZWaveError: Failed to send the message after 3 attempts (ZW0202)
    at Driver.sendMessage (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4366:23)
    at Driver.sendCommandInternal (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4563:28)
    at Driver.sendCommand (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4674:15)
    at Proxy.get (/usr/src/app/node_modules/@zwave-js/cc/src/cc/VersionCC.ts:243:40)
    at ZWaveController.getAvailableFirmwareUpdates (/usr/src/app/node_modules/zwave-js/src/lib/controller/Controller.ts:5861:61)
    at Function.handle (/usr/src/app/node_modules/@zwave-js/server/dist/lib/controller/message_handler.js:201:30)
    at Client.receiveMessage (/usr/src/app/node_modules/@zwave-js/server/dist/lib/server.js:106:62)
2023-06-28 18:14:51.383 ERROR Z-WAVE-SERVER: Cannot check for firmware updates for node 18: Failed to query fingerprint from the node! (ZW0260)
ZWaveError: Cannot check for firmware updates for node 18: Failed to query fingerprint from the node! (ZW0260)
    at ZWaveController.getAvailableFirmwareUpdates (/usr/src/app/node_modules/zwave-js/src/lib/controller/Controller.ts:5854:10)
    at Function.handle (/usr/src/app/node_modules/@zwave-js/server/dist/lib/controller/message_handler.js:201:30)
    at Client.receiveMessage (/usr/src/app/node_modules/@zwave-js/server/dist/lib/server.js:106:62)
2023-06-28 18:20:01.140 ERROR Z-WAVE-SERVER: Failed to send the message after 3 attempts (ZW0202)
ZWaveError: Failed to send the message after 3 attempts (ZW0202)
    at Driver.sendMessage (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4366:23)
    at Driver.sendCommandInternal (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4563:28)
    at Driver.sendCommand (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4674:15)
    at Proxy.get (/usr/src/app/node_modules/@zwave-js/cc/src/cc/VersionCC.ts:243:40)
    at ZWaveController.getAvailableFirmwareUpdates (/usr/src/app/node_modules/zwave-js/src/lib/controller/Controller.ts:5861:61)
    at Function.handle (/usr/src/app/node_modules/@zwave-js/server/dist/lib/controller/message_handler.js:201:30)
    at Client.receiveMessage (/usr/src/app/node_modules/@zwave-js/server/dist/lib/server.js:106:62)
2023-06-28 18:32:01.144 ERROR Z-WAVE-SERVER: Timeout while waiting for an ACK from the controller (ZW0200)
ZWaveError: Timeout while waiting for an ACK from the controller (ZW0200)
    at Driver.sendMessage (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4366:23)
    at Driver.sendCommandInternal (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4563:28)
    at Driver.sendCommand (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4674:15)
    at Proxy.get (/usr/src/app/node_modules/@zwave-js/cc/src/cc/ManufacturerSpecificCC.ts:109:24)
    at ZWaveController.getAvailableFirmwareUpdates (/usr/src/app/node_modules/zwave-js/src/lib/controller/Controller.ts:5851:5)
    at Function.handle (/usr/src/app/node_modules/@zwave-js/server/dist/lib/controller/message_handler.js:201:30)
    at Client.receiveMessage (/usr/src/app/node_modules/@zwave-js/server/dist/lib/server.js:106:62)
``

There are some log lines of the driver:

```

2023-06-28T16:25:46.883Z SERIAL « 0x010f00a80001100631050122010b00a4f6                                (17 bytes)
2023-06-28T16:25:46.889Z SERIAL » [ACK]                                                                   (0x06)
2023-06-28T16:25:46.897Z CNTRLR   [Node 016] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
2023-06-28T16:25:46.899Z CNTRLR   [Node 016] [~] [Multilevel Sensor] Air temperature: 26.5 => 26.7  [Endpoint 0]
2023-06-28T16:25:46.900Z DRIVER « [Node 016] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -92 dBm
                                  └─[MultilevelSensorCCReport]
[... cut ...] n is when those `Failed to execute controller command after 1/3 attempts.` appear.  I have rebuild the zwave network without security and the behavior seems similar.  The zwave network was build without HA (not even deployed), all actions were done from the Zwave JS UI interface and the first day (sorry I have no data) after adding all devices it was fine (normal cpu usage, but also no actions performed from the controller). A day later, same devices and same configuration those problems appear.

closed 2023-07-16 by zwave-js-assistant[bot]