zwave-js/zwave-js #5959: FGWREU-111 - Report state null with physical buttons

jtonnelier, 2023-06-28:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

After upgrading to ZWaveJSUI 8.19.0 (before I used 8.13.1) , my fibaro state is incorrect when I use physical clic (with specific Target Value it's OK).

In 8.13.1, there is no log about currentValue, but now with 8.19.0 I see somes new logs (check Attach Drivers Logfile - 8.19.0).

There is nothing is newValue and I got a null value for state:

[https://community.jeedom.com/uploads/default/original/3X/8/b/8b42fe8065b1be85daf1d223efbc61a803119409.png](image)

For your information, my shutter is a bubendorff with 4 wires (N, L and 2 commands wires), so I can't use calibrate feature, I want to set it manually (information about my shop).

I configured the FGWREU-111 like this:

- Parameter 151: 5 (5 - Roller blind with built-in driver)
- Parameter 155: 5W (if I disable it, nothing work)
- Parameter 156 / 157: Set time manually

Maybe it's relies with [https://github.com/zwave-js/node-zwave-js/issues/5825](#5825)

If you need more information, please tell me.

### Device information

Manufacturer: Fibaro
Model name: FGWREU-111 
Node ID in your network: 17

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 11.0.0
`zwave-js-ui` branch: 8.19.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Yes update today to 8.19.0 from 8.13.1 but 8.13.1 don't work too.

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

8.13.1:
```
Arg 0:
└─commandClassName: Multilevel Switch
└─commandClass: 38
└─property: targetValue
└─endpoint: 1
└─newValue: 0
└─prevValue: 0
└─propertyName: targetValue
2023-05-31T20:21:41.627Z - value updated
Arg 0:
└─commandClassName: Multilevel Switch
└─commandClass: 38
└─property: duration
└─endpoint: 1
└─newValue: unknown
└─prevValue: unknown
└─propertyName: duration
2023-05-31T20:21:42.510Z - value updated
Arg 0:
└─commandClassName: Multilevel Switch
└─commandClass: 38
└─property: duration
└─endpoint: 1
└─newValue: unknown
└─prevValue: unknown
└─propertyName: duration
2023-05-31T20:21:44.751Z - value updated
Arg 0:
└─commandClassName: Meter
└─commandClass: 50
└─property: value
└─propertyKey: 66049
└─endpoint: 1
└─newValue: 0
└─prevValue: 0.3
└─propertyName: value
└─propertyKeyName: Electric_W_Consumed
```

8.19.0:
```
*2023-06-28T18:09:17.832Z* - **value updated** Arg 0: └─commandClassName: Multilevel Switch └─commandClass: 38 └─property: currentValue └─endpoint: 1 └─newValue └─prevValue: 7 └─propertyName: currentValue

*2023-06-28T18:09:20.193Z* - **value updated** Arg 0: └─commandClassName: Meter └─commandClass: 50 └─property: value └─propertyKey: 66049 └─endpoint: 1 └─newValue: 0 └─prevValue: 0.4 └─propertyName: value └─propertyKeyName: Electric_W_Consumed
```

jtonnelier, 2023-06-28:
[zwavejs_2023-06-28.log](https://github.com/zwave-js/node-zwave-js/files/11899849/zwavejs_2023-06-28.log)

Logfile :)

AlCalzone, 2023-06-29:
I'm afraid there's nothing to be done about that:
```
2023-06-28T20:45:08.841Z SERIAL « 0x010f0004001109600d01002603fefefe5b                                (17 bytes)
                                                              ^^^^^^
```
These 3 bytes mean "unknown" "unknown" "unknown", so the device doesn't know the state itself.

> About that:
[image]

I've opened https://github.com/zwave-js/zwave-js-ui/issues/3153 to make it clearer what `null` means.

jtonnelier, 2023-06-29:
Bad news...

I think it's a firmware issue, because I think module know the state.

In fact, when I use first physicals buttons and after set position with zwavejs, the shutter go to the good position everytime!

If the state was really unknow, it can go to the right position, isn't it?

AlCalzone, 2023-06-29:
I'm inclined to agree

jtonnelier, 2023-06-29:
Impossible to send these informations to Fibaro...

Thanks for your help!

closed 2023-06-29 by jtonnelier