zwave-js/zwave-js #5962: ZSE29 wake up events aren't reported

bobbeims, 2023-06-30:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Wake up events for the Zooz ZSE29 are never detected.

The ZSE29 is supposed to wake up every 4 hours, or can be forced to wake up by removing and re-inserting the batteries. I have never been able to detect a wake up event. I submitted a support request to the Zooz team and worked with them to try everything they could think of to figure out the problem. 

The Zooz team eventually had their lab techs attempt to diagnose the problem, but were unable to do so in Home Assistant. They tested SmartThings, Hubitat, Z-Box and in those platforms the ZSE29 wake up events were detected.

The default configuration for the ZSE29 is to wake up once every 4 hours.

### Device information

Manufacturer: ZSE29
Model name: Outdoor Motion Sensor
Node ID in your network: 054 and 060

I've attached diagnostics files for the ZST10-700 stick as well as the two ZSE29s.
[zwave_js-ZST10-700.json.txt](https://github.com/zwave-js/node-zwave-js/files/11923343/zwave_js-ZST10-700.json.txt)
[zwave_js-Outdoor Motion Sensor 2.json.txt](https://github.com/zwave-js/node-zwave-js/files/11923345/zwave_js-Outdoor.Motion.Sensor.2.json.txt)
[zwave_js-Outdoor Motion Sensor 1.json.txt](https://github.com/zwave-js/node-zwave-js/files/11923346/zwave_js-Outdoor.Motion.Sensor.1.json.txt)

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js: 10.23.2
zwave-js-ui: 8.18.1

[config_entry-zwave_js-diagnostics.json.txt](https://github.com/zwave-js/node-zwave-js/files/11923357/config_entry-zwave_js-diagnostics.json.txt)

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### If yes, where did it work?

Don't know, as I'd never tried to update parameters until recently. That's the only reason I ran into the no-wakeup-detected scenario.

### Attach Driver Logfile

[zwavejs_2023-06-29.log](https://github.com/zwave-js/node-zwave-js/files/11923361/zwavejs_2023-06-29.log)

AlCalzone, 2023-07-01:
I guess you wanted to close the other issue, not this one?

bobbeims, 2023-07-01:
@AlCalzone you’re right … doh!

AlCalzone, 2023-07-05:
I don't see any wakeup events either. It could be that the wakeup destination isn't set, so the device does not know who to notify.
Can you re-interview one of those (need to wake them up manually) and post the log of that?

bobbeims, 2023-07-13:
@AlCalzone sorry for the slow reply. My ZSE29s are located where I have to get ladders out to access them, so it took a few days to find time to pull one down, try to force a wake up and capture the logs, etc.

I've attached both the z-ui and wavejs logs that were captured with debug level enabled. I pulled the unit down at about 3:20 PM (US Central time) and went through the process to trigger a wake up a couple of times and initiated a re-interview. I put the unit back in place by 4:00 PM CDT.

[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/12039996/zwave-js-ui-store.zip)

bobbeims, 2023-07-17:
I posted debug logs 4 days ago but haven’t seen a response, yet. Please reopen the ticket.

bobbeims, 2023-07-17:
@AlCalzone i can’t reopen this, since it was closed by the triage bot. Can you re-open it?

AlCalzone, 2023-07-17:
@bobbeims I'm afraid there's no interview in that log, nor is there any mention of the word "awake" or "wake up". Did you see the interview happen in the UI?

bobbeims, 2023-07-18:
@AlCalzone during my email conversations with the Zooz tech support I tried all sorts of methods to cause a wake up. I could never see any evidence in the UI that a wake up event was making it through the stack.

After the Zooz folks had asked me to try all of those methods is when we gave up and they asked me to report the issue to the HA dev community.

I’m hoping we can get you in direct contact with the Zooz team to diagnose the issue. It is obviously going to take some sort of network snooping at the PHY level to figure out where the disconnect is happening. 20 years ago I did that sort of low-level driver development and diagnosis, but I don’t have any equipment or access to a lab for Z-wave packet sniffing.

AlCalzone, 2023-07-18:
I have contact with Zooz and they know where to reach me. I'm afraid though that they'll just send me one of your logs I've already seen.
That said, I do have the device and I just tested:
```
09:53:28.575 DRIVER « [Node 002] [REQ] [BridgeApplicationCommand]
                      │ RSSI: -69 dBm
                      └─[WakeUpCCWakeUpNotification]
09:53:28.577 CNTRLR « [Node 002] received wakeup notification
2023-07-18 09:53:28.578 INFO Z-WAVE: [Node 002] Is now awake
09:53:28.580 CNTRLR   [Node 002] The node is now awake.
```
So waking up works on mine, but it is super finnicky. I have to click the tamper switch 3 times quickly and then hold for a second, and this only works some of the time.

bobbeims, 2023-07-18:
Well, thanks for that idea. Next time I have a chance, I’ll give that a try (I’m traveling this week). The ironic thing is that the Zooz folks sent me an updated firmware file they thought might address the issue, but if the HA stack never sees a wake-up, the firmware update never starts.

Thanks for your help, but I guess I’ll close this ticket for now and reopen to provide a more info if I ever manage to get that firmware loaded.

bobbeims, 2023-10-06:
In case anyone stumbles across this ticket looking for a solution, here's a quick status update:

- I just bought a new ZSE29 that came in with the v2.1 FW
- Since my last attempts to get this to work, I had updated to the latest HA build (2003.10) which included updates to wave-js to 12.0.3 and wave-js-ui to 2.1
- the newer Home Assistant stack detected the ZSE29 wakeup when I powered it up, and performed an update of the FW to 2.2
- after these HA and FW updates, the default configured waked ups of the ZSE29 are now detected, which means I've been able to re-configure advanced parameters as desired