zwave-js/zwave-js #6007: Minoston MP31ZP parameters do not match documentation or actual device

JeremyChen1213, 2023-07-12:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Device profile parameter has no effect

What do you observe?
Device profile parameter has no effect

What did you expect to happen?
Device profile parameter has no effect

Steps to reproduce the behavior:
  1. Go to 'Device Controller UI'
  2. Click on 'CONFIGURE'
  3. Set parameter
  4. See parameters has no effect

### Device information

Manufacturer: Minoston
Model name: MP31ZP
Node ID in your network: 7

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:DB Version: 11.4.0

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

-

AlCalzone, 2023-07-12:
This is nothing we can fix, since from a Z-Wave perspective the MP31ZP is indistinguishable from the MP21ZP.

https://github.com/zwave-js/node-zwave-js/issues/4955#issuecomment-1229219216
https://github.com/zwave-js/node-zwave-js/pull/4882
https://github.com/zwave-js/node-zwave-js/pull/4894

AlCalzone, 2023-07-12:
...unless they finally fixed their firmware, in which case I need the driver log the issue template asked for.

JeremyChen1213, 2023-07-12:
But I bought MP21Z and MP31Z, their firmware also seems to be modified, no longer match the 'Configure' parameter in Z-Wave JS.

JeremyChen1213, 2023-07-12:
[home-assistant_zwave_js_2023-07-12T10-12-43.354Z.log](https://github.com/zwave-js/node-zwave-js/files/12026664/home-assistant_zwave_js_2023-07-12T10-12-43.354Z.log)
I don't know if this log file is useful to you

AlCalzone, 2023-07-12:
No, I need this one: https://www.home-assistant.io/integrations/zwave_js#how-to-access-the-z-wave-logs
While you have that log tab open, re-interview the device.

JeremyChen1213, 2023-07-17:
Hello, sorry. I got the device model wrong when asking the question. The correct one should be Minoston MP21Z, MP31Z parameters are incomplete, setting Auto-Off Timer is invalid.The description of setting LED Indicator is wrong, and the value does not match.
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/12065709/zwave_js.log)

AlCalzone, 2023-07-17:
> **While** you have that log tab open, re-interview the device.

I don't see an interview in that log.

> setting Auto-Off Timer is invalid.The description of setting LED Indicator is wrong, and the value does not match.

Do you mean on the MP21Z or MP31Z?

JeremyChen1213, 2023-07-19:
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/12090881/zwave_js.log)
I opened the log tab and re-added the device.

JeremyChen1213, 2023-07-19:
MP31Z

AlCalzone, 2023-07-19:
That device identifies as follows:
```
manufacturer: NIE Technology Co., Ltd. (0x0312)
product type: 0xff00
product id:   0xff0c
```
which is the **MP21Z** according to the information at https://products.z-wavealliance.org/products/3973

The**MP31Z** is supposed to have a different product ID `0xff0e` according to https://products.z-wavealliance.org/products/4626, but it seems they never changed that ID in the device firmware.

Which brings us back to my original comment: Minoston messed up, so we cannot distinguish the MP31Z from the MP21Z until they fix it with a firmware update (I wouldn't get my hopes up for that).
To set config params, you'll have to use Z-Wave JS UI and the controls at the bottom of the configuration panel where you specify the parameter number and value size yourself (see device manual for details).

JeremyChen1213, 2023-07-20:
[image]
The https://products.z-wavealliance.org/products/4626 you mentioned should be that MP31ZP is a power monitoring product. But the MP31Z I'm talking about is a pure relay product.

JeremyChen1213, 2023-07-20:
I consulted Minoston's after-sales service, and they gave me the reply that MP21Z and MP31Z are the same product, only the shape changes. That's why I came here to ask if I can solve the problem of parameter mismatch

JeremyChen1213, 2023-07-20:
[image]
I saw in the log of Home Assistant that there should be 5 parameters, but after loading, there are only parameter 1 and parameter 2, and the description of parameter 1 does not match the value. Parameter 2 set value has no function.

AlCalzone, 2023-07-20:
> only the shape changes. [...] parameter mismatch

They are not really the same product then if they have different parameters though? Or is the information for the MP21Z also wrong? The parameter information you're showing only acts as a fallback if we have no matching configuration file.

Since you are already in contact with them, can you ask them if they have engineering sheets for those devices? I.e. this information for each of the MP21Z, MP21ZP (?), MP31Z, MP31ZP:
- Which device / name
- Product ID and type
- configuration parameters

JeremyChen1213, 2023-07-22:
[Minoston MP21Z  Mini Smart Plug.docx](https://github.com/zwave-js/node-zwave-js/files/12135591/Minoston.MP21Z.Mini.Smart.Plug.docx)
[Minoston MP31Z  Mini Smart Plug.docx](https://github.com/zwave-js/node-zwave-js/files/12135592/Minoston.MP31Z.Mini.Smart.Plug.docx)
[MP21ZP  Mini Power Meter Plug.docx](https://github.com/zwave-js/node-zwave-js/files/12135593/MP21ZP.Mini.Power.Meter.Plug.docx)
[MP31ZP  Mini Power Meter Plug.docx](https://github.com/zwave-js/node-zwave-js/files/12135594/MP31ZP.Mini.Power.Meter.Plug.docx)

JeremyChen1213, 2023-07-22:
According to after-sales description
The functional parameters and device ID of MP21Z and MP31Z are exactly the same, only the appearance is different.
The functional parameters and device ID of MP21ZP and MP31ZP are exactly the same, only the appearance is different.

AlCalzone, 2023-07-22:
Ok that certainly explains some things. Some people who have reported these issues in the past must have mixed up the Z and ZP variants.

JeremyChen1213, 2023-07-24:
Then, can it be repaired, and the value corresponding to these parameters?

AlCalzone, 2023-07-24:
I think so, yes!

closed 2023-07-24 by zwave-js-bot

JeremyChen1213, 2023-07-25:
Okay, thank you for your timely reply and help, when can I use the latest version?

AlCalzone, 2023-07-25:
The release went out yesterday. Now it's a matter of time that your application picks it up

JeremyChen1213, 2023-07-28:
Sorry to bother you again. Thank you very much for your help, Configuration is very good, but I found something wrong in PARAMETER6. Then I went to ask the Minoston after-sales service, and found that they were working wrong, and the PARAMETER6 parameters sent to me were wrong.The attached picture is the correct one.
[image]

JeremyChen1213, 2023-07-28:
Additional note I'm talking about MP21Z/MP31Z.

AlCalzone, 2023-07-28:
https://github.com/zwave-js/node-zwave-js/pull/6087 changes that

JeremyChen1213, 2023-07-29:
Okay, thank you for your timely reply and help

JeremyChen1213, 2023-08-01:
Hello, sorry to bother you again. Since I consulted and received the configuration documents of MP21ZP/MP31ZP before, I received a proactive contact from Minoston today and told them that the firmware has been upgraded according to customer needs, and a lot of configuration parameters have been added. I would like to trouble you to add configuration parameters again.
I am so sorry. Attached is the latest configuration document.
[MP21ZP:MP31ZP Meter Switch User Manual.pdf](https://github.com/zwave-js/node-zwave-js/files/12223386/MP21ZP.MP31ZP.Meter.Switch.User.Manual.pdf)

AlCalzone, 2023-08-01:
Could you add this information to https://github.com/zwave-js/node-zwave-js/issues/6097 which seems to be the same issue?

JeremyChen1213, 2023-08-02:
Of course, I will see how to add.

AlCalzone, 2023-08-02:
I think the updated manual is already posted there

PR #6056 (2023-07-24): fix(config): split and correct Minoston MP21/31Z/ZP config files
fixes: #6007
fixes: https://github.com/zwave-js/node-zwave-js/issues/5328 (the right way)