zwave-js/zwave-js #6010: Most of the device paramaters are missing

skull791, 2023-07-13:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

building 36 ADC-T40K HD

### Manufacturer ID

400

### Product Type

1

### Product ID

10

### Firmware Version

1.6

### Is the device listed on the Z-Wave Alliance website?

no

skull791, 2023-07-13:
> Do you have a manual or something that explains which device parameter exist and what they do?

I do not have parameter manual and installation manual have nothing in it. I can’t find anything online anywhere like I was able for my adc-t3000 thermostat. I have a screen shots from alarm.com interface that has some items that can be adjusted. I know there are more since alarm.com likes to only give you some. I attaching screen shots if that would help. Is there like a zwave decoder or sniffer that can be run to see what parameters are available ?Sent from my iPhone

AlCalzone, 2023-07-13:
I went through a similar procedure with another user for this:
https://github.com/zwave-js/node-zwave-js/pull/5557

That involved setting up a Zniffer, changing each parameter to their min/max limits via the alarm.com interface, documenting each change and then using that info to reverse-engineer the Zniffer traces to write the config file.

skull791, 2023-07-13:
Ok. I’m up for anything that we can do to get this set up. I know there are also options that are not showing in alarm.com menu that are there like adc t3000 had. Is there a way to find them out. I think I have pdf of all parameters of t-3000 that I can send to you.  Not sure if that would help.Now how do I set this up to do all testing and Geri g all parameters. Can you walk me through so I can supply you with everything you need for configuration file Thank youAdam Sent from my iPhone

AlCalzone, 2023-07-20:
You need a Windows PC, the [Zniffer software](https://siliconlabs.my.site.com/s/contentdocument/0691M000003WIBt) (available after logging in, but I can send it to you if that doesn't work), if Security is used the network keys, and a Zniffer stick.

Zniffers are essentially Z-Wave sticks [flashed](https://community.silabs.com/s/article/z-wave-500-converting-a-uzb3-controller-to-a-zniffer?language=en_US) with the firmware from the above download.
Ideally use an UZB3 for your region:
- https://eu.mouser.com/ProductDetail/Silicon-Labs/ACC-UZB3-U-STA?qs=PqoDHHvF64%2FtTVCvZ7NeaA%3D%3D for US
- https://www.digikey.com/en/products/detail/silicon-labs/ACC-UZB3-E-STA/6111631 for EU

When you have that set up, start the Zniffer application, configure it to use the Zniffer stick and start capturing.
Then change each parameter to their min/max limits via the alarm.com interface, documenting each change (which parameter was changed to which value when?).
When done, send me that information along with the Zniffer trace file. We can do that [on Discord](https://discord.gg/HFqcyFNfWd).

skull791, 2023-07-21:
Ok. If you don’t mind can you send me that software. Also can I reflags my stuck back to normal firmware after we are done. Can you also send me stick flashin tools and firmware.Thank you Sent from my iPhone

AlCalzone, 2023-07-25:
Please send me an email to the address in my profile. I can't see yours here.

skull791, 2023-07-25:
Just send you email to your email

sdt16, 2024-01-31:
I’m willing to take this on. Just ordered the ACC-UZB3-U-STA. I have a SILabs account, so I can grab the software. The rub is I don’t have an alarm.com system, so hopefully there’s another way to enumerate everything, or at least make some progress.

AlCalzone, 2024-02-01:
You can use a driver function in Z-Wave JS UI:
```ts
const node = driver.controller.nodes.get( 2 ); // assuming 2 is your node ID
node.commandClasses.Configuration.scanParametersLegacy()
```
This should scan through all possible parameters and print their size and value to the log.
You can then use the "custom configuration" fields to GET and SET their values.

The hard part will be figuring out what actually happens when changing the parameters.