zwave-js/zwave-js #6012: Node interview stuck at "node info" for Ring Keypad V2 and server not responding

Charles-HL, 2023-07-13:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Description:
I am encountering an issue while attempting to pair a Ring Keypad V2 with my setup consisting of a Raspberry Pi 4 and an AEOTEC ZW090-C Gen5 Plus. Following the official documentation, I deployed the Docker using Docker Compose and configured the S2 and S0 keys.
When I initiate the pairing process through the "manage node" feature, the keypad is successfully detected, and I am prompted to enter the DSK pin. However, in the node table, the interview status of the keypad remains stuck at "node info" with a yellow status. Subsequently, the processor utilization spikes to 60% to 100%, and after some time, the status at the top right of the UI turns red, indicating disconnection. Attempting to refresh the page yields no response, while the CPU usage remains persistently high. Even after restarting the Docker, I can briefly connect to the UI for a few seconds before encountering the same problem again.

To reproduce:
Set up a Raspberry Pi 4 and an AEOTEC ZW090-C Gen5 Plus
Deploy the Docker environment using Docker Compose.
Access the user interface (UI) of the Docker deployment.
Navigate to the "manage node" section.
Initiate the pairing process for the Ring Keypad V2.
When prompted, enter the DSK pin for the keypad.
Observe the node table in the UI and notice that the interview status of the keypad remains at "node info" with a yellow status.
Monitor the CPU usage, which may increase to 60% to 100%.
After some time, the status at the top right of the UI changes to red, indicating disconnection.
Attempt to refresh the page, but observe that there is no response.
Note that the CPU usage continues to remain high.
Restart the Docker environment.
Briefly connect to the UI for a few seconds after the restart.
Notice that the issue reoccurs, with the keypad remaining disconnected and the CPU usage spiking again.

### Device information

Manufacturer: Ring
Model name: Keypad Gen 2 Model 5B28S8
Node ID in your network: 2

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[... 1653 lines cut ...]
    {
      "index": 0,
      "label": "Root Endpoint"
    }
  ],
  "isSecure": true,
  "security": "S2_AccessControl",
  "isControllerNode": false,
  "keepAwake": false,
  "deviceClass": {
    "basic": 4,
    "generic": 64,
    "specific": 11
  },
  "firmwareCapabilities": {
    "firmwareUpgradable": false
  },
  "deviceId": "undefined-undefined-undefined",
  "lastActive": 1689243332715,
  "lastReceive": 1689243332715,
  "lastTransmit": 1689243332715,
  "errorReceive": false,
  "errorTransmit": false
}
`

Charles-HL, 2023-07-13:
@AlCalzone  Those are the logs when I pair the device without security. On those logs, the node id is 3, the pairing works sucessfully with the status complete, but the device is unusable as expected
[z-ui_2023-07-13.log](https://github.com/zwave-js/node-zwave-js/files/12040124/z-ui_2023-07-13.log)
[zwavejs_2023-07-13.log](https://github.com/zwave-js/node-zwave-js/files/12040125/zwavejs_2023-07-13.log)

Charles-HL, 2023-07-14:
@AlCalzone  With only S0 security, the device is still unusable but the device name and model are detected.
When I pair the device I have this message
[image]
I have this S0 key BDA6C4737BEA173D1FCA0650A890424B and the node id is 4.
The logs: [zwavejs_2023-07-14.log](https://github.com/zwave-js/node-zwave-js/files/12050284/zwavejs_2023-07-14.log)

Indianb0y016, 2023-07-15:
I knew something was wrong when I issued a reinterview command to my second ring keypad today and the network just crashed.
This behavior is completely abnormal, since it didnt give issues like this before. I used to be able to pair the devices just fine before. Sadly I dont have logs or dates of when I paired them. I can try and check my various backups for any logs or info. 
Since I am also facing this issue, let me know if I can provide any logs that may help.

AlCalzone, 2023-07-16:
> the device is still unusable

Something is off with the S0 communication here - I need to look into that.

Charles-HL, 2023-07-16:
I switched to zwavejs/zwave-js-ui:8.16.1 , in this version all work fine

Indianb0y016, 2023-07-16:
I just wanted to add something else as well. Although NOT specific to the ring keypad, version 1.19 also changed how some of my devices report configuration parameters. My radio CT100 thermostat, with the current configuration file reports every available configuration parameter as defined in the config file in the db. This is on version 1.18.1.
HOWEVER, on version 1.19, the configuration parameters are all messed up and not respecting the file anymore. Even after tweaking with the file to determine why it wasn't respecting it, I couldn't figure out why. But after yesterdays rollback to 1.18.1, the configuration is perfectly fine. Heres an example.
Parameter 2 is a partial parameter, so it has multiple options.

**1.18.1**
2[0xff0000000] 
value: 2 
Reports HVAC type correctly with the respective label

**1.19**
2[0xff0000000]
Unable to identify this, and instead, a new configuration parameter in the zwavejs is found, under "param002" with value: 1
The original HVAC type parameter entry is now blank and unknown. There are other configuration parameters that popup and report "undefined"

I hope this may be of some use

andltayl, 2023-07-18:
I also have this exact issue. If there is anything I can do to help I'd be happy to do so.

PR #6029 (2023-07-18): fix: infinite loop persisting 4-byte bitfield parameters with the high bit set
Looks like we fell into a JavaScript trap here :(

fixes: #6012
closed 2023-07-18 by zwave-js-bot