zwave-js/zwave-js #6015: Missing node information after (re)interview

kdknigga, 2023-07-14:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I've noticed problems recently (it's hard to nail it to a version right now, but I'm testing now with zwave-js-ui: 8.20.0.14b61ff, zwave-js: 11.4.2) with interviewing nodes.  It seems that often interviews will not successfully populate all of the information about a node.  Even nodes that have been successfully interviewed before now come up with a lot of missing information after a re-interview.

For example, I have an Inovelli LZW42 that's been a part of my network for a long time and used to be identified just fine by zwavejs.  Now, after a reinterview (several, in fact), I get "Unknown manufacturer 0xXXXX" and "Unknown product 0xXXXX" and the configuration editor shows no parameters.

Another example is an Aeotec Siren 6 that was working fine.  After a reinterview, while I see parameters listed under Configuration, few if any of them are named.  For parameter 1, instead of it being broken into partial parameters with nice names like Light Effect Index and Tone Play Mode, I just get "[7-112-0-1] param001 (property)".

### Device information

Manufacturer: Inovelli
Model name:  LZW42
Node ID in your network: 3

Manufacturer: Aeotec
Model name:  Siren 6 (ZW164)
Node ID in your network: 7

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwave-js-ui: 8.20.0.14b61ff
zwave-js: 11.4.2

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

This worked a few months ago.  I'll try to step back in versions to find where it broke and update the ticket.

### Attach Driver Logfile

[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/12053291/zwavejs_current.log)

kdknigga, 2023-07-14:
I stepped through some recent versions of ZUI and found that this is probably two problems.

The partial parameters problem that's happening on the siren (node 7) starts with ZUI 8.19.

The problem where the LZW42 (node 3) stops successfully interviewing at all starts with ZUI 8.20.

mundschenk-at, 2023-07-15:
I've noticed this with a new include as well (Aetoec Smart Switch 7 EU version), some parameters are discovered fine, others are listed as e.g. `<node>-112-0-9] param009 (property)`. This is on the HA community add-on using Z-Wave JS UI v8.19.0.