zwave-js/zwave-js #6017: CT101 Thermostat Showing as two climate entities, and no fan

blacktirion, 2023-07-16:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

See attached Bug from the following Home Assistant Issue: https://github.com/home-assistant/core/issues/48452

### Device information

Manufacturer:  Radio Thermostat of America
Model name:  CT101
Node ID in your network: 3

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: stable
`zwave-js-ui` branch: stable

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

See attached Bug from the following Home Assistant Issue: https://github.com/home-assistant/core/issues/48452

If more are required other than this already filed issue, I can gather if needed. First time filing, so I can definitely gather if required.

AlCalzone, 2023-07-16:
Please make a [driver log](https://www.home-assistant.io/integrations/zwave_js/#how-to-access-the-z-wave-logs) of a re-interview, loglevel `debug` and attach it here as a file (drag & drop into the text field).

blacktirion, 2023-07-17:
Attaching zwave_js.log Driver log re-interview as loglevel `debug`
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/12071251/zwave_js.log)

blacktirion, 2023-07-24:
Is there any more information required from me?

AlCalzone, 2023-07-24:
Sorry, I just missed the ping. From that driver log it looks like the fan mode and fan state are queried and values for those are created/exposed. Given that the HA issue is already 2 years old, I think it's worth taking another look at the HA side of this.

@raman325 @marcelveldt

AlCalzone, 2023-07-24:
I'm not sure that's the same problem. According to the interview above, the device has 2 endpoints in addition to the root, all of which are `General Thermostat V2`. That device class only has mandatory support for Thermostat Mode and Thermostat Setpoint.

The 2 endpoints have identical capabilities
```
2023-07-17T15:26:24.923Z CNTRLR « [Node 003] received response for device endpoints:
                                  endpoint count (individual): 2
                                  count is dynamic:            false
                                  identical capabilities:      true
```
and only support these two non-mandatory CCs:
```
supported CCs:
· Multilevel Sensor
· Association
```

Only the root device has support for the fan CCs:
```
2023-07-17T15:26:23.473Z CNTRLR « [Node 003] node info received
supported CCs:
· Basic
· Clock
· Indicator
· Manufacturer Specific
· Multilevel Sensor
· Thermostat Mode
· Thermostat Operating State
· Thermostat Fan Mode
· Thermostat Fan State
· Thermostat Setpoint
· Version
· Configuration
· Battery
· Association
· Multi Channel
```

Therefore the fan stuff belongs on the root endpoint and isn't mapped despite the compat flag. Maybe we need to force-remove all thermostat related CCs from the endpoints and just keep the ones on the root? Seems like that functionality is entirely duplicated anyways.

raman325, 2023-07-26:
Adding my comment from Discord here for visibility:

> discovery is triggered by the presence of certain values, in this case a thermostat mode CC value (https://github.com/home-assistant/core/blob/dev/homeassistant/components/zwave_js/discovery.py#L641-L650), which in turns triggers entity creation. During climate entity creation, we look for all of the other values that provide the additional  capabilities (https://github.com/home-assistant/core/blob/dev/homeassistant/components/zwave_js/climate.py#L143-L193). Typically we check for additional values on the same endpoint as the discovered value, but I also had a CT101 and ran into a similar issue where values that needed to be related were on different endpoints, so we updated the climate platform to search for values on other endpoints if it can't find it on the same endpoint as the discovered value (https://github.com/home-assistant/core/blob/dev/homeassistant/components/zwave_js/entity.py#L284-L298).

> The discovery logic does not have any intelligence between discovered values, so the two climate entities get created because the two thermostat mode values are found from each of the non root endpoints. But both should have fan mode support even if the value exists on the root endpoint because of the logic to check other endpoints

raman325, 2023-07-26:
@blacktirion can you please restate what the issue is that you are experiencing? I have a hard time believing your issue is exactly the same as what was reported in that issue because we made modifications to account for this scenario (per above)

blacktirion, 2023-07-26:
[image]
Here is my entity state, from the page on the device integration page. 

Additionally, here's the entity itself. 
[image]
Notice that I DO have fan control here, but as I've seen on other Zwave thermostats, this should also have a fan entity, should it not?

Lastly, I noticed that the active mode reported on the thermostat does not always match the current mode that is active. For example, it will show "Cooling" when it's Idle, or "Idle" when it is cooling. 

Another example of this dual entity behavior happens on my brother's house. He is also running the CT101, same firmware level. However, he's on the Aeotec Gen5 Zwave stick, where as I am on a Zooz 800 LR stick.

I hope this explains more about what I'm seeing. I can post more if I need. 

The fan entity missing/being disabled does not really bother me, so much as the dual climate entities as well as the active mode not matching.

blacktirion, 2023-07-26:
That is what should be the fan control, but it is disabled by the integration by default. Enabling it will bring it to active, but when you attempt to perform an action, it says "failed to call service fan/turn_on connection lost". 

Since the fan is controllable by the climate entity, instead of fan entity, I will be fine with that. Additionally, either of the climate entities seem to control the unit, so I am OK with disabling it, if that is the recommendation. I wanted to mostly open this issue to make sure other people reading down the line had some more information. I was hoping the issue where the active state of the unit not syncing was related to the dual entity presentation, but if it is not, it does not seem like much I can do. 

If the solution is to control the fan state via the climate entity + disable the second entity, I will accept that. I figured it should have been reported down stream from the original issue on the other post and it doesn't seem like that OP ever got around to it. 

Thanks for your time @kpine.

blacktirion, 2023-07-26:
> Are you waiting long enough for that? If so, please provide HA logs here, or feel free to create a new issue in the HA project. There shouldn't be any problems using the entity.

Yup, it took about 30-60 seconds to become active, and it "lit up" but clicking those buttons did nothing. I'll open a separate case on the HA project if it becomes an issue. 

> If you want to debug this problem further, I think you'll need to enable driver logs and try to observe whether changes to the thermostat operating state is being communicated by the device.

Polling the device by using the zwave_js.refresh_value service with the refresh_all_values attribute did seem to synchronize it. This itself might be an HA issue as well. 

> Also check the group associations and make sure the Lifeline group is assigned to the controller, but if were not you'd get no events at all. Or maybe only the root endpoint gets the events, which would be a problem with the device configuration.

This is a bit over my head :) I'll see if I can figure it out. I assume that since it IS getting data back via a manual poll, that it is functioning properly, but since my previous venture into Zwave was with SmartThings, I was using a custom driver, and I don't recall exactly how often it refreshed the state.

I'm not enough of a z-wave expert to inform anything, so I'll just do what the experts recommend :)

raman325, 2023-07-26:
> Polling the device by using the zwave_js.refresh_value service with the refresh_all_values attribute did seem to synchronize it. This itself might be an HA issue as well.

To confirm this, check the logs. Specifically, open the live log window, switch to debug mode, and then attempt to switch your thermostat to cooling/idle and check the logs to see if it reflects the change in operating mode. As @kpine mentioned, if you don't see it in the logs, that means HA can't see it and it is either a device or a driver problem. My bet is on the device because the driver (and the HA integration) handles these value updates fine for other devices that use the same command class.

> This is a bit over my head :) I'll see if I can figure it out. I assume that since it IS getting data back via a manual poll, that it is functioning properly, but since my previous venture into [...]

Many Z-Wave controllers (SmartThings included) implement automatic polling to bypass issues like this. zwave-js has opted not to do that automatically and to give control to the user on applying any polling that is needed. This is better for battery and performance, but it means that some devices will have stale state until you call the refresh service, which causes the driver to poll the device for the latest state information.

apella12, 2023-07-30:
I have these devices and they are a PITA.  The only reliable (timely) value is the air temperature (validated with a zniffer). I have a rule to 1) poll the Opstate (66) when Air=Setpoint and Opstate (running (1 or 2) to get the Opstate to zero and 2) when Air > Setpoint by the (cooling) hysteresis (112-0-7) and Opstate =0 to get the Opstate to 2.  I'd say it triggers the polls about half the time. 

I'm not a HA user, so don't know how to translate all that to your system

AlCalzone, 2023-07-31:
@apella12 do you know if the thermostat-related functionality on the additional endpoints is needed at all? Or do they expose everything but the additional sensors on the root endpoint 0?
If the latter is the case, we could remove all non-root endpoint functionality, (maybe except the multilevel sensor on endpoint 2), similar to what we did on the CT32.

apella12, 2023-07-31:
I do not have complete understanding, but the endpoint quirks (as opposed to the reporting quirks) are related to the humidity reading.  The humidity is on endpoint 2. Further, at least for the OH binding it comes across in a non-standard way (multiInstance but uses the multiSensor format). 
[image]

I have used this unmerged PR https://github.com/openhab/org.openhab.binding.zwave/pull/1344/files to get the humidity reading for my devices locally.  In my testing of Zwave-js-ui there has been no problem with humidity reporting as it seems to convert the frame to multisensor. 

AFAIK for all the non-humidity items are only root or endpoint 1. From my zwave-js-ui/mqtt it looks like CC 64 and 67 are on on EP 1, but the duplication of those CC's on EP 2 are not used).

 I hope that helps

AlCalzone, 2023-08-11:
I have a PR up that should fix this issue. Would be good if someone with the device could verify it works though.

The file: https://raw.githubusercontent.com/zwave-js/node-zwave-js/f5b81cfd6ec2c1de053565429498620f4b424853/packages/config/config/devices/0x0098/ct101.json
How to test: https://zwave-js.github.io/zwave-js-ui/#/device-support/custom-device-files

apella12, 2023-08-11:
two questions before I embark on this adventure with a working installation.
1) Currently I do not see EP 0 on the proposed EP1 and EP2 removals.  Will that appear on rediscovery?
[image]

2) Can I get back to what I have now by deleting the file in the config folder?

AlCalzone, 2023-08-11:
yes and yes (2. needs another re-interview then)

apella12, 2023-08-11:
Ok seems to work fine (for about an hour anyway).  Note that Basic CC still has the two EPs.  Not sure if it is used for anything.  After reinterview and MQTT clean-up
[image]

AlCalzone, 2023-08-12:
> Basic CC still has the two EPs. Not sure if it is used for anything

I wasn't sure either. The manual doesn't mention it aside from it being supported. Guess I can remove that too.

apella12, 2023-08-12:
FYI- I looked a bit about this "issue" and it is probably obvious to you, but I'm not using the HA preformatted json, instead subscribing to the relevant messages, regardless of EP, so never had the "issue" described. With the PR, I just redirected my subscriptions to EP 0, so not a test of the HA json.

closed 2023-08-14 by zwave-js-bot