zwave-js/zwave-js #6030: Aeotec Siren/Doorbell 6 Interview Issues / Missing Capabilities

polskikrol, 2023-07-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Issue is related to "Aeotec Indoor Siren 6 and Doorbell 6 (ZW164) Undefined values under multiple command classes #2568". Simply having this device on the Zwave network and trying to interview/upgrade firmware via Zwave JS is not possible.

What do you observe?

Multiple interviews conclude VERY quickly with very little information. The FW: vundefined is seen and not all capabilities, ie tones etc are uncovered.

### Device information

  "productLabel": "ZW162",
  "productDescription": "Doorbell 6",
  "manufacturer": "Aeotec Ltd.",
Node ID in your network: 21

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.20.0.14b61ff
zwave-js: 11.4.2

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

.

AlCalzone, 2023-07-19:
What the bot said. This **really** needs a driver log of a re-interview to investigate.

polskikrol, 2023-07-19:
Hopefully this is what is needed:

```
2023-07-19 08:54:22.990 INFO Z-WAVE: [Node 021] Interview started
2023-07-19 08:54:22.994 INFO Z-WAVE: Success zwave api call refreshInfo undefined
2023-07-19 08:54:23.007 INFO Z-WAVE: [Node 021] Interview stage PROTOCOLINFO completed
2023-07-19 08:54:23.049 INFO Z-WAVE: [Node 021] Interview stage NODEINFO completed
2023-07-19 08:54:23.090 INFO Z-WAVE: [Node 021] Metadata updated: 114-0-manufacturerId
2023-07-19 08:54:23.091 INFO Z-WAVE: [Node 021] Value added: 114-0-manufacturerId => 881
2023-07-19 08:54:23.091 INFO Z-WAVE: [Node 021] Metadata updated: 114-0-productType
2023-07-19 08:54:23.092 INFO Z-WAVE: [Node 021] Value added: 114-0-productType => 259
2023-07-19 08:54:23.093 INFO Z-WAVE: [Node 021] Metadata updated: 114-0-productId
2023-07-19 08:54:23.093 INFO Z-WAVE: [Node 021] Value added: 114-0-productId => 162
2023-07-19 08:54:40.237 INFO Z-WAVE: [Node 021] Interview stage COMMANDCLASSES completed
2023-07-19 08:54:40.240 INFO Z-WAVE: [Node 021] Interview stage OVERWRITECONFIG completed
2023-07-19 08:54:40.241 INFO Z-WAVE: [Node 021] Interview stage COMPLETE completed
2023-07-19 08:54:40.245 INFO Z-WAVE: [Node 021] Value added 21-114-0-manufacturerId => 881
2023-07-19 08:54:40.245 INFO Z-WAVE: [Node 021] Value added 21-114-0-productType => 259
2023-07-19 08:54:40.245 INFO Z-WAVE: [Node 021] Value added 21-114-0-productId => 162
2023-07-19 08:54:40.246 INFO Z-WAVE: [Node 021] Ready: Aeotec Ltd. - ZW162 (Doorbell 6)
2023-07-19 08:54:40.251 INFO Z-WAVE: [Node 021] Interview COMPLETED, all values are updated
```

Also, have uploaded the driver log as well:

[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/12095161/zwavejs_current.log)

AlCalzone, 2023-07-19:
You previously included the device using Security (S0 or S2), but did not configure the network keys. Therefore Z-Wave JS cannot access most of the functionality of the device.

Either configure the same keys that were used to include, or if that isn't possible exclude the device, factory reset it, then include it again.

polskikrol, 2023-07-19:
The setup I have is a bit different. Vera Plus is the primary controller, and Aeotec Z-stick 7 was placed into learning mode via PC Controller 5, and then afterwards put into Inter NUC running ZWAVE JS UI and HA on docker containers. I was able to reset the Doorbell, and re-include it on the primary controller (Vera Plus) and it was given a new zwave ID. I also was able to obtain the hex of the S0 security key, and copied that over to the ZWAVE JS UI config. However, I do not see the nodes update on the Aeotec Z-stick 7. Cant delete the removed zwave ID node for the Doorbell, and dont see the new one pop up. Is there something I am missing for the network to update on ZWAVE JS UI? Do I need to move the Z-stick back to PC Controller 5 and re-learn the network? Appreciate the advise and assistance as I wrap my head around this.

polskikrol, 2023-07-19:
On the Vera Plus, under ZWAVE advanced options I have a few items that can be selected:
- Copy Z-Wave network from a primary controller (appears to be if one wants to take Z-stick 7 zwave config and toss it onto the Vera)
- Controller shift (trasfer role of primary to another z-wave controller)
- Get network update from SUC/SIS (update slave from the primary to which it is added, used when vera is secondary)
So, all of these options on Vera would appear to be useful when Vera is secondary and not primary as is the current case. Didnt see any similar options on ZWAVE JS, unless I am blind. Would the better approach be to use Vera as secondary, and transfer primary role over to Zstick 7 and ZWAVE JS? Then presume I could update vera's copy of the zwave config from ZWAVE JS.

AlCalzone, 2023-07-20:
Z-Wave JS does not support being the secondary controller, which is probably why you're not seeing the changes you're expecting.

closed 2023-08-07 by zwave-js-assistant[bot]