zwave-js/zwave-js #6032: Door Lock fails to set time

kdknigga, 2023-07-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Unknown

What do you observe?
The clock on the lock never gets successfully set.  I can query the lock for the time and date and it's always wrong.

```
2023-07-18T22:29:34.387Z DRIVER » [Node 087] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      63
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 168
                                    │ security class:  S2_AccessControl
                                    └─[TimeCCTimeGet]

2023-07-18T22:29:35.708Z DRIVER « [Node 087] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 95
                                    │ security class:  S2_AccessControl
                                    └─[TimeCCTimeReport]
                                        time: 00:05:33

2023-07-18T22:29:35.744Z DRIVER » [Node 087] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      64
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 169
                                    │ security class:  S2_AccessControl
                                    └─[TimeCCDateGet]

2023-07-18T22:29:35.818Z DRIVER « [Node 087] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 96
                                    │ security class:  S2_AccessControl
                                    └─[TimeCCDateReport]
                                        date: 1970-01-01
```
(full log attached)

Pulling the battery on the lock and reinserting it will trigger the lock to ask for the time, but it setting it seems to fail.

```
2023-07-18T22:24:09.045Z DRIVER « [Node 087] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 90
                                    │ security class:  S2_AccessControl
                                    └─[SupervisionCCGet]
                                      │ session id:      2
                                      │ request updates: false
                                      └─[TimeCCTimeGet]

2023-07-18T22:24:09.066Z DRIVER » [Node 087] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      55
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 163
                                    │ security class:  S2_AccessControl
                                    └─[SupervisionCCGet]
                                      │ session id:      62
                                      │ request updates: true
                                      └─[TimeCCTimeReport]
                                          time: 17:24:09

2023-07-18T22:24:09.117Z DRIVER « [Node 087] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 91
                                    │ security class:  S2_AccessControl
                                    └─[SupervisionCCReport]
                                        session id:          62
                                        more updates follow: false
                                        status:              Fail
                                        duration:            0s

2023-07-18T22:24:09.144Z DRIVER » [Node 087] [REQ] [SendData]
                                  │ transmit options: 0x24
                                  │ callback id:      0
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 164
                                    │ security class:  S2_AccessControl
                                    └─[SupervisionCCReport]
                                        session id:          2
                                        more updates follow: false
                                        status:              Success
                                        duration:            0s
```
(full log attached)

What did you expect to happen?
The time to get set correctly.

I half expect this to be a firmware problem as this works with another lock I have using S2 Access Control and Time CC v1, but I don't know what I'm looking at enough to be sure.

### Device information

Manufacturer: Schlage
Model name:  Keypad Lever (JFE109)
Node ID in your network: 87

[node_87.json.txt](https://github.com/zwave-js/node-zwave-js/files/12089479/node_87.json.txt)

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwave-js-ui: 8.19.0
zwave-js: 11.0.0

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### If yes, where did it work?

Not to my knowledge, at least.

### Attach Driver Logfile

[zwavejs_2023-07-18.log](https://github.com/zwave-js/node-zwave-js/files/12089482/zwavejs_2023-07-18.log)

AlCalzone, 2023-07-19:
> half expect this to be a firmware problem

I do too: This incoming request makes no sense:
```
2023-07-18T22:24:09.045Z DRIVER « [Node 087] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 90
                                    │ security class:  S2_AccessControl
                                    └─[SupervisionCCGet]
                                      │ session id:      2
                                      │ request updates: false
                                      └─[TimeCCTimeGet]
```
It shouldn't be using Supervision, as the `TimeCCTimeGet` expects a response. The response to that
```
2023-07-18T22:24:09.066Z DRIVER » [Node 087] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      55
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 163
                                    │ security class:  S2_AccessControl
                                    └─[SupervisionCCGet]
                                      │ session id:      62
                                      │ request updates: true
                                      └─[TimeCCTimeReport]
                                          time: 17:24:09
```
makes a little more sense since it makes sure the device understands the report, but I suspect that it doesn't expect a supervision-encapsulated report and thus fails.

AlCalzone, 2023-07-19:
@kdknigga does changing the time using the `setDateAndTime` API work or is this a different issue than the one on Discord?

AlCalzone, 2023-07-19:
I have a potential fix up at https://github.com/zwave-js/node-zwave-js/pull/6034 but I'm not sure if that will solve the actual issue. This makes it so that the responses to supervised Get requests are no longer supervised. While the request being supervised is nonsense, this hopefully makes it so the device understands the response.

kdknigga, 2023-07-19:
> @kdknigga does changing the time using the `setDateAndTime` API work or is this a different issue than the one on Discord?

As far as I can tell, the `setDateAndTime` call does nothing for this node... possibly any node that supports Time v1.  When I do it for a node that supports Clock, I see a Clock Set command go out.  When I do it for this node, though, I see nothing get sent.  I'm not sure what it would send, as there's no "set" in Time.  Perhaps an unsolicited report?

I'll be able to experiment more in about 13 hours.

kdknigga, 2023-07-19:
@AlCalzone , snuck in a little time to test before work.

When I click "Set Date and Time" on a device that supports Clock, I see:

```
2023-07-19T11:39:56.931Z DRIVER » [Node 054] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      63
                                  └─[ClockCCSet]
                                      clock setting: Wednesday, 06:39
```

I get the same result when I run the following (which I assume ZUI is doing?):

```
const node = driver.controller.nodes.get(54);
node.setDateAndTime();
```

When I click that button for a device that supports Time, I get literally nothing in the logs even through ZUI reports "API syncNodeDateAndTime ended successfully".

The driver function as above also reports nothing in the logs.

I thought to try an unsolicited reportTime, and that seems to have worked!

```
const node = driver.controller.nodes.get(87);
const time = node.commandClasses.Time;

if (time.isSupported()) {
    await time.reportTime(hour=6, minute=47, second=0);
}
```

```
2023-07-19T11:47:00.477Z SERIAL » 0x010a001357039f01a1254be3                                          (12 bytes)
2023-07-19T11:47:00.480Z DRIVER » [Node 087] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      75
                                  └─[Security2CCNonceGet]
                                      sequence number: 161
2023-07-19T11:47:00.483Z SERIAL « [ACK]                                                                   (0x06)
2023-07-19T11:47:00.486Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2023-07-19T11:47:00.487Z SERIAL » [ACK]                                                                   (0x06)
2023-07-19T11:47:00.490Z DRIVER « [RES] [SendData]
                                    was sent: true
2023-07-19T11:47:01.745Z SERIAL « 0x011800134b00007e00c37f7f7f7f010103000000004204000047              (26 bytes)
2023-07-19T11:47:01.751Z SERIAL » [ACK]                                                                   (0x06)
2023-07-19T11:47:01.756Z DRIVER « [REQ] [SendData]
                                    callback id:            75
                                    transmit status:        OK, took 1260 ms
                                    routing attempts:       4
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    ACK RSSI:               -61 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1
                                    beam:                   1000 ms
2023-07-19T11:47:01.774Z SERIAL « 0x011c00040057149f025501570833826f26a63e19d94d151833d01ec300ec      (30 bytes)
2023-07-19T11:47:01.776Z SERIAL » [ACK]                                                                   (0x06)
2023-07-19T11:47:01.778Z DRIVER « [Node 087] [REQ] [ApplicationCommand]
                                  └─[Security2CCNonceReport]
                                      sequence number:  85
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0x570833826f26a63e19d94d151833d01e
2023-07-19T11:47:01.787Z SERIAL » 0x012a001357239f03a2011241be12b77cb523eeb2b99bfc0cf9f7755a346a0736a (44 bytes)
                                  415a1e217ae9c2659254c2e
2023-07-19T11:47:01.794Z DRIVER » [Node 087] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      76
[... 28 lines cut ...]
2023-07-19T11:47:20.803Z SERIAL « [ACK]                                                                   (0x06)
2023-07-19T11:47:20.807Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2023-07-19T11:47:20.808Z SERIAL » [ACK]                                                                   (0x06)
2023-07-19T11:47:20.815Z DRIVER « [RES] [SendData]
                                    was sent: true
2023-07-19T11:47:22.043Z SERIAL « 0x011800134d00007c00c37f7f7f7f010103000000004204000043              (26 bytes)
2023-07-19T11:47:22.045Z SERIAL » [ACK]                                                                   (0x06)
2023-07-19T11:47:22.052Z DRIVER « [REQ] [SendData]
                                    callback id:            77
                                    transmit status:        OK, took 1240 ms
                                    routing attempts:       4
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    ACK RSSI:               -61 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1
                                    beam:                   1000 ms
2023-07-19T11:47:22.074Z SERIAL « 0x011900040057119f035600c06c77a2eeb209c1210d7d3b10c3003a            (27 bytes)
2023-07-19T11:47:22.079Z SERIAL » [ACK]                                                                   (0x06)
2023-07-19T11:47:22.082Z DRIVER « [Node 087] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 86
                                    │ security class:  S2_AccessControl
                                    └─[TimeCCTimeReport]
                                        time: 06:47:20
```

AlCalzone, 2023-07-19:
> I thought to try an unsolicited reportTime, and that seems to have worked!

Wow, that totally goes against my understanding of the specs:
> The Time Command Class, version 2 is used to **read date and time** from a supporting node in a Z
Wave network

but it totally makes sense to me that they chose to implement it that way.

Guess I can add sending an unsolicited report to the list of things the driver will try.

kdknigga, 2023-07-19:
> Wow, that totally goes against my understanding of the specs:

Well, this device might not be functioning fully within the spec.  It certainly seems to have other issues.

I have at least one more device that supports Time, so I'll try experimenting with it tonight.

kdknigga, 2023-07-20:
Here are the results of experimenting with my other lock (the one that works).

Send getTime and getDate to see the current state:
```
2023-07-20T02:37:04.001Z DRIVER » [Node 094] [REQ] [SendData]                                                                                                                                                                           │ transmit options: 0x25                                                                                                                                                                              │ callback id:      67                                                                                                                                                                                └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 159
                                    │ security class:  S2_AccessControl
                                    └─[TimeCCTimeGet]

2023-07-20T02:37:07.029Z DRIVER « [Node 094] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 202
                                    │ security class:  S2_AccessControl
                                    └─[TimeCCTimeReport]
                                        time: 21:37:09

2023-07-20T02:37:07.047Z DRIVER » [Node 094] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      68
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 160
                                    │ security class:  S2_AccessControl
                                    └─[TimeCCDateGet]

2023-07-20T02:37:07.457Z DRIVER « [Node 094] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 203
                                    │ security class:  S2_AccessControl
                                    └─[TimeCCDateReport]
                                        date: 2023-07-19
```

Ok, looks good.

Send the lock an unsolicited time report with an obviously different time:
```
2023-07-20T02:37:35.645Z DRIVER » [Node 094] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      69
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 161
                                    │ security class:  S2_AccessControl
                                    └─[TimeCCTimeReport]
                                        time: 17:00:00

2023-07-20T02:37:38.570Z DRIVER « [Node 094] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 204
                                    │ security class:  S2_AccessControl
                                    └─[TimeCCDateGet]

2023-07-20T02:37:38.589Z DRIVER » [Node 094] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      70
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 162
                                    │ security class:  S2_AccessControl
                                    └─[TimeCCDateReport]
                                        date: 2023-07-19
```

It responds with a Date Get!  That's interesting.

Now I'll query the time and date again to see if it changed based on my unsolicited time report.
```
2023-07-20T02:38:50.127Z DRIVER » [Node 094] [REQ] [SendData]                                                                                                                                                                           │ transmit options: 0x25                                                                                                                                                                              │ callback id:      71                                                                                                                                                                                └─[Security2CCMessageEncapsulation]                                                                                                                                                                     │ sequence number: 163
                                    │ security class:  S2_AccessControl
                                    └─[TimeCCTimeGet]

2023-07-20T02:38:53.190Z DRIVER « [Node 094] [REQ] [ApplicationCommand]
[... 28 lines cut ...]
                                  │ callback id:      80
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 165
                                    │ security class:  S2_AccessControl
                                    └─[TimeCCDateReport]
                                        date: 2023-07-19

...

...

...

2023-07-20T02:56:24.917Z DRIVER » [REQ] [GetBackgroundRSSI]

2023-07-20T02:56:24.926Z DRIVER « [RES] [GetBackgroundRSSI]
                                    channel 0: No signal detected
                                    channel 1: No signal detected
```

Nothing.

But, it's good to know at least two devices both set their internal clocks based off of unsolicited time reports.

I don't know if I have any other devices that support Time CC, and I'm not sure how to check without looking every one up individually in the Z-Wave Alliance's product database.

AlCalzone, 2023-07-20:
Cool, thanks for testing. I'll just build that in. If it works - good. If not we don't lose anything.

kdknigga, 2023-07-20:
Thanks a lot!

PR #6042 (2023-07-24): fix: try sending unsolicited reports to set the time/date via `Time CC`
fixes: #6032

@kdknigga this needs a test from you since I don't have any devices that only support Time CC.
Can you do that from a test build, e.g. with a [custom Docker container](https://zwave-js.github.io/zwave-js-ui/#/development/custom-docker?id=building-a-container-using-dockerfilecontrib)?
Otherwise we can just iterate after release.
closed 2023-07-24 by zwave-js-bot