zwave-js/zwave-js #6036: > Got the USB stick from Amazon with US frequency and would like to update firmware to have EU frequency instead.

MarcusSteen, 2023-07-19:
> Got the USB stick from Amazon with US frequency and would like to update firmware to have EU frequency instead.

AFAIK you can just switch the region in the settings on that firmware.

_Originally posted by @AlCalzone in https://github.com/zwave-js/node-zwave-js/issues/5910#issuecomment-1636262505_

AlCalzone, 2023-07-19:
???

MarcusSteen, 2023-07-19:
Thank you very much for the recent update of the Z-Wave JS UI module - it looks much better now:
[image]

After the recent update of the Z-Wave JS UI module it is not possible to update the frequency by using the switch in the module.
Whatever I select in the menu it always turns back to US frequency again.
Just read the change log from Silicon Labs  [https://www.silabs.com/documents/public/release-notes/SRN14889-7.18.3.0.pdf] - does the Z-Wave JS UI module take care of this remark?

[image]

The log file looks like this:
[image]

Anything that I could do different?

AlCalzone, 2023-07-19:
What you're referencing applies to the initial programming, not changing the region on the fly.
Please disable application logs, and set your driver logs to loglevel debug, then make another log of that attempt.
Please no screenshots, just copy/paste the log output here, then format it as code with this:
[image]

MarcusSteen, 2023-07-19:
`2023-07-19 13:27:52.749 INFO APP: GET /health/zwave 200 2.915 ms - 2140
2023-07-19 13:28:20.724 INFO Z-WAVE: Calling api setRFRegion with args: [ 0, [length]: 1 ]
2023-07-19T11:28:20.751Z CNTRLR   Performing soft reset...
2023-07-19T11:28:20.978Z DRIVER     no handlers registered!
2023-07-19T11:28:21.764Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2023-07-19T11:28:22.089Z DRIVER     no handlers registered!
2023-07-19T11:28:22.873Z CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try i
                                  n 1100 ms.
2023-07-19 13:28:22.925 INFO APP: GET /health/zwave 200 6.586 ms - 2140
2023-07-19T11:28:24.197Z DRIVER     no handlers registered!
2023-07-19T11:28:24.983Z CNTRLR   Soft reset failed: Timeout while waiting for an ACK from the controller (ZW020
                                  0)
2023-07-19T11:28:24.985Z CNTRLR   Waiting for the controller to reconnect...
2023-07-19T11:28:26.488Z CNTRLR   Waiting for the Serial API to start...
2023-07-19T11:28:31.491Z CNTRLR   Did not receive notification that Serial API has started, checking if it respo
                                  nds...
2023-07-19T11:28:31.520Z CNTRLR   Serial API responded
2023-07-19 13:28:31.538 INFO Z-WAVE: Success zwave api call setRFRegion true`

MarcusSteen, 2023-07-19:
2023-07-19 13:27:52.749 INFO APP: GET /health/zwave 200 2.915 ms - 2140
2023-07-19 13:28:20.724 INFO Z-WAVE: Calling api setRFRegion with args: [ 0, [length]: 1 ]
2023-07-19T11:28:20.751Z CNTRLR   Performing soft reset...
2023-07-19T11:28:20.978Z DRIVER     no handlers registered!
2023-07-19T11:28:21.764Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2023-07-19T11:28:22.089Z DRIVER     no handlers registered!
2023-07-19T11:28:22.873Z CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try i
                                  n 1100 ms.
2023-07-19 13:28:22.925 INFO APP: GET /health/zwave 200 6.586 ms - 2140
2023-07-19T11:28:24.197Z DRIVER     no handlers registered!
2023-07-19T11:28:24.983Z CNTRLR   Soft reset failed: Timeout while waiting for an ACK from the controller (ZW020
                                  0)
2023-07-19T11:28:24.985Z CNTRLR   Waiting for the controller to reconnect...
2023-07-19T11:28:26.488Z CNTRLR   Waiting for the Serial API to start...
2023-07-19T11:28:31.491Z CNTRLR   Did not receive notification that Serial API has started, checking if it respo
                                  nds...
2023-07-19T11:28:31.520Z CNTRLR   Serial API responded
2023-07-19 13:28:31.538 INFO Z-WAVE: Success zwave api call setRFRegion true

AlCalzone, 2023-07-19:
> Was this done correct?

Ehh...

I still see [application logs](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=application-logs) in there, so you did not disable them.
Also the [driver logs](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) are not on loglevel debug.
And as far as formatting goes it looks like you first pressed the code button, then inserted it. Need to do it the other way round.

MarcusSteen, 2023-07-19:
2023-07-19T14:04:01.283Z DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                                  ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                                    ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                                   ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                                  ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                                  ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
2023-07-19T14:04:01.283Z DRIVER   version 11.5.0
2023-07-19T14:04:01.283Z DRIVER   
2023-07-19T14:04:01.284Z DRIVER   starting driver...
2023-07-19T14:04:01.284Z DRIVER   opening serial port /dev/ttyUSB0
2023-07-19T14:04:01.294Z DRIVER   serial port opened
2023-07-19T14:04:01.295Z SERIAL » 0x15                                                                 (1 bytes)
2023-07-19T14:04:02.313Z DRIVER   loading configuration...
2023-07-19T14:04:02.322Z CONFIG   Using external configuration dir /data/db
2023-07-19T14:04:02.323Z CONFIG   version 11.5.0
2023-07-19T14:04:03.807Z DRIVER   beginning interview...
2023-07-19T14:04:03.808Z DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                                  1 registered
2023-07-19T14:04:03.809Z DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                                  1 registered
2023-07-19T14:04:03.809Z DRIVER   added request handler for ReplaceFailedNode (0x63)...
                                  1 registered
2023-07-19T14:04:03.809Z CNTRLR   querying controller IDs...
2023-07-19T14:04:03.816Z SERIAL » 0x01030020dc                                                         (5 bytes)
2023-07-19T14:04:03.817Z DRIVER » [REQ] [GetControllerId]
2023-07-19T14:04:03.823Z SERIAL « [ACK]                                                                   (0x06)
2023-07-19T14:04:03.824Z SERIAL « 0x01080120f11f556e0102                                              (10 bytes)
2023-07-19T14:04:03.825Z SERIAL » [ACK]                                                                   (0x06)
2023-07-19T14:04:03.827Z DRIVER « [RES] [GetControllerId]
                                    home ID:     0xf11f556e
                                    own node ID: 1
2023-07-19T14:04:03.830Z CNTRLR   received controller IDs:
                                    home ID:     0xf11f556e
                                    own node ID: 1
2023-07-19T14:04:03.830Z CNTRLR   querying API capabilities...
2023-07-19T14:04:03.834Z SERIAL » 0x01030007fb                                                         (5 bytes)
2023-07-19T14:04:03.835Z DRIVER » [REQ] [GetSerialApiCapabilities]
2023-07-19T14:04:03.843Z SERIAL « [ACK]                                                                   (0x06)
2023-07-19T14:04:03.844Z SERIAL « 0x012b01070713027a0001000df6873e88cf2bc04ffbd7fde01700008000808680b (45 bytes)
                                  a0500700000ee7fc0000000a0
2023-07-19T14:04:03.846Z SERIAL » [ACK]                                                                   (0x06)
2023-07-19T14:04:03.847Z DRIVER « [RES] [GetSerialApiCapabilities]
                                    payload: 0x0713027a0001000df6873e88cf2bc04ffbd7fde01700008000808680ba0500700
                                  000ee7fc0000000
2023-07-19T14:04:03.850Z CNTRLR   received API capabilities:
                                    firmware version:    7.19
                                    manufacturer ID:     0x027a
                                    product type:        0x01
                                    product ID:          0x0d
                                    supported functions: 
                                    · GetSerialApiInitData (0x02)
                                    · SetApplicationNodeInformation (0x03)
                                    · GetControllerCapabilities (0x05)
                                    · SetSerialApiTimeouts (0x06)
                                    · GetSerialApiCapabilities (0x07)
                                    · SoftReset (0x08)
                                    · GetProtocolVersion (0x09)
                                    · SerialAPIStarted (0x0a)
                                    · SerialAPISetup (0x0b)
                                    · SetRFReceiveMode (0x10)
                                    · FUNC_ID_ZW_SEND_NODE_INFORMATION (0x12)
                                    · SendData (0x13)
                                    · SendDataMulticast (0x14)
                                    · GetControllerVersion (0x15)
                                    · SendDataAbort (0x16)
                                    · FUNC_ID_ZW_GET_RANDOM (0x1c)
                                    · GetControllerId (0x20)
                                    · UNKNOWN_FUNC_MEMORY_GET_BYTE (0x21)
                                    · UNKNOWN_FUNC_MEMORY_PUT_BYTE (0x22)
                                    · UNKNOWN_FUNC_MEMORY_GET_BUFFER (0x23)
[... 892 lines cut ...]
2023-07-19T14:09:00.991Z DRIVER « [RES] [GetBackgroundRSSI]
                                    channel 0: -111 dBm
                                    channel 1: -111 dBm
                                    channel 2: -111 dBm
2023-07-19T14:09:02.477Z CNTRLR   Did not receive notification that Serial API has started, checking if it respo
[... cut ...]
2023-07-19T14:09:02.517Z SERIAL « [ACK]                                                                   (0x06)
2023-07-19T14:09:02.519Z SERIAL « 0x0105010b2001d1                                                     (7 bytes)
2023-07-19T14:09:02.520Z SERIAL » [ACK]                                                                   (0x06)
2023-07-19T14:09:02.521Z DRIVER « [RES] [SerialAPISetup]
                                    command: GetRFRegion
                                    region:  USA
``

MarcusSteen, 2023-07-19:
Thanks for your support.

closed 2023-07-23 by zwave-js-assistant[bot]