zwave-js/zwave-js #6037: Missing device configuration: Oomi Color Strip

polskikrol, 2023-07-19:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Oomi Color Strip

### Manufacturer ID

N/A

### Product Type

N/A

### Product ID

N/A

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/2362?selectedFrequencyId=-1

### Do you have a manual?

Attaching logs for the interview for Node 20.
[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/12095329/zwave-js-ui-store.zip)

Also, here is the manual.
[Oomi ColorStrip manual - 2.pdf](https://github.com/zwave-js/node-zwave-js/files/12095336/Oomi.ColorStrip.manual.-.2.pdf)

This is a re badge of the Aeotec LED Strip... https://products.z-wavealliance.org/products/2498?selectedFrequencyId=-1. According to manufacturer support, it can be flashed with the EXE firmware to re badge as Aeotec. Firmware 1.02 is available for this.

AlCalzone, 2023-07-19:
@zwave-js-bot import config 2362 from zwa

AlCalzone, 2023-07-19:
> Attaching logs for the interview for Node 20.
> [zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/12095329/zwave-js-ui-store.zip)

Can you try again please? Looks like the interview didn't succeed because some strange route management was going on and the stick was unresponsive.

polskikrol, 2023-07-19:
Strange. I 'healed' the node, and same issue. I also went under log, and removed the current log files. I then do a test, and both log files show 22 bytes or so. However, when I select both files and download, it is a few MB. So maybe I dont know how to clear the logs properly or something? But figured if I trashed the file, it would write a new one and not be loaded with info.

polskikrol, 2023-08-25:
I ended up using my Z-Stick 5 with the PC controller firmware updater to change the firmware to ZW121 v1.3. As this was a rebadge of the Aeon (Aeotec) LED Strip. Then I unpaired it from Z-Stick 5, repaired it with Vera Plus, and relearned the network via Z-Stick 7 and HA. Re-interviewed and it comes up fine now.

closed 2023-08-25 by polskikrol