zwave-js/zwave-js #6048: Unable to control old Leviton switches (VRS05/VRS15)

denix0, 2023-07-21:
I have several old Leviton switches (and dimmers) around the house - those are of the first generation of Z-Wave from about 10-15 years ago and I've been using them w/o problems with several different controllers and hubs throughout the years.

I'm setting up everything now in Z-Wave JS UI and noticed that I cannot control them either directly from the UI or from HomeAssistant via WebSocket. Moreover, in the UI under "Binary Switch v0" section there's "Current value" field, but there's NO "Target value" at all, which I normally get for other devices with e.g. "Binary Switch v1":

[image]
[image]

I'm using Zooz 800-series ZST39 LR stick. The problematic devices are Leviton VRS05 and VRS15 switches, but I haven't tried dimmers yet.

They report Z-Wave protocol version 2.9 and Z-Wave chip firmware version "0.4" or "0.5"

I wonder if this is due to "v0" in the "Binary Switch" class:

[image]

Or if it's a driver issue...

I would appreciate any help and gladly provide logs or try different options. Thanks in advance!

AlCalzone, 2023-07-21:
This feels like an incomplete interview. Please re-interview one of the nodes and make a driver log (loglevel debug) of that.
Ideally clear the logfile before doing that (open it in the store tab, remove everything, save). I just need the `zwavejs_... .log` file.

Also let me know the # of the node you re-interviewed

denix0, 2023-07-26:
@AlCalzone sure, can do! Please give me a bit more time - trying so shuffle bunch of instances around in between day job...

FYI, just an upfront clarification... This is not a one-off user error - I've already mentioned that I have several of such similar devices (VRS05/VRS15) and all of them show this issue. I also have at least 2 separate instances where I've tried it - HomeAssistant running Z-Wave JS + UI addons and a standalone Docker container with the same. I've tried including and excluding these devices multiple of times, as well as re-interviewing them after the fact. Seems like a systematic issue - all attempts lead to the same outcome.

Regardless, I'll get back to you with the logs you've requested.

denix0, 2023-07-28:
Ok, I've been able to confirm that it still fails exactly the same with Home Assistant default Z-Wave JS support (so it's not UI specific) and also with a different controller stick (QuickStick Combo HUSBZB-1 = Nortek Z-Wave+Zigbee)

Here are the logs of including/interviewing the device attached. Please let me know if you need anything else. Thanks!

[zwave-js-ui-store2.zip](https://github.com/zwave-js/node-zwave-js/files/12189146/zwave-js-ui-store2.zip)

AlCalzone, 2023-07-28:
The interview appears to be successful, except that the CC versions cannot be queried, which might be the reason you're not seeing the values to control the device. I'll look into that.

AlCalzone, 2023-07-28:
Ok I can reproduce. The timed out version query was the issue.

closed 2023-07-28 by zwave-js-bot

denix0, 2023-08-07:
@AlCalzone I finally got a chance to update and test the fix - everything now works. Thanks a lot!

PR #6089 (2023-07-28): fix: actually set 1 as the CC version when assuming V1 is supported
fixes: #6048