zwave-js/zwave-js #6052: Missing device configuration: <enter manufacturer and device name>

kmccb, 2023-07-21:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Inovelli NZW31T

### Manufacturer ID

0x015d

### Product Type

0x1f02

### Product ID

0x1f02

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/2679

closed 2023-07-24 by zwave-js-bot

kmccb, 2023-07-25:
I may have submitted the ticket wrong because it's still not working.. The device is an Inovelli NZW31T which is listed: https://devices.zwave-js.io/?jumpTo=0x0312:0x0111:0x261c:0.0#

However, in HA the device is discovered as 0x015d-0x1f02-0x1f02.. 

0x015d is Willis Electric Co which looks like another name for Inovelli.. 

0x1f02 needs to be moved under 0x015d (Willis)..

AlCalzone, 2023-07-25:
Are you sure your application has been updated to `node-zwave-js` version `11.6.0` yet?

kmccb, 2023-07-25:
So it's nothing on my end? I just need to wait until Z-Wave JS UI HA Addon is updated?