zwave-js/zwave-js #6055: NVM restore from an Aeotec  Z-Stick Gen5 device to an Aeotec  Z-Stick Gen7 failed

damsellem, 2023-07-24:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

PKG executable

### Z-Wave JS UI version

8.13.1.33a6f57

### ZwaveJS version

10.14.0

### Describe the bug

When restoring an Aeotec  Z-Stick Gen5 NVM backup to an Aeotec  Z-Stick Gen7, treatement fails

### To Reproduce

Backup Z-Stick Gen5 NVM.
Swap sticks.
Restore NVMxxxxxxxxxxxxxxxxx.bin 
Conversion reaches 100%
Restore aborts with the following message
**INVALID NVM JSON: Controller is part of another network but has no learned Home ID! (ZW0281)**

### Expected behavior

Restore & Conversion end succesfully.

### Additional context

Soft Reset & Hard Reset executed on Aeotec Gen 7 where useless 

[z-ui_2023-07-22_logs.zip](https://github.com/zwave-js/zwave-js-ui/files/12135763/z-ui_2023-07-22_logs.zip)

AlCalzone, 2023-07-24:
Was your Gen5 a secondary controller?

damsellem, 2023-07-24:
Hello,
thanks for your answer
it is my primary (and unique) controller
Regards

AlCalzone, 2023-07-24:
To investigate this, I'll need your NVM backups from both sticks. Send them to the email in my profile please.

damsellem, 2023-07-27:
backups sent

AlCalzone, 2023-07-27:
Interesting - there is one byte set to 255 I've never seen have that value before.

closed 2023-07-27 by zwave-js-bot

damsellem, 2023-07-27:
Great ! and how quickly !
Thank you

PR #6082 (2023-07-27): fix(nvmedit): recover from invalid controller capabilities byte in 500 series NVM
fixes: #6055