zwave-js/zwave-js #6069: Unable to write parameter with device FGKF601 (remote)

bdbogjoe, 2023-07-25:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

8.21.0.b1dc848

### ZwaveJS version

11.5.0

### Describe the bug

while trying to change any parameter i get error in logs : 

2023-07-25 16:15:04.864 INFO Z-WAVE: Writing 61 to 66-112-0-2
2023-07-25 16:15:21.926 ERROR Z-WAVE: Error while writing 61 on 112-0-2: Message dropped because of an unexpected error: Cannot read properties of undefined (reading 'throw') (ZW0202)
2023-07-25 16:15:21.930 ERROR Z-WAVE: Unable to write 61 on 112-0-2: Fail

### To Reproduce

- open panel from zwave-ju ui
- change any parameter
- submit change

### Expected behavior

parameter should be updated

AlCalzone, 2023-07-25:
Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).

bdbogjoe, 2023-07-25:
driver logs : 

2023-07-25T17:23:21.982Z SERIAL « 0x010700131f010332c4                                                 (9 bytes)
2023-07-25T17:23:21.983Z SERIAL » [ACK]                                                                   (0x06)
2023-07-25T17:23:21.985Z DRIVER « [REQ] [SendData]
                                    callback id:     31
                                    transmit status: NoAck
2023-07-25T17:23:23.475Z SERIAL » 0x010d0013420670040202003d2520e9                                    (15 bytes)
2023-07-25T17:23:23.476Z DRIVER » [Node 066] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      32
                                  └─[ConfigurationCCSet]
                                      parameter #:      2
                                      reset to default: false
                                      value size:       2
                                      value format:     SignedInteger
                                      value:            61
2023-07-25T17:23:23.484Z SERIAL « [ACK]                                                                   (0x06)
2023-07-25T17:23:23.487Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2023-07-25T17:23:23.488Z SERIAL » [ACK]                                                                   (0x06)
2023-07-25T17:23:23.491Z DRIVER « [RES] [SendData]
                                    was sent: true

AlCalzone, 2023-07-25:
Seems cut off - is there nothing after it?

bdbogjoe, 2023-07-25:
will try to get more but difficult to filter, i have lot of events

AlCalzone, 2023-07-25:
I don't mind, just include the next 15 seconds or so.

apella12, 2023-07-30:
So I'm just testing zwave-js-ui so this could be off base, but I do have this device. Are you waking it after you set the parameter (o and +).  I had no problem today changing to 61 and back to 60

closed 2023-08-10 by zwave-js-assistant[bot]