zwave-js/zwave-js #6084: Inovelli VZW31 - Param 99 - Unable to set effect to Clear(255)

stu1811, 2023-07-27:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

According the config database param 99 byte 4 is supposed to support a clear option of 255. This option does not appear in the zwavejs ui. I also cannot set 255 via a service call to  zwave_js.set_config_parameter.
https://devices.zwave-js.io/?jumpTo=0x031e:0x0015:0x0001:0.3
[image]

```
service: zwave_js.set_config_parameter
data:
  endpoint: 0
  bitmask: "0xff000000"
  parameter: "99"
  value: "255"
target:
  entity_id: light.driveway_light
```
Error:
```
Failed to call service zwave_js.set_config_parameter. Node(node_id=70) - InvalidNewValue: Must provide a value for 70-112-0-99-4278190080 that represents a valid state key or label from {"0": "Off", "1": "Solid", "2": "Fast Blink", "3": "Slow Blink", "4": "Pulse", "5": "Chase", "6": "Open/Close", "7": "Small to Big", "8": "Aurora", "9": "Slow Fall", "10": "Medium Fall", "11": "Fast Fall", "12": "Slow Rise", "13": "Medium Rise", "14": "Fast Rise", "15": "Medium Blink", "16": "Slow Chase", "17": "Fast Chase", "18": "Slow Siren", "19": "Fast Siren"}
```

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwave-js-ui: 8.21.2
zwave-js: 11.6.0

NOTE: I think I selected the correct option for "How are you using `node-zwave-js`?". I'm using the HA addon for zwave-js-ui and zwave-js integration.

Home Assistant 2023.7.3
Supervisor 2023.07.1
Frontend 20230705.1 - latest

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

Device is added with S2.
In the UI when I select debug info here param 99 effect. Notice 255 or clear is missing.
```
{
      "id": "70-112-0-99-4278190080",
      "nodeId": 70,
      "toUpdate": false,
      "commandClass": 112,
      "commandClassName": "Configuration",
      "endpoint": 0,
      "property": 99,
      "propertyName": "All LED Strip Effect - Effect",
      "propertyKey": 4278190080,
      "type": "number",
      "readable": true,
      "writeable": true,
      "label": "All LED Strip Effect - Effect",
      "default": 0,
      "stateless": false,
      "commandClassVersion": 4,
      "min": 0,
      "max": 19,
      "list": true,
      "allowManualEntry": false,
      "states": [
        {
          "text": "Off",
          "value": 0
        },
        {
          "text": "Solid",
          "value": 1
        },
        {
          "text": "Fast Blink",
          "value": 2
        },
        {
          "text": "Slow Blink",
          "value": 3
        },
        {
          "text": "Pulse",
          "value": 4
        },
        {
          "text": "Chase",
          "value": 5
        },
        {
          "text": "Open/Close",
          "value": 6
        },
        {
          "text": "Small to Big",
          "value": 7
        },
        {
          "text": "Aurora",
          "value": 8
        },
        {
          "text": "Slow Fall",
          "value": 9
        },
        {
          "text": "Medium Fall",
          "value": 10
        },
        {
          "text": "Fast Fall",
          "value": 11
        },
        {
          "text": "Slow Rise",
          "value": 12
        },
        {
          "text": "Medium Rise",
          "value": 13
        },
        {
          "text": "Fast Rise",
          "value": 14
        },
        {
          "text": "Medium Blink",
          "value": 15
        },
        {
          "text": "Slow Chase",
          "value": 16
        },
        {
          "text": "Fast Chase",
          "value": 17
        },
        {
          "text": "Slow Siren",
          "value": 18
        },
        {
          "text": "Fast Siren",
          "value": 19
        }
      ],
      "value": 0,
      "lastUpdate": 1690425721783,
      "newValue": 0
    },
```

AlCalzone, 2023-07-27:
Are you on the latest addon version?
Even if you are, I'm a bit confused why that option is missing. There weren't any changes for that configuration file after being added.

stu1811, 2023-07-27:
Doh. I had a custom template. I removed it refreshed the values. All good now.

closed 2023-07-27 by stu1811