zwave-js/zwave-js #6096: Z-WAVE-SERVER: The node failed to decode the message. (ZW1405)

Sraw, 2023-07-30:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
In the HomeAssistant, if I turn both plugs of a single device on or off, then one of them will fail and give the following message:

```
2023-07-30 10:42:02.696 ERROR Z-WAVE-SERVER: The node failed to decode the message. (ZW1405)
ZWaveError: The node failed to decode the message. (ZW1405)
at Driver.sendMessage (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4647:23)
at Driver.sendCommandInternal (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4844:28)
at Driver.sendSupervisedCommand (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4889:27)
at Driver.sendCommand (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:4937:30)
at Proxy.set (/usr/src/app/node_modules/@zwave-js/cc/src/cc/BinarySwitchCC.ts:132:24)
at Proxy.<anonymous> (/usr/src/app/node_modules/@zwave-js/cc/src/cc/BinarySwitchCC.ts:154:16)
at ZWaveNode.setValue (/usr/src/app/node_modules/zwave-js/src/lib/node/Node.ts:1062:39)
at NodeMessageHandler.handle (/usr/src/app/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:20:43)
at Object.node (/usr/src/app/node_modules/@zwave-js/server/dist/lib/server.js:42:96)
at Client.receiveMessage (/usr/src/app/node_modules/@zwave-js/server/dist/lib/server.js:107:99)
```

What do you observe?
One of the plugs doesn't turn on/off due to the error message attached above.

What did you expect to happen?
Both plugs can work seamlessly.

Steps to reproduce the behavior:
Device: Zooz | 700 Series Outdoor Double Plug | ZEN14

  1. Go to HomeAssistant, and add both plugs for this device into the dashboard so that there will be a single button to control both plugs.
  2. Click on the button in the title to control both plugs.
  3. See error.

### Device information

Manufacturer: Zooz
Model name: ZEN14
Node ID in your network: 7

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 8.21.2
`node-zwave-js` branch: N/A
`zwave-js-ui` branch: N/A

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

In version 8.21.0, although sometimes it would still throw this error, it is very rare. While in the 8.21.2 version, it fails every time.

### Attach Driver Logfile

File attached.

[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/12209879/zwavejs_current.log)

AlCalzone, 2023-07-31:
I think what's happening here is this:
The device has two outlets/endpoints, plus it reports the combined state (is any outlet on) on endpoint 0.

- **Z**-Wave JS -> **D**evice: turn outlet 1 on, tell me if it worked
- D -> Z: it worked
- Z -> D: turn outlet 2 on, tell me if it worked
- at the exact same time: D -> Z: at least one outlet is on, tell me if you understood

This results in a collision, because both those messages use the same encryption state, so the other side is unable to decode. Z-Wave JS uses a trick, so it is actually able to decode the device's command, even though it shouldn't. But the device doesn't.

Recovering from this isn't easy, especially when both sides are trying to do it at the same time.

There should be a way around this though by simply turning both outlets on with a single command. I'll try this now.

AlCalzone, 2023-07-31:
The next release will contain an updated config file for this device.
If you re-interview then, you'll see an additional binary switch (endpoint 0), which controls both outlets at once. I was able to use that without any issues.

That said - if you need to turn one on and one off at the same time, you're better off waiting a few hundred ms inbetween.

Sraw, 2023-07-31:
Thanks for the help! That makes perfect sense to me!

AlCalzone, 2023-08-01:
FYI, I could reproduce, both in tests and with a real device. I've implemented a fix for this particular collision, which fixes the issue for me.
This should help in more cases, since devices sending reports while we're trying to control them is a common scenario.

closed 2023-08-01 by zwave-js-bot