zwave-js/zwave-js #6097: Additional configuration parameters were introduced in a new firmware version and need to be added

fabnavigator, 2023-07-31:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Configuration parameters 9 - 13 need to be added to the configuration database.

### Device information

Manufacturer: Minoston
Model name: MP21ZP / MP31ZP

https://devices.zwave-js.io/?jumpTo=0x0312:0xff00:0xff0e:0.0

I attached the user manual from the manufacturer that shows the five new configuration parameters (9 - 13).

[Meter Switch User Manual.pdf](https://github.com/zwave-js/node-zwave-js/files/12210933/Meter.Switch.User.Manual.pdf)

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [X] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

N/A.

fabnavigator, 2023-07-31:
I really don't think I need one for this type of issue.

AlCalzone, 2023-07-31:
In which firmware version were those added?

fabnavigator, 2023-07-31:
1.20

AlCalzone, 2023-07-31:
Thanks! I may get to it at some point, but if someone wants to chime in, here are some instructions.

fabnavigator, 2023-07-31:
@AlCalzone, Thank you. I might take a stab at it.

fabnavigator, 2023-08-02:
@AlCalzone, I've taken a stab at this. I'm testing with three devices, all with different firmware versions (1.10.1, 1.11.1, and 1.20.1). Everything looks good so far, except I see an extra configuration parameter (LED Indicator Brightness) assigned to the device on 1.10.1. I don't know where it is coming from. Any thoughts?

Also, I found out the new parameters were introduced in version 1.11.1, not 1.20.1 as I said previously.

[image]
ess) assigned to the device on 1.10.1. I don't know where it is coming from. Any thoughts?

Driver:
```
{
	"manufacturer": "Minoston",
	"manufacturerId": "0x0312",
	"label": "MP21ZP / MP31ZP",
	"description": "Mini Plug with Power Meter",
	"devices": [
		{
			"productType": "0xff00",
			"productId": "0xff0e",
			"zwaveAllianceId": 4247
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "255.255"
	},
	"paramInformation": [
		{
			"#": "1",
			"$import": "templates/minoston_template.json#led_indicator_four_options_inverted"
		},
		{
			"#": "9",
			"$if": "firmwareVersion >= 1.11.1",
			"$import": "templates/minoston_template.json#led_indicator_brightness"
		},
		{
			"#": "2",
			"$import": "templates/minoston_template.json#auto_off_timer"
		},
		{
			"#": "3",
			"$import": "templates/minoston_template.json#auto_on_timer"
		},
		{
			"#": "4",
			"$import": "~/templates/master_template.json#state_after_power_failure_prev_off_on"
		},
		{
			"#": "10",
			"$if": "firmwareVersion >= 1.11.1",
			"$import": "templates/minoston_template.json#power_reporting"
		},
		{
			"#": "5",
			"$import": "templates/minoston_template.json#power_reporting_threshold"
		},
		{
			"#": "6",
			"$import": "templates/minoston_template.json#power_reporting_interval"
		},
		{
			"#": "11",
			"$if": "firmwareVersion >= 1.11.1",
			"$import": "templates/minoston_template.json#current_reporting"
		},
		{
			"#": "7",
			"$import": "templates/minoston_template.json#current_reporting_threshold"
		},
		{
[... 560 lines cut ...]
				"label": "Disabled",
				"value": 1
			}
		]
    },
	"current_reporting_interval": {
		"label": "Current Reporting Interval",
		"valueSize": 4,
		"unit": "minutes",
		"minValue": 1,
		"maxValue": 65535,
		"defaultValue": 60,
		"unsigned": true
    },
	"voltage_reporting_interval": {
		"label": "Voltage Reporting Interval",
		"valueSize": 4,
		"unit": "minutes",
		"minValue": 1,
		"maxValue": 65535,
		"defaultValue": 60,
		"unsigned": true
    }
}
```

AlCalzone, 2023-08-02:
> Everything looks good so far, except I see an extra configuration parameter (LED Indicator Brightness) assigned to the device on 1.10.1. I don't know where it is coming from.

Might be auto-discovered. Can you share a driver log of a re-interview on loglevel `debug` and attach it here as a file (drag & drop into the text field).

fabnavigator, 2023-08-03:
I finally got the manufacturer to admit that the documentation was incorrect. That parameter is supposed to be there in that version. I'll update my files.

fabnavigator, 2023-08-03:
I was probably supposed to leave this open until the software was updated.

AlCalzone, 2023-08-08:
Can you please provide a pull request with your changes? Makes it easier to review/do revisions.

fabnavigator, 2023-08-08:
@AlCalzone The PR workflow failed, and I don't know that to do.

AlCalzone, 2023-08-09:
All good - I left you a review for learning purposes and fixed the issues.

PR #6139 (2023-08-09): fix(config): add parameters 9-13 to Minoston MP21ZP / MP31ZP
fixes: #6097

1. Added to minoston_template.json:
power_reporting
current_reporting
current_reporting_interval
voltage_reporting_interval
2. Added to mp21zp_mp31zp.json:
parameters 9 - 13
3. Reordered parameters in mp21zp_mp31zp.json to group like parameters together
closed 2023-08-09 by AlCalzone