zwave-js/zwave-js #6098: Security info not showing correctly

bostjane, 2023-07-31:
Hi,

I wanted to move Aetoec multisenzor 6 from S0 security to no security, but wanted to keep the same node ID. So I manually reset the device and proceed with "Replace failed node" process. Device was successfully added, it works fine, but in GUI I see question mark in security column. I tried also re-interview node, but nothing helped.

How can I fix this? Version is 8.21.2
[Capture](https://github.com/zwave-js/zwave-js-ui/assets/7306524/428291f8-ee46-482f-a9b5-34aedbfba20e)

AlCalzone, 2023-07-31:
Ok I cannot reproduce. I even tried removing all security keys but S0. What I did:

- include device using S0, wait for interview to finish
- shut down Z-UI
- factory reset device
- start Z-UI
- replace failed node, select "no encryption"

Result:
[image]

bostjane, 2023-08-01:
> > So I manually reset the device and proceed with "Replace failed node" process.
> 
> When you factory reset the device ensure your stick is not connected, then use replace failed node against that node, also some nodes have a specific procedure to put them in secure inclusion, check you did that correctly. Also please enable debug logs so in case this fails we can check what's going on
> 
> Make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop into the text field).
> 
> cc @AlCalzone

Yes, I manually reset the device while stick was disconnected. Maybe the problem is because I needed to go with Replace failed node" process twice as it failed for the first time - I can't remember exactly why.
I case log doesn't tell you what is the problem maybe I can send you cache or other configuration file that is causing the problem.

[zwavejs_2023-08-01.log](https://github.com/zwave-js/node-zwave-js/files/12224728/zwavejs_2023-08-01.log)

AlCalzone, 2023-08-01:
The connection to that device is terrible - 9.6kbit/s, latencies upwards of 300ms and several timeouts during the interview. I would not be surprised if that's the reason.

AlCalzone, 2023-08-02:
Ok that interview looks totally fine. There should be a `<homeid>.jsonl` file in the store, homeid is a 8-digit hex number. Can you send me that one?

bostjane, 2023-08-03:
[zwave.zip](https://github.com/zwave-js/node-zwave-js/files/12248114/zwave.zip)

AlCalzone, 2023-08-03:
Ok looks like the info about security classes isn't being stored for that one for whatever reason.

bostjane, 2023-08-17:
Yesterday I tried another thing. I restarted ZWave JS and deleted all three configuration files that contained my HomeID. Therefore Zwave JS had to re-interview all the nodes. So it was practically the same when I moved from OZW to Zwave JS.
This time maybe 1/3 of devices got question mark on security column. But when I moved from OZW it got them all correctly (same network and devices).

I guess something changed along the way in Zwave JS that introduced a bug which prevents security info to be stored.

AlCalzone, 2023-08-17:
> This time maybe 1/3 of devices got question mark on security column.

have you got a driver log of this?

bostjane, 2023-08-17:
I tried again with my test network with just 3 nodes, actually 2 (num 3 and 5), because node 7 has dead battery and it's normal that it didn't respond.
As you can see, security info for both nodes was not saved.

[zwavejs_2023-08-17.log](https://github.com/zwave-js/node-zwave-js/files/12372055/zwavejs_2023-08-17.log)
[image]

bostjane, 2023-08-17:
And here are config files if you need them too.
[store.zip](https://github.com/zwave-js/node-zwave-js/files/12372066/store.zip)

closed 2023-08-18 by zwave-js-bot

AlCalzone, 2023-08-18:
Should be fixed in the next release (after another re-interview)

PR #6187 (2023-08-18): fix: unknown security class when node does not support S0 and/or S2
fixes: #6098