zwave-js/zwave-js #6105: Binary Switch vundefined

MickaelColombieSnC, 2023-08-01:
Hello everyone, 
I added this json (customized)
[mh-s411-7102.zip](https://github.com/zwave-js/zwave-js-ui/files/12226090/mh-s411-7102.zip)

Then, I included a MCOHome MH-S411 panel switch to Zwave JS UI.
Everything seems fine, but when I turn On the switch (using targetValue or currentValue), the currentVallue change to True then to False one sec later, but the light stay On.
If I turn Off the switch (using targetValue or currentValue), the light switch Off.

I also remarked that, just after the inclusion, I only have the Binary Switch v2 command class, but as soon as I change the switch state once, a new command class appear (Binary Switch vundefined). Those currentValue doesn't have the same problem as Binary Switch v2, but I can't trigger anything using them. (37-1-currentValue or target value won't trigger my switch, they are just following 37-0- state)
 
I also tried witthout the MH-S411's json loaded, but I get the same result.
Could someone enlight me on this issue ?

Thanks in advance !
[image]

[image]

[mh-s411 debug info.txt](https://github.com/zwave-js/zwave-js-ui/files/12226184/mh-s411.debug.info.txt)

AlCalzone, 2023-08-01:
Seems like the device is reporting from endpoints 1 and 2 after you control endpoint 0. The info above seems to indicate that the device only has one binary switch on endpoint 0 though.

Please make a [driver log](https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs) of a re-interview, loglevel `debug` and attach it here as a file (drag & drop into the text field).

MickaelColombieSnC, 2023-08-01:
[z-ui_2023-08-01.log](https://github.com/zwave-js/node-zwave-js/files/12228767/z-ui_2023-08-01.log)

Hi, here are the log, I made the reinterview, and then triggered the endpoint 0 that made the 1 and 2 appear.

AlCalzone, 2023-08-01:
That's the wrong log. Check the link I sent you again :)

MickaelColombieSnC, 2023-08-01:
[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/12229476/zwave-js-ui-store.zip)
Sorry :)

MickaelColombieSnC, 2023-08-01:
[zwave-js-ui-store (1).zip](https://github.com/zwave-js/node-zwave-js/files/12230049/zwave-js-ui-store.1.zip)
I confused zwave js ui log and driver log sorry again, this one might be good ;)

AlCalzone, 2023-08-02:
Ok it looks like the device really wants to report from endpoint 1 - although I'm not sure why they even bothered to make it have an additional endpoint. Maybe that makes the firmware more similar to the variants with more than 1 button.

We'll have to try some things to get this working properly.
Please add
```
"preserveEndpoints": [1]
```
under the `"compat"` field, then restart Z-UI so it picks up the new file, and re-interview the device again. Ideally make a log of this too in case we have to do more tests.

You should now have values on endpoint 1 to control the device instead of endpoint 0 like previously. If everything works as expected, this should also reflect the correct state.

AlCalzone, 2023-08-02:
Eh, I wasn't clear. `preserveEndpoints` is another `compat` flag, so it needs to go inside the `"compat"` object. LIke this:
```jsonc
	"compat": {
                "commandClasses" : {
                        "remove" : {
                                "0x2c": {
                                        "endpoints": "*"
                                },
                                "0x80": {
                                        "endpoints": "*"
                                }
                        }
                },
                "preserveEndpoints": [1]
        },
```

closed 2023-08-02 by MickaelColombieSnC

AlCalzone, 2023-08-02:
Not much in that log, but I trust that it works. Would you be willing to provide a pull request with that config file? We don't seem to have one with that ID yet.