zwave-js/zwave-js #6125: YRD446-ZW2 different product code, coming up as unidentified

88fingerslukee, 2023-08-03:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Hi All,

I have an YRD446-ZW2 and it is iin the DB but it has a different product code and therefore is coming up as unidentified. How do I get this added?

[image]

thanks!
88

### Device information

Manufacturer: Yale
Model name: YRD446-ZW2
Node ID in your network: 50

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

20:44:03.548 SERIAL « 0x0121000400321b9881fba0c6aa4769d530ee95ceac58ead28b4863ac1b5d50c72 (35 bytes)
5b621
20:44:03.549 SERIAL » [ACK] (0x06)
20:44:03.550 CNTRLR [Node 050] [~] [Door Lock] currentMode: 255 => 255 [Endpoint 0]
2023-08-03 20:44:03.551 INFO Z-WAVE: [Node 050] Value updated: 98-0-currentMode 255 => 255
20:44:03.551 CNTRLR [Node 050] [~] [Door Lock] outsideHandlesCanOpenDoor: false,false [Endpoint 0]
,false,false => false,false,false,false
2023-08-03 20:44:03.552 INFO Z-WAVE: [Node 050] Value updated: 98-0-outsideHandlesCanOpenDoor false,false,false,false => false,false,false,false
20:44:03.552 CNTRLR [Node 050] [~] [Door Lock] insideHandlesCanOpenDoor: false,false, [Endpoint 0]
false,false => false,false,false,false
2023-08-03 20:44:03.553 INFO Z-WAVE: [Node 050] Value updated: 98-0-insideHandlesCanOpenDoor false,false,false,false => false,false,false,false
20:44:03.553 CNTRLR [Node 050] [~] [Door Lock] doorStatus: "open" => "open" [Endpoint 0]
2023-08-03 20:44:03.553 INFO Z-WAVE: [Node 050] Value updated: 98-0-doorStatus open => open
20:44:03.554 CNTRLR [Node 050] [~] [Door Lock] boltStatus: "unlocked" => "locked" [Endpoint 0]
2023-08-03 20:44:03.554 INFO Z-WAVE: [Node 050] Value updated: 98-0-boltStatus unlocked => locked
20:44:03.554 CNTRLR [Node 050] [~] [Door Lock] latchStatus: "open" => "open" [Endpoint 0]
2023-08-03 20:44:03.554 INFO Z-WAVE: [Node 050] Value updated: 98-0-latchStatus open => open
20:44:03.555 DRIVER « [Node 050] [REQ] [ApplicationCommand]

AlCalzone, 2023-08-08:
@zwave-js-bot add fingerprint `0x8002:0x1600` to `0x0129:0x8002:0x0600:0.0`

closed 2023-08-08 by zwave-js-bot

88fingerslukee, 2023-08-10:
Thank you for doing this. How do update my device to show the proper information? When I re-interview the node it still comes up as unidentified.

AlCalzone, 2023-08-10:
This change is not yet released. When it is and your application has updated, a re-interview should pick it up.

PR #6135 (2023-08-08): fix(config): add fingerprint to Yale YRD446-ZW2
fixes: #6125