zwave-js/zwave-js #6126: OTW Update Aeotec Stick 700

higs76, 2023-08-04:
### Checklist

- [ ]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

8.16.2

### ZwaveJS version

10.21.0

### Describe the bug

First during update i've got the same bug like this issue #5360 ([https://github.com/zwave-js/node-zwave-js/issues/5360](https://github.com/zwave-js/node-zwave-js/issues/5360))
Same go to windows, install Simplicity Studio... same device not found.
Back to HA, test boot Loader mode>Save
And use OTW, select firmware file "zwave_ncp_serial_api_controller_BRD4206A" (version 7.19.2)
Updating controller firmware, please wait 0% not update % but behing my stick appear
after 20mins always 0%

Hesitate to stop or close HA Page.

[image]

### To Reproduce

Update OTW with bootloader On

### Expected behavior

Close "X" or end detection/progression.

AlCalzone, 2023-08-04:
That's still not a driver log. Please read the posted link again.

Micka41, 2023-08-04:
Sorry
[zwave-js-ui-store (1).zip](https://github.com/zwave-js/node-zwave-js/files/12258894/zwave-js-ui-store.1.zip)

Micka41, 2023-08-04:
An other without NVM backup :
[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/12258976/zwave-js-ui-store.zip)

AlCalzone, 2023-08-08:
The bootloader on your stick is aborting the update with error `0x18`. This is a question for the manufacturer, but probably means you're trying to flash a firmware that isn't compatible or not actually an update:
> https://siliconlabs.my.site.com/community/s/article/z-wave-700-otw-of-controller?language=en_US
> 2. If getting error 0x50 or 0x18 during firmware update, please make sure:
a. bootloader and encryption keys have already been flashed into the device via SWD interface before you perform the OTW/OTA update.
b. GBL file should have an increased firmware version comparing the one that the device is currently running.

closed 2023-08-21 by zwave-js-assistant[bot]