zwave-js/zwave-js #6127: Coordinator crashes after NVM backup.

hmjgriffon1, 2023-08-04:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

8.22.0

### ZwaveJS version

11.8.0

### Describe the bug

When NVM backup runs, coordinator becomes unresponsive, devs asked me to post logs here.

### To Reproduce

I can manually run an NVM backup.

### Expected behavior

backups to run without issue.

### Additional context

I've got the 700 series znet homeseer coordinator if that helps.
[zwave-js-ui-files.zip](https://github.com/zwave-js/zwave-js-ui/files/12263028/zwave-js-ui-files.zip)

asayler, 2023-08-04:
Seeing this issue as well via the Z-wave JS UI add-on in HA.

- Z-Wave JS UI Add On 1.15.3
- Zooz ZAC93 800-series GPIO controller running FW v1.0
- Logs: https://gist.githubusercontent.com/asayler/8584c01835dcbe6f613f70ce16d322fa/raw/5c61334d517a7854e26d7b2723b17aa4e17eba94/zwavejs_2023-08-04.log (issue starts around the `2023-08-04T10:00` timestamp just after the NVM backup and soft reset completes).

It's not yet clear to me if the backup itself is triggering the issue, or if the associated soft reset is. These backups/restes were working fine in the prior add on version, however.

Lumute, 2023-08-04:
I have the same issue, it started after upgrading the Z-Wave JS Add-on to 1.15.3 during nightly automated NVM backups

I have two Z-Wave networks both upgraded the same day and it is happening on both of them, I do have the Soft Reset enabled but it has always been like that. Here are my logs:

Z-Wave Network 1 using UZB-7 v7.17.2: [link](https://gist.githubusercontent.com/Lumute/248cb265426f5a7f27e551b2a754d6d5/raw/6d73e2c788c1d5ed110cdc7cd92cf78300fe1faf/zwavejs_1_2023-08-04.log)
  NVM backup at 2023-08-04T08:30:18.253Z on Silly log level
  NVM backup at 2023-08-04T15:35:18.000Z on Debug log level

Z-Wave Network 2 using UZB-7 v7.19.3: [link](https://gist.github.com/Lumute/248cb265426f5a7f27e551b2a754d6d5/raw/6d73e2c788c1d5ed110cdc7cd92cf78300fe1faf/zwavejs_2_2023-08-04.log)
  NVM backup at 2023-08-04T08:40:18.465Z on Silly log level
  NVM backup at 2023-08-04T15:30:18.755Z on Debug log level

tokyotexture, 2023-08-04:
Getting very similar errors making my network unusable (very erratic/unstable) after upgrading to 1.15.3. I have not set up backups, so this issue might get triggered by more than just backups. I didn't save logs and have already rolled back, but I was seeing a lot of 

```
2023-08-03T16:53:52.974Z DRIVER   Dropping message because it could not be deserialized: The command class Contr
oller Replication is not implemented (ZW0303)
```

I was also getting really weird device ID numbers (like 7000-something) randomly while restarting and other funky stuff. Rolling back to 1.15.0 and everything is back to normal.

jeffjensen, 2023-08-06:
Encountering this as well.

ColColonCleaner, 2023-08-06:
Same problem here. Every day after the backup all commands start dropping. The workaround I've been using is just going to to the settings in zui and clicking save so it restarts the coordinator. Unplugging the stick and plugging it back in also seems to solve it until the next day. Disabling backups until this is resolved.

epower53, 2023-08-06:
Same issue... automated nightly backup process causes my network to lock up. For now I've downgraded to JS-UI 8.21.2 w/ zwave-JS 11.6, and all is good. Will wait until this issue has been rooted out before I attempt any more updates.

closed 2023-08-07 by zwave-js-bot

AlCalzone, 2023-08-07:
Will be fixed in 11.9.1 (in a couple of minutes).