zwave-js/zwave-js #6128: Receiving node id out range (> 256)

ocalvo, 2023-08-06:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

YES, BUT a Z-Wave JS UI developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Unknown.

What do you observe?
Incoming notifications with node id that are out of range (> 256).

What did you expect to happen?
Notifications coming from node id that are actually part of the network.

Steps to reproduce the behavior:
  1. Wait 24-48 hours
  2. Notifications start to arrive with invalid node ids.

### Device information

Manufacturer: Aeotec
Model name: GEn7 USB
Node ID in your network: 1

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwave-js-ui` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

```
2023-08-02T14:23:20.585Z SERIAL » 0x01030016ea                                                         (5 bytes)
2023-08-02T14:23:20.588Z DRIVER » [REQ] [SendDataAbort]
2023-08-02T14:23:20.594Z SERIAL « [ACK]                                                                   (0x06)
2023-08-02T14:23:21.094Z CNTRLR   [Node 050] ping failed: Timeout while waiting for a callback from the controll
                                  er (ZW0200)
2023-08-02T14:23:21.100Z SERIAL » 0x012200a900010032149f03ff001dfca3d98136c379e87e2ab6491874da2500000 (36 bytes)
                                  00081f7
2023-08-02T14:23:21.110Z DRIVER » [Node 050] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      129
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 255
                                    │ security class:  S2_Authenticated
                                    └─[SupervisionCCGet]
                                      │ session id:      5
                                      │ request updates: true
                                      └─[MultilevelSwitchCCSet]
                                          target value: 255
                                          duration:     default
2023-08-02T14:23:21.117Z SERIAL « [ACK]                                                                   (0x06)
2023-08-02T14:23:21.124Z SERIAL « 0x010401a90152                                                       (6 bytes)
2023-08-02T14:23:21.128Z SERIAL » [ACK]                                                                   (0x06)
2023-08-02T14:23:21.130Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2023-08-02T14:23:25.745Z SERIAL « 0x012300a80001711a9f037800f202ed3578af76e210bd4da39a31c26b3c69312fb (37 bytes)
                                  8c800b853
2023-08-02T14:23:25.747Z DRIVER   Dropping message because it could not be deserialized: The command class Node 
                                  Provisioning is not implemented (ZW0303)
2023-08-02T14:23:25.749Z SERIAL » [ACK]                                                                   (0x06)
2023-08-02T14:23:26.810Z SERIAL « 0x011200a800011a097105000000ff07080000b063                          (20 bytes)
2023-08-02T14:23:26.812Z SERIAL » [ACK]                                                                   (0x06)
2023-08-02T14:23:26.814Z CNTRLR   [Node 2417] is unknown - discarding received command...
2023-08-02T14:23:26.817Z DRIVER « [Node 2417] [REQ] [BridgeApplicationCommand]                       [discarded]
                                  │ target node: 282
                                  └─[NoOperationCC]
                                      payload: 0x00ff07
2023-08-02T14:23:26.920Z SERIAL « 0x010d00a800011a043003ff0c00b035                                    (15 bytes)
2023-08-02T14:23:26.922Z DRIVER   Dropping message because it could not be deserialized: The command class unkno
                                  wn (0xff0c) is not implemented (ZW0303)
2023-08-02T14:23:26.927Z SERIAL » [ACK]                                                                   (0x06)

```

ocalvo, 2023-08-06:
Note the node id with 2417, This is indicating some kind of data corruption.

closed 2023-08-07 by zwave-js-bot

AlCalzone, 2023-08-07:
Will be fixed in 11.9.1 (in a couple of minutes).