zwave-js/zwave-js #6131: Loose all nodes after updating from 8.22.0 to 8.22.1

jyvern, 2023-08-08:
### Checklist

- [ ]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

8.22.1

### ZwaveJS version

11.8.0

### Describe the bug

After autoupdating with watchtower to version 8.22.1, all nodes disapeared and I have this kind of message :
2023-08-07T20:27:24.115Z CNTRLR   [Node 006] is unknown - discarding received command...
2023-08-07T20:27:24.116Z DRIVER « [Node 006] [REQ] [ApplicationCommand]                              [discarded]
                                  └─[CRC16CCCommandEncapsulation]
                                    └─[MeterCCReport]
                                        type:        Electric
                                        scale:       V
                                        rate type:   Consumed
                                        value:       233.207
                                        time delta:  5 seconds
                                        prev. value: 233.71

### To Reproduce

Just try the 8.22.1 version

### Expected behavior

No more nodes displayed in UI

AlCalzone, 2023-08-08:
I can reproduce with my old Razberry.

jonofmac, 2023-08-08:
I did a boot with the log set to silly. Not a lot in it after several seconds. 

`2023-08-08T08:08:56.777Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2023-08-08T08:08:56.781Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2023-08-08T08:08:58.024Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2023-08-08T08:08:58.557Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2023-08-08T08:08:58.634Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2023-08-08T08:09:00.178Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2023-08-08T08:09:00.718Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2023-08-08T08:09:03.361Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2023-08-08T08:09:03.897Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2023-08-08T08:09:07.790Z DRIVER   Dropping message because network keys are not set or the driver is not yet rea
                                  dy to receive secure messages.
2023-08-08T08:09:07.794Z DRIVER   Dropping message because network keys are not set or the driver is not yet rea
                                  dy to receive secure messages.
2023-08-08T08:09:08.437Z DRIVER   Dropping message because network keys are not set or the driver is not yet rea
                                  dy to receive secure messages.
2023-08-08T08:09:10.218Z DRIVER   Dropping message because network keys are not set or the driver is not yet rea
                                  dy to receive secure messages.
2023-08-08T08:09:10.763Z DRIVER   Dropping message because network keys are not set or the driver is not yet rea
                                  dy to receive secure messages.
2023-08-08T08:09:11.298Z DRIVER   Dropping message because network keys are not set or the driver is not yet rea
                                  dy to receive secure messages.
2023-08-08T08:09:13.603Z DRIVER   Dropping message because network keys are not set or the driver is not yet rea
                                  dy to receive secure messages.
2023-08-08T08:09:14.137Z DRIVER   Dropping message because network keys are not set or the driver is not yet rea
                                  dy to receive secure messages.
2023-08-08T08:09:20.266Z DRIVER   Dropping message because network keys are not set or the driver is not yet rea
                                  dy to receive secure messages.
2023-08-08T08:09:20.808Z DRIVER   Dropping message because network keys are not set or the driver is not yet rea
                                  dy to receive secure messages.`

closed 2023-08-08 by zwave-js-bot

ocalvo, 2023-12-15:
I have lost about 50% of the nodes from my network. Down to 63 nodes from 138 nodes.

What steps can I take to recover without having to readd all the nodes?