zwave-js/zwave-js #6143: Zipato keyboard changes state on its own

OUARZA, 2023-08-10:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Many of us have encountered a problem with the Zipato keyboard since using Zwave-JS.
I have the impression that when the equipment wakes up (at regular intervals), the last command played (passing a badge for example), is replayed even though no one has passed the badge.
Can you analyze the situation?
Version 8.22.0

What do you observe?

What did you expect to happen?

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

[image]

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwave-js-ui` branch:
8.22.0

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### If yes, where did it work?

I switched from Open Zwave to ZwaveJS

### Attach Driver Logfile

I don't know how to generate the file.
I use the Zwave-JS plugin from Jeedom.
I have access to the Zwave-JS UI, I look later how to find the file.

AlCalzone, 2023-08-10:
@OUARZA check the link the bot posted. How to get the file is explained below the section on creating them.

OUARZA, 2023-08-10:
@AlCalzone 
Yes yes I do that as soon as possible.
Do we agree that you need a log when the problem occurs?

AlCalzone, 2023-08-10:
correct

OUARZA, 2023-08-13:
@AlCalzone 
Good morning,
I just added the logs to my original post.
I had the problem today at 9:27'23 with node 49 and at 9:39'28 with node 50.
Mathieu

AlCalzone, 2023-08-14:
@OUARZA for some reason, the keypad is considered to be operating in "pull" mode, where the controller has to actively read its notifications from time to time. A keypad should rather be operating in "push" mode where it sends notifications to the controller.

To know why it is detected as "pull", I'll need another driver log, this time when re-interviewing one of the keypads. Remember to wake it up after starting the re-interview, so it actually gets done.

OUARZA, 2023-08-14:
@AlCalzone 
Good morning,
If I understand correctly, you want the log when I activated a mode for example and swiped a badge?

AlCalzone, 2023-08-14:
Do this:
[image]

And then wake up the device manually. According to [this manual](https://www.zipato.com/wp-content/uploads/2015/07/wt-rfid-Zipato-Mini-Keypad-RFiD-Z-Wave-User-Manual-v1.4.pdf), swiping a tag should wake it up.
You'll see the interview status change when it does:
[image]

OUARZA, 2023-08-14:
@AlCalzone 
It's done.
It changed from nodeinfo to Complete.
[zwave-js-ui-store (1).zip](https://github.com/zwave-js/node-zwave-js/files/12334135/zwave-js-ui-store.1.zip)

AlCalzone, 2023-08-14:
Thank, I think I found the issue. Will be fixed next release.

closed 2023-08-14 by zwave-js-bot

OUARZA, 2023-08-14:
thank you so much

PR #6157 (2023-08-14): fix: only rely on `AGI CC` to determine notification mode if supported
Without this check, `groupsIssueingNotifications` will always be an empty array for nodes that don't support AGI CC. This will incorrectly consider them using pull mode

fixes: #6143